olikraus/u8g2 #746: 128x32 SSD1306 glitches on 8MHz

g2tec, 2018-11-09:
Good day.

I'm driving a small 128x32 SSD1306 i2c OLED display with a ProMini clone running in 8MHz.

The testing gig was a 16MHz Nano (328p) and everything worked flawslessly.

When porting to the 8MHz Promini (328p @ 8MHz) the drawXMB command works as intended, but a simple print shows garbage.

Uploading the exact same code set to 16MHz solves the issue, but everything runs slower.

The init I found to work is "U8G2_SSD1306_128X32_UNIVISION_1_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);"

What can be done to solve this issue?

Thank you, sorry for the bad english.

g2tec, 2018-11-09:
Managed to narrow the issue.

Not library involved.

Sorry!

closed 2018-11-09 by g2tec

olikraus, 2018-11-09:
No problem. :)