olikraus/u8g2 #748: Arduino crash after update library

wchpikus, 2018-11-15:
Hello
After update library to newest version, esp8622 crash.
2.22.18 works ok.
2.23.18 works ok

logs with working firmware and working library:
__________________________

Firmware: 
    Chip Id: 001DA1DF
    Core version: 2_4_1
    SDK version: 2.2.1(cfd48f3)
    Boot version: 31
    Boot mode: 1
__________________________

Flash chip information: 
    Flash chip Id: 001625C2 (for example: Id=001640E0  Manuf=E0, Device=4016 (swap bytes))
    Sketch thinks Flash RAM is size: 4.00 MB
    Actual size based on chip Id: 4.00 MB ... given by (2^( "Device" - 1) / 8 / 1024
    Flash frequency: 40.00 MHz
    Flash write mode: DOUT
__________________________

File system (SPIFFS): 
    Total KB: 2880.13 KB
    Used KB: 0.00 KB
    Block size: 8192
    Page size: 256
    Maximum open files: 5
    Maximum path length: 32

SPIFFS directory {/} :
__________________________

CPU frequency: 80 MHz

#####################
contacts count= .
** Search aux module **
......network volvo finded!....
Connecting to volvo_aux
.................WiFi connected
IP address: 
192.168.11.1
GATEWAY: 192.168.11.254
gateway
MDNS responder started
HTTP server started
MAC_1 = c8:d7:b0:1a:34:e4

logs from 2.24.3 library:

          Soft WDT reset

                        ctx: cont
                                  sp: 3fff38a0 end: 3fff3ca0 offset: 01b0

>>>stack>>>
           3fff3a50:  00000018 0000005c 3fff3b78 40210239
3fff3a60:  00000017 000000f4 3fff28d7 4021020b
3fff3a70:  0000005c 00000020 3fff3b78 4020a7d1
3fff3a80:  3fff56ac 00000020 3fff3b78 40201c89
3fff3a90:  3fff28d4 003f8000 3fff47c0 3fff5698
3fff3aa0:  3fff3b10 00000001 3fff3b78 4020a831
3fff3ab0:  00000000 00000000 0000001f 40201dc6
3fff3ac0:  0000001c 0000001f 00000020 3fff20dc
3fff3ad0:  00000000 00000000 3fff3b78 40201a3e
3fff3ae0:  40201090 3fff1ee4 3fff3b78 00000000
3fff3af0:  00000000 00000000 3fff3b78 40201e7f
3fff3b00:  00000001 00000040 3fff3b78 4020a890
3fff3b10:  3fff20d4 3f000020 3fff3b78 402012ee
3fff3b20:  3fff1ee4 3fff1aa1 3fff3b78 40201188
3fff3b30:  00000008 00000008 00000001 40201118
3fff3b40:  3fff3b78 00000040 00000008 0000ffff
3fff3b50:  3fff1ee4 3ffe9fbc 3fff3b78 402011e7
3fff3b60:  3fff1ee4 3ffe9fbc 00000078 4020de8a
3fff3b70:  40106fd0 00000000 3ffeb7c4 00000000
3fff3b80:  40201e14 40201c2c 402101cc 402100b0
3fff3b90:  00000019 00000000 001c1ee4 00001dff
3fff3ba0:  0000ff5c ffffff00 ffffffff 020fffff
3fff3bb0:  ffffff05 ffffffff 3fffffff 4020a668
3fff3bc0:  3ffeb740 3fff20d4 01000008 00000040
3fff3bd0:  00400000 00400100 01000000 00400000
3fff3be0:  ffff0000 ffff0000 00000000 40207f20
3fff3bf0:  3fff27d4 00070020 00002000 00000000
3fff3c00:  3fff1e00 3fff1aa1 3fff1ee4 40201a60
3fff3c10:  3fff14a0 3fff29d0 3fff1ee4 40201b01
3fff3c20:  010101e4 40201b5c 00000000 40207e8e
3fff3c30:  4020231a 000003e8 3fff1ee4 00000000
3fff3c40:  3fff1ee4 3fff1aa1 3fff2bb8 4020b2fb
3fff3c50:  feefeffe feefeffe feefeffe feefeffe
3fff3c60:  0000000f 00000001 feefeffe feefeffe
3fff3c70:  feefeffe feefeffe feefeffe 3fff2c78
3fff3c80:  3fffdad0 00000000 3fff2c70 4021757c
3fff3c90:  feefeffe feefeffe 3fff2c80 40100710
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v614f7c32
~ld

I spend wto days to searching code and after downgrade library to 2.23.18 everything come back to normal work..
My exception decode not work in arduino, so someone can decode this error..

wchpikus, 2018-11-16:
Oled ssd1322 spi.
I check this (speed) and give you sign;)

wchpikus, 2018-11-16:
#if ARDUINO >= 10600
      /* not sure when the setClock function was introduced, but it is there since 1.6.0 */
      /* if there is any error with Wire.setClock() just remove this function call */
      //Wire.setClock(u8x8->bus_clock); 
#endif

**still crash..**
 Soft WDT reset

                        ctx: cont
                                  sp: 3fff3e10 end: 3fff4200 offset: 01b0

>>>stack>>>
           3fff3fc0:  00000018 0000005c 3fff40e8 4020f9e5
3fff3fd0:  00000017 00000054 3fff2e36 4020f9b7
3fff3fe0:  0000005c 00000020 3fff40e8 40209865
3fff3ff0:  3fff59dc 00000020 3fff40e8 40201c89
3fff4000:  3fff2e34 003f8000 3fff4c68 40206e29
3fff4010:  3fff4080 00000001 3fff40e8 402098c5
3fff4020:  4000050c 003f80fe 003f8100 40201dc6
3fff4030:  0000001c 0000001f 00000020 3fff263c
3fff4040:  00000000 00000000 3fff40e8 40201a3e
3fff4050:  3fff2ea0 00000001 3fff40e8 00000000
3fff4060:  00000000 00000000 3fff40e8 40201e7f
3fff4070:  00000001 00000040 3fff40e8 40209924
3fff4080:  3fff2634 3f000020 3fff40e8 402012ee
3fff4090:  3fff2444 3fff2011 3fff40e8 40201188
3fff40a0:  00000008 00000008 00000001 40201118
3fff40b0:  3fff40e8 00000040 00000008 00000000
3fff40c0:  0000001d 00000003 3fff40e8 402011e7
3fff40d0:  0000001d 00000003 00000001 4020d854
3fff40e0:  40106fd0 00000000 3ffebd14 402098f4
3fff40f0:  40201e14 40201c2c 4020f978 4020f85c
3fff4100:  0000005c 00000000 001c0020 00002eff
3fff4110:  0000ff00 ffffff00 ffffffff 020fffff
3fff4120:  ffffff05 ffffffff 3fffffff 402096fc
3fff4130:  3ffebc90 3fff2634 01000008 00000040
3fff4140:  00400000 00400100 01000000 00400000
3fff4150:  ffff0000 ffff0000 00000000 40207f20
3fff4160:  4010500c 0011b8f2 3fff2f30 00000000
3fff4170:  3fff1a00 3fff2f30 3fff31e0 3fff2f30
3fff4180:  3fff2444 00000000 40216cbc 3fff3101
3fff4190:  0101001a 40201b5c 00000000 4020230f
3fff41a0:  3fff2444 3fff2011 3fff3118 4020a384
3fff41b0:  feefeffe feefeffe feefeffe feefeffe
3fff41c0:  00000009 00000001 feefeffe feefeffe
3fff41d0:  feefeffe feefeffe feefeffe 3fff31d8
3fff41e0:  3fffdad0 00000000 3fff31d0 40216d00
3fff41f0:  feefeffe feefeffe 3fff31e0 40100710
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v614f7c32
~ld

olikraus, 2018-11-16:
hmm.. then I do not know. This was (from my perspective) the only critical change for. Of course there are many other changes, but i think nothing which could cause a reset.

What examples are affected? Or: Is there any example which still works?

wchpikus, 2018-11-16:
Affected soft reset..nothing on diplay.
On startup i have some animation.

https://youtu.be/HCaQ4OEMig0

The same dump compiled with older version start normal, so it must be something with spi i thing.

olikraus, 2018-11-16:
I can not debug your code. It might be helpful if you could cut down your code until you find the statement which crashes your code. Or use one of the existing examples from u8g2. If they crash, it could be more easier to find the problem.

olikraus, 2018-11-17:
I have tested the latest u8g2 version with my Adafruit Feather ESP8266 board with ESP8266  v2.4.2 Environment on an Arduino 1.8.4 IDE.

Everything works as expected. No crash visible.

wchpikus, 2018-11-17:
i do not know..
crash when display data...
 u8g2.begin(); 
  Serial.begin(115200);
  delay(10);
  SPIFFS.begin();
  
if(start_screen == 0){
  #ifdef rotation_180
  u8g2.setDisplayRotation(U8G2_R2);
  #endif 
  start_screen = 1;
}
  
  if(start_screen == 1){
  u8g2.setContrast(backligth);  
  delay(10);
  u8g2.setBitmapMode(true /* transparent*/);
  u8g2.drawXBMP(0, 0, 256, 64, welcomex256);
  u8g2.sendBuffer();   //somewhere here is crash 
  delay(1000); 
  u8g2.firstPage();
  animation();
  u8g2.drawXBMP(96, -1, 64, 64, rot1);   //-1 is allowed?
  u8g2.sendBuffer();
  delay(1000);
  start_screen = 2;  
  }

olikraus, 2018-11-17:
Hmm... regarding the width of 256 for the bitmap. Did you enable 16 bit mode?
what is your display?

wchpikus, 2018-11-17:
Yes, 16bit enable

olikraus, 2018-11-17:
Also note, that the 16 mode will be disabled during update/downgrade.

wchpikus, 2018-11-17:
u8g2.drawXBMP(0, 0, 256, 64, welcomex256);

static unsigned char welcomex256[] U8X8_PROGMEM = {
0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0xfc, 0x3f,
   0xc0, 0xff, 0x1f, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x00, 0xc0, 0x07, 0xc0, 0x43, 0x00, 0x30, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x70, 0x00, 0x00, 0x8e, 0x00, 0x30, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x0c, 0x00, 0x00,
   0x30, 0x03, 0x30, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x03, 0x00, 0x00, 0xc0, 0x02, 0x30, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0xc0, 0x7f, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x80, 0x01, 0x00, 0x00, 0x80, 0x03, 0x30, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x00, 0xf8, 0xff, 0x03, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x60, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x30, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0xfe, 0xff, 0x0f, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x30, 0x00, 0x00, 0x00, 0x00, 0x00, 0x30, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0xfe, 0xff, 0x0f,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x7c, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x18, 0x00, 0xf8, 0x1f, 0x00, 0x00, 0x20, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x80, 0xff, 0xff, 0x3f, 0x00, 0x00, 0x00, 0x00,
   0xe0, 0xff, 0x07, 0x00, 0x00, 0x00, 0x00, 0x00, 0x0c, 0x00, 0xff, 0xff,
   0x01, 0x00, 0x20, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x80, 0xff, 0xff, 0x3f, 0x00, 0x00, 0x00, 0x00, 0xf8, 0xff, 0x3f, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x06, 0xc0, 0xff, 0xff, 0x07, 0x00, 0x20, 0x00,
   0x00, 0x00, 0x80, 0xff, 0x01, 0x00, 0x00, 0x00, 0xc0, 0xff, 0xfd, 0x7f,
   0x00, 0x00, 0x00, 0x00, 0xfe, 0xff, 0x7f, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x02, 0xf0, 0x03, 0xc0, 0x1f, 0x00, 0x22, 0x00, 0x00, 0x00, 0xe0, 0xff,
   0x0f, 0x00, 0x00, 0x00, 0xe0, 0xff, 0xfd, 0xff, 0x00, 0x00, 0x00, 0x00,
   0xff, 0xff, 0xff, 0x01, 0x00, 0x00, 0x00, 0x00, 0x01, 0xf8, 0x00, 0x00,
   0x3f, 0x00, 0x21, 0x00, 0x00, 0x00, 0xf0, 0xff, 0x1f, 0x00, 0x00, 0x00,
   0xf0, 0xff, 0xf1, 0xff, 0x01, 0x00, 0x00, 0xc0, 0xff, 0xff, 0xff, 0x03,
   0x00, 0x00, 0x00, 0x80, 0x01, 0x3e, 0x00, 0x00, 0x7c, 0x80, 0x25, 0x00,
   0x00, 0x00, 0xfe, 0xff, 0xff, 0x00, 0x00, 0x00, 0xf0, 0xff, 0xe1, 0xff,
   0x01, 0x00, 0x00, 0xe0, 0xff, 0xff, 0xff, 0x07, 0x00, 0x00, 0x00, 0xc0,
   0x00, 0x0f, 0x00, 0x00, 0xf0, 0x00, 0x2b, 0x00, 0x00, 0x00, 0xff, 0xff,
   0xff, 0x01, 0x00, 0x00, 0xf0, 0xff, 0xe1, 0xff, 0x01, 0x00, 0x00, 0xe0,
   0xff, 0xff, 0xff, 0x0f, 0x00, 0x00, 0x00, 0xc0, 0x80, 0x07, 0x00, 0x00,
   0xe0, 0x01, 0x1a, 0x00, 0x00, 0x80, 0xff, 0xff, 0xff, 0x03, 0x00, 0x00,
   0xf0, 0xff, 0x81, 0xff, 0x01, 0x00, 0xe0, 0xff, 0xff, 0xff, 0xff, 0xff,
   0x07, 0x00, 0x00, 0x60, 0xc0, 0x03, 0x00, 0x00, 0xc0, 0x03, 0x06, 0x00,
   0x00, 0xc0, 0xff, 0xff, 0xff, 0x07, 0x00, 0x00, 0xf0, 0xff, 0x81, 0xff,
   0x01, 0x00, 0xf8, 0xff, 0xff, 0xff, 0xff, 0xff, 0x1f, 0x00, 0x00, 0x60,
   0xc0, 0x01, 0x00, 0x00, 0x80, 0x03, 0x04, 0x00, 0x00, 0xe0, 0xff, 0xff,
   0xff, 0x0f, 0x00, 0x00, 0xf0, 0xff, 0x11, 0xff, 0x01, 0x00, 0xfe, 0xff,
   0xff, 0xff, 0x0f, 0x00, 0x7c, 0x00, 0x00, 0x20, 0xe0, 0x01, 0x00, 0x00,
   0x00, 0x07, 0x0c, 0x00, 0x00, 0xf0, 0xff, 0xff, 0xff, 0x1f, 0x00, 0x00,
   0xf0, 0xff, 0x11, 0xfe, 0x01, 0x00, 0xfe, 0xff, 0xff, 0xff, 0x01, 0x00,
   0xe0, 0x00, 0x00, 0x30, 0xf0, 0x00, 0x00, 0x00, 0x00, 0x07, 0x08, 0x00,
   0x00, 0xf0, 0xff, 0xff, 0xff, 0x1f, 0x00, 0x00, 0xf0, 0xc7, 0x71, 0xfc,
   0x01, 0x00, 0xff, 0xff, 0xff, 0xff, 0x00, 0x00, 0xc0, 0x00, 0x00, 0x30,
   0x70, 0x00, 0x00, 0x00, 0x00, 0x0e, 0x08, 0x00, 0x00, 0xf8, 0xff, 0xff,
   0xff, 0x3f, 0x00, 0x00, 0xf0, 0xc7, 0x71, 0xfc, 0x01, 0x80, 0xff, 0xff,
   0xff, 0x7f, 0x00, 0x00, 0xc0, 0x01, 0x00, 0x3c, 0xf8, 0xff, 0xff, 0xff,
   0xff, 0x0f, 0x3e, 0x00, 0x00, 0xfc, 0xff, 0xff, 0xff, 0x7f, 0x00, 0x00,
   0xf0, 0x87, 0x31, 0xfc, 0x01, 0x80, 0xcf, 0xe3, 0x18, 0x3f, 0xfc, 0x9f,
   0x87, 0x03, 0x00, 0xf4, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0x2f, 0x00,
   0x00, 0xfc, 0x3f, 0xfe, 0xff, 0x7f, 0x00, 0x00, 0xf0, 0x0f, 0x11, 0xfe,
[... 79 lines cut ...]
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x80, 0x03, 0x00, 0x00, 0x80, 0x03, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
[... cut ...]
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00 };

wchpikus, 2018-11-17:
After upgrade i uncomments #DEFINE 16BITS on u8g2.h

wchpikus, 2018-12-02:
any update?
i still using 2.23.18, newest version still crash..

wchpikus, 2018-12-03:
Please give me your @, my is wchpikus@wp.pl.
I send you some simple code

wchpikus, 2018-12-03:
ok, i have it..could you tell me something:
on main file I declared:
U8G2_SSD1322_NHD_256X64_F_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 15, /* dc=*/ 2, /* reset=*/ 5); 
and it is ok.
But what i have to add in other function, what using this library, located in other file?
When i repeat this:
U8G2_SSD1322_NHD_256X64_F_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 15, /* dc=*/ 2, /* reset=*/ 5); 
it made crash..
but only after upgrade..

e.g..
on other file i have function:
#include <ESP8266WiFi.h>
#include <U8g2lib.h>

void animation(){
  int delay_value = 20;
  int i;
  uint8_t c =96; //ustawienie grafiki tak jak welcomex256 znak volvo 120
  int d =-1; //ustawienie grafiki welcomex256 znak volvo -1
  u8g2.setBitmapMode(true /* transparent*/);
  u8g2.firstPage();
  uint8_t bitmap;
  u8g2.setDisplayRotation(U8G2_R2);
  
  for (i = 0; i<=24; i++){
    u8g2.drawXBMP(120-i, -1, 64, 64, rot1); //0=-1
    u8g2.sendBuffer();
    delay(10);
    u8g2.firstPage();
  }.........

and 'u8g2' was not declared in this scope on line u8g2.setBitmapMode(true /* transparent*/);
So it look no declared driver..

On previous i put on the function U8G2_SSD1322_NHD_256X64_F_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 15, /* dc=*/ 2, /* reset=*/ 5); and it works..

wchpikus, 2018-12-03:
Ok i try this.
Tnx for response;)

wchpikus, 2018-12-04:
ok i done this.
in global:
U8G2_SSD1322_NHD_256X64_F_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 15, /* dc=*/ 2, /* reset=*/ 5);  
U8G2 *u8g2_ptr;

in setup:
 U8G2 *u8g2_ptr = &u8g2_obj;
 and error compilation:
'u8g2_obj' was not declared in this scope (setup)

wchpikus, 2018-12-04:
ok in global added:
U8G2_SSD1322_NHD_256X64_F_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 15, /* dc=*/ 2, /* reset=*/ 5);  // Enable U8G2_16BIT in u8g2.h hardware
U8G2_SSD1322_NHD_256X64_F_4W_HW_SPI u8g2_obj(U8G2_R0, /* cs=*/ 15, /* dc=*/ 2, /* reset=*/ 5);  
U8G2 *u8g2_ptr; 

in setup:
U8G2 *u8g2_ptr = &u8g2_obj;

in other file:
extern U8G2 *u8g2_ptr;

ok but i can leave U8G2_SSD1322_NHD_256X64_F_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 15, /* dc=*/ 2, /* reset=*/ 5); on main file?
without it in main not works compilation.
'u8g2' was not declared in this scope

when i add this line in global, in main is ok, but in other files:
'u8g2' was not declared in this scope
in first finded function e.g u8g2->setBitmapMode

wchpikus, 2018-12-04:
ok i tested it and:

in global:
U8G2_SSD1322_NHD_256X64_F_4W_HW_SPI u8g2_obj(U8G2_R0, /* cs=*/ 15, /* dc=*/ 2, /* reset=*/ 5);  
U8G2 *u8g2_ptr;     

in setup:
 U8G2 *u8g2_ptr = &u8g2_obj;

and in main ino and files inside changed u8g2. to u8g2->

in main after compilation:
'u8g2' was not declared in this scope ( u8g2->firstPage(); )

Can i use this setup:
U8G2_SSD1322_NHD_256X64_F_4W_HW_SPI u8g2_obj(U8G2_R0, /* cs=*/ 15, /* dc=*/ 2, /* reset=*/ 5);  
and
U8G2_SSD1322_NHD_256X64_F_4W_HW_SPI u8g2(U8G2_R0, /* cs=/ 15, / dc=/ 2, / reset=/ 5);
 in the same time?

when i made this:
main ino ok, no errors, but in other files, after add extern U8G2 *u8g2_ptr; in global, the same..
'u8g2' was not declared in this scope

extern U8G2 *u8g2_ptr; do not link to setup of oled hardware?

olikraus, 2018-12-04:
> 'u8g2' was not declared in this scope ( u8g2->firstPage(); )

of course yo need to use u8g2_ptr in this case.
Or you use

U8G2 *u8g2;

in setup:
U8G2 *u8g2 = &u8g2_obj;

wchpikus, 2018-12-04:
ok so after test:

compilation ok, 
in global:
U8G2_SSD1322_NHD_256X64_F_4W_HW_SPI u8g2_obj(U8G2_R0, /* cs=*/ 15, /* dc=*/ 2, /* reset=*/ 5);
U8G2_SSD1322_NHD_256X64_F_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 15, /* dc=*/ 2, /* reset=*/ 5);
U8G2 *u8g2_ptr;

in setup:
U8G2 *u8g2 = &u8g2_obj;

functions looks u8g2->begin();

in other files:
extern U8G2 *u8g2;

library 2,24,3 crash.. after draw first bitmap...

>>>stack>>>
3fff45e0:  3fff2870 0000000b 40219828 40207e8e  
3fff45f0:  4020231a 000003e8 3fff2870 0000000b  
3fff4600:  3fff2870 3fff2352 3fff356c 4020c2c5  
3fff4610:  4020ae8c feefeffe 4021a90c 4021a8f4  
3fff4620:  feefeffe feefeffe feefeffe feefeffe  
3fff4630:  feefeffe feefeffe feefeffe 3fff362c  
3fff4640:  3fffdad0 00000000 3fff3624 4021986c  
3fff4650:  feefeffe feefeffe 3fff3640 40100710  
<<<stack<<<

Decoding stack results
0x40219828: esp_yield() at \Arduino15\packages\esp8266\hardware\esp8266\2.4.1\cores\esp8266\core_esp8266_main.cpp line 87
0x40207e8e: u8g2_SetBufferCurrTileRow at C:\Users\w.chmielewski\Documents\Arduino\libraries\U8g2\src\clib\u8g2_buffer.c line 105
0x4020231a: delay at \Arduino15\packages\esp8266\hardware\esp8266\2.4.1\cores\esp8266\core_esp8266_wiring.c line 54
0x4020c2c5: setup() at C:\Users\w.chmielewski\Documents\Arduino\libraries\U8g2\src/U8g2lib.h line 212
0x4020ae8c: FtpServ::handleNewClient(void*, AsyncClient*) at C:\Users\W77FE~1.CHM\AppData\Local\Temp\arduino_build_782096\sketch/ftpServ.h line 37
0x4021a90c: std::_Function_base::_Base_manager ::_M_manager(std::_Any_data&, std::_Any_data const&, std::_Manager_operation) at \arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2/functional line 1934
0x4021a8f4: std::_Function_handler ::_M_invoke(std::_Any_data const&, void*, AsyncClient*) at \arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2/functional line 2069
0x4021986c: loop_wrapper() at \Arduino15\packages\esp8266\hardware\esp8266\2.4.1\cores\esp8266\core_esp8266_main.cpp line 118

Downgrade u8g2 to 2.23.18 no help, so this solution is not work..

wchpikus, 2018-12-04:
Did you use interrupts in this driver or other interrupts can hang your code?

wchpikus, 2018-12-04:
ok i have it..
f..s....!@#$#@$
Do not use :

olikraus, 2018-12-04:
No interrupts, no dynamic memory

wchpikus, 2018-12-09:
sorry, for this time no success with new library..
what is differences between 2.3.18 vs 2.4.3?

olikraus, 2018-12-10:
Changes are here:
https://github.com/olikraus/u8g2/issues?q=is%3Aclosed+milestone%3A2.24

olikraus, 2018-12-10:
A detailed commit history probably is best viewed here:
https://github.com/olikraus/U8g2_Arduino/commits/master

wchpikus, 2018-12-12:
Could i send you on priv my project?
i give up with new library, i can not use them..

wchpikus, 2018-12-12:
btw..
i can not use

U8G2_SSD1322_NHD_256X64_F_4W_HW_SPI u8g2_obj(U8G2_R0, /* cs=*/ 15, /* dc=*/ 2, /* reset=*/ 5);
U8G2 *u8g2_ptr;

and in extern files:
U8G2 *u8g2 = &u8g2_obj;

i try use it in 2.23.18 and make crash, works only when i declare in all function used oled:
e.g.

void display_connection(){   
U8G2_SSD1322_NHD_256X64_F_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 15, /* dc=*/ 2, /* reset=*/ 5);       
u8g2.setContrast(backligth); ...

and in all function i have to add this..

olikraus, 2018-12-12:
> Could i send you on priv my project?

It would not make sense, I do not have time to debug your code and I also do not have your environment.

> i give up with new library, i can not use them..

I am sorry for this, but no problem. It is open source... maybe someone else can find the problem.

wchpikus, 2018-12-12:
i understand.. 
Maybe someone find what reset mcu, but i found that:

"All application code must be in either a task or a timer
All tasks and timers should complete in less than 2ms and must complete in less than 500ms or the watchdog timer will reset the MCU.
Reoccurring timers should not be scheduled more often than every 5ms."

maybe some function take a lot of time and >2ms?

olikraus, 2018-12-12:
> maybe some function take a lot of time and >2ms?

Sure, this can happen.

olikraus, 2018-12-12:
as mentioned above: I can not identify any major change which could explain you problem.

olikraus, 2018-12-20:
ToDo: test with my own ESP8266 board.

chenli2015, 2019-01-13:
Maybe spi‘s problem！This SPI bus connects to the flash also. Did you try i2c?

wchpikus, 2019-01-13:
I know..internal yes, this chip using 4M flash..
But i still using 2.23.18 and this is ok..
Do you have esp to test newest version?
I have spi (hardware) connected to oled.

wchpikus, 2019-01-13:
It can be lcd with ssd1322.
I can test with other lcd with newest library and older.

olikraus, 2019-02-09:
any new results on this? I did not find time for further debugging... so I move this to the next milestone...

olikraus, 2019-05-19:
Today I did several tests with my ESP8266 board (https://github.com/olikraus/u8g2/issues/53). Everything works without problems (Arduino 1.8.4 and ESP8266 2.5.1)

wchpikus, 2019-05-20:
Ok, now i see i have older esp toolchain..strange.
I update it, mayby here is problem.
Tnx for test )

closed 2019-11-20 by olikraus