olikraus/u8g2 #750: SPI clock speed for SH1106

semicolonTransistor, 2018-11-23:
I have experienced some performance issues when I ported my project to U8g2 for the amazing font and graphics support. My project is running on an STM32F103C8 and the display in question is a generic SH1106. After some probing around with my scope I found that U8G2 was only clocking the SPI bus at 2.25 MHz. I read that earlier this year in #551 that the maximum allowed speed for SH1106 was lowered from 8MHz to 4MHz which resulted in the next lower possible SPI speed on the STM32 being used, 2.25 MHz. After reading #551 I believe the screen issue experienced by a0908896067 was caused by not calling nextPage() when using the library in paged mode. I have been running the SPI bus at 36MHz before porting my library to U8g2 without any problems. I don't know why the datasheet only specifies 4MHz. And since SH1106 is a pretty common Oled chipset and only this year did someone raise an issue I think it is fair to assume that people were not having problems with SH1106 at the original clock speed of 8MHz. I tried to raise the clock speed by modifying the file mentioned in #551 but it made no difference to SPI clock rate. It is really hard to track down a single number in a set of source files. 
Here are my 2 suggestions:
1. Raise the clock speed of SH1106 back to 8MHz
2. Maybe add in a way to override the clock speed programmatically. Perhaps an additional parameter in the constructor.

Thanks for the powerful and lean library and keep up the grate work.

olikraus, 2018-11-23:
Your second suggestion is already implemented: #705

olikraus, 2018-11-23:
> Thanks, I see now. Sorry I did not investigate it thoroughly enough and missed the setBusFrequency method.

no problem

> Please consider my first suggestion as I think a lot of people may benefit from the extra performance.

Sure, but my point of view is this:
If I increase clock frequency by default, then there probably will be some users for which this speed will not work (due to long wires or other hw problems). The screen will stay blank and the users will complain here or in the Arduino forum. U8g2 will be made guilty for this malfunction and I will have a hard time for each of those cases to figure out the root cause of the malfunction.

On the other side, if I keep the clock speed slow, the display will at least work. If people are interessted, they will ask about performance improvements and I can provide a specific answers quickly.

So from support perspective it is much better to make a library reliable. But this also means it will be slow.
If there is something to learn as a library author it is this: Whatever you do, there is someone who complains: Some users request new features, others complain the size of the lib. Other users complain about the speed, again some user will not be able to see somehting.

That's the life of a library author ;-)

semicolonTransistor, 2018-11-24:
No problem, I see your point. I'll close this issue. Thanks for your time.

closed 2018-11-24 by semicolonTransistor