olikraus/u8g2 #752: Missing Constructor

SteveC00k, 2018-11-24:
SED1520 122X32
This does not exist except on the constructor list pages
https://github.com/olikraus/u8g2/wiki/u8g2setupcpp

olikraus, 2018-11-24:
Will sbn1661 work?

SteveC00k, 2018-11-25:
No the sbn1661 doesn't work as it causes the display to show everything mirrored. I have attempted to use it and even attempted to alter one of the headers as suggested by another to revers things but it does not work.

SteveC00k, 2018-11-25:
This is the display I have when attempting to use the sbn1661:
[image]

[image]

olikraus, 2018-11-25:
Regarding your initial post:

> SED1520 122X32
This does not exist except on the constructor list pages

What exactly does this mean? I see the constructor in the examples (at least the full buffer examples).

> No the sbn1661 doesn't work as it causes the display to show everything mirrored.

You display differs from the one which I have used for testing.
You could use the Mirrir option (https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#rotation): Use `U8G2_MIRROR` instead of `U8G2_R0`.

Will this solve your problem?

SteveC00k, 2018-11-25:
Found out that the hello world example I was using didn't have the SED1520. 
Found another had an example and I copied it. Used the mirror option. The text is oriented correctly but half of the text is chopped off.

SteveC00k, 2018-11-25:
Ok managed to get the display working. Need to learn more about the lib. Its huge.

closed 2018-12-20 by olikraus