olikraus/u8g2 #753: update display timeout > ignore update display?

StefanMeGit, 2018-11-24:
Is there any change to check that the display is working while updating it? 

I want to handle the case if the SSD1306 breaks or disconnect while running u8g2.clearBuffer(); 8g2.sendBuffer(); or u8g2.firstPage(); u8g2.nextPage()

My program simply stops when I disconnect the display which is not good...

olikraus, 2018-11-24:
What kind of communication do you use? I2C?
Which Arduino Board?

I assume, the problem will not happen with SPI. It should also not happen with SW-I2C. Actually the problem is not a problem of U8g2 but a problem of the Arduino Wire library on AVR systems (Arduino Uno).
The problem might be solved by using proper pullup resistors. Well all in all I can not say much without having more information from your side.

StefanMeGit, 2018-11-24:
Oh sorry... I run it on an Samd21 (Adafruit Feather M0) on HW I2C (4 wire connection).

`U8G2_SSD1306_128X64_NONAME_F_HW_I2C` 

PullUp resistor to what?

olikraus, 2018-11-25:
Just to clarify: I2C, right? (not SPI, because 4-wire connection usually referes to SPI)

Pull-up resistors for I2C refer to 2K to 10K resistors which connect the clock and data line towards power supply (3.3V in your case).
See here: https://www.i-programmer.info/programming/hardware/9124-exploring-edison-i2c-bus.html

StefanMeGit, 2018-11-25:
Ok, but I use 4 wires for I2C: 
VCC 
GROUND 
SDA 
SCL
Didn't u need more wires for SPI? 
VCC
GROUND
MOSI  
CLK 
DC  
CS  
RESET 

Ok, this pull-ups I am missing. The should be close to the SDA SCL pins on the Ardunios u guess...

olikraus, 2018-11-25:
> Ok thanks, so the pull-up restores should solve this problem?

It is more like an idea. I know that the hardware I2C subsystem of the AVR (ATMega uC) will halt the uC if there is no I2C device and no pullups. Not sure about the MKR uC.

closed 2018-12-20 by olikraus