olikraus/u8g2 #756: Skewed Text winstar

goodsireric, 2018-11-28:
Using a Winstar #738 and the 128x32 library almost works. When using horizontal mode, the text appears fine for 4 rows, then on the next 4 rows is shifts to the right by 4 pixels. 
When using paging mode, the text appears offset way at the top of the screen and the remaining screen is random. (attached images of the skewing and offset)

Can this be fixed by tweaking the title settings? When adjusting the title_width and title_height, the text can become more jumbled or just become completely random. 

Thank you so much for your work on this library, it has been extremely helpful. 
When I get this display working, would you like me to issue a pull request to add this winstar display?

[Winstar128x32_Display_Example.zip](https://github.com/olikraus/u8g2/files/2626982/Winstar128x32_Display_Example.zip)

goodsireric, 2018-11-29:
Managed to get the display working by removing all address offsetting commands "u8x8_cad_SendCmd" inside U8X8_MSG_DISPLAY_DRAW_TILE. I have reached out to Winstar to find out if there is an alternate method of setting the x,y coordinates.

olikraus, 2018-11-29:
> When I get this display working, would you like me to issue a pull request to add this winstar display?

sure, but we should probably build everything into a new device. Which u8x8 device did you modify?

olikraus, 2018-12-20:
@goodsireric: Can you post the modified code?

olikraus, 2019-01-02:
Can you paste the modified part of the code here? 
Thanks

goodsireric, 2019-01-05:
Still having an issue unfortunately, see attached image. It's shifting to the right by a three rows, I think that the commands aren't being sent/accepted properly before the screen data is sent. This is via SPI.

[image]

olikraus, 2019-01-05:
Which constructor did you use?

goodsireric, 2019-01-07:
Finally solved the last problem by setting the x-offset to 125. Setting flipmode_offset_x to 125 makes 180 degree flip work as well. Turned out to be a two-layer problem, as nrf52 SDK has a problem with sending single characters over SPI without a workaround. 

The WEO012832DWPP3N00000 uses the SSD1306 controller, and is a newer variant on that older display you found. 
[Winstar WEO012832DWPP3N00000 Data Sheet.pdf](https://github.com/olikraus/u8g2/files/2733817/Winstar.WEO012832DWPP3N00000.Data.Sheet.pdf)

I have attached the constructor file, feel free to use it in any way you want. I really appreciate all of your support and excellent work documenting your library.  

[u8x8_d_ssd1306_128x32 _Winstar.zip](https://github.com/olikraus/u8g2/files/2733830/u8x8_d_ssd1306_128x32._Winstar.zip)

olikraus, 2019-01-07:
Thanks!
Will have a look and update the code.

olikraus, 2019-02-09:
Created a U8G2_SSD1306_128X32_WINSTAR constructor.

closed 2019-02-09 by olikraus