olikraus/u8g2 #757: setBusClock not changing clock rate

versamodule, 2018-11-30:
I am using U8G2_SSD1322_NHD_128X64_F_4W_SW_SPI u8g2, on a ESP32. 
Out of the box works just fine. In the reference manual i see there is setBusClock. 
I am calling u8g2.setBusClock(2000000);     after u8g2.begin();

The problem is that no matter what freq, I use in there, it does not change the rate the OLED updates. 
I have verified this with the oscilloscope, and I initializing it in the wrong place ?

olikraus, 2018-11-30:
U8g2 just calls the Arduino HAL for this (SPI.h). Maybe there's an issue in the Arduino ESP32 software. can you test SPI.h whether the clock setting works?

versamodule, 2018-11-30:
Yes, I started writing my own driver for my OLED, before I found yours, and said why re-invent it. Mine works when using the following, and when I change SPI_CLOCK_DIV8 to say SPI_CLOCK_DIV128 it does change the clock speed accordingly. 

  pinMode(CS_PIN, OUTPUT);
  digitalWrite(CS_PIN,HIGH);
  SPI.begin( CLK_PIN,  MISO_PIN,  MOSI_PIN,  CS_PIN);
  SPI.setDataMode(SPI_MODE0);
  SPI.setClockDivider(SPI_CLOCK_DIV8);

olikraus, 2018-11-30:
Actually SPI.setClockDivider is outdated and U8g2 will use SPI.beginTransaction to set the clock speed.
Can you test whether SPI.beginTransaction allows you to modify the clock speed?

Reference: 
https://github.com/olikraus/u8g2/blob/master/cppsrc/U8x8lib.cpp#L733
https://www.arduino.cc/en/Reference/SPIbeginTransaction

versamodule, 2018-11-30:
I placed a bunch of printf("1"); , printf("2");, printf("3"); and so on throughout U8x8lib.cpp to see whats getting called. Its my way of debugging ;) 

```
#if ARDUINO >= 10600
      SPI.beginTransaction(SPISettings(u8x8->bus_clock, MSBFIRST, internal_spi_mode));
      printf("2\r\n");
#else
```

Never gets called. What does get called is 

```
#ifdef U8X8_USE_PINS

/*
  use U8X8_PIN_NONE as value for "reset", if there is no reset line
*/

void u8x8_SetPin_4Wire_SW_SPI(u8x8_t *u8x8, uint8_t clock, uint8_t data, uint8_t cs, uint8_t dc, uint8_t reset)
{
    printf("6\r\n");
  u8x8_SetPin(u8x8, U8X8_PIN_SPI_CLOCK, clock);
  u8x8_SetPin(u8x8, U8X8_PIN_SPI_DATA, data);
  u8x8_SetPin(u8x8, U8X8_PIN_CS, cs);
  u8x8_SetPin(u8x8, U8X8_PIN_DC, dc);
  u8x8_SetPin(u8x8, U8X8_PIN_RESET, reset);
}
```

versamodule, 2018-11-30:
I think i may have found the issue. 
```
#else
  /* fallback */
  uint8_t u8x8_byte_arduino_4wire_sw_spi(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)
  {
    printf("9\r\n");
    return u8x8_byte_4wire_sw_spi(u8x8, msg,arg_int, arg_ptr);
  }
  
#endif
```

This was getting called repeatedly, so looking at u8x8_byte_4wire_sw_spi, in u8x8_byte.c  it is bit banging the spi. and not using the actual SPI driver. Should there be something else i need to define to prevent that ?

olikraus, 2018-11-30:
oh.... i should have read more carefully your posts:

> I am using U8G2_SSD1322_NHD_128X64_F_4W_SW_SPI u8g2, on a ESP32.

This constructor will obviously call the bitbanging procedure (SW_SPI). Better use
U8G2_SSD1322_NHD_128X64_F_4W_HW_SPI  ;-)

versamodule, 2018-12-01:
Your absolutely right!  I was so focused on a constructor to allow me to set my own pins that I never read the entire part of it. I only got as far as "U8G2_SSD1322_NHD_128X64". 
I will give it a shot, and I am sure it will work like its supposed to. 
Just out of curiosity, why doesn't the "U8G2_SSD1322_NHD_128X64_F_4W_HW_SPI" have setable pins like the "U8G2_SSD1322_NHD_128X64_F_4W_SW_SPI" does ? 

This would be handy to have on a ESP32.

olikraus, 2018-12-01:
> Just out of curiosity, why doesn't the "U8G2_SSD1322_NHD_128X64_F_4W_HW_SPI" have setable pins like the "U8G2_SSD1322_NHD_128X64_F_4W_SW_SPI" does ?

It has, but the argument sequence is different. It is like this:
Most uC have only fixed pins for the HW SPI subsystem. For example on AVR systems it does not make sense to add pin numbers for HW SPI, because these pins are fixed and can not be changed by any library. However, some uC include a switch matrix to map the HW SPI pins on any GPIO pin. In such a case the U8g2 constructor offers optional pins to carry out this mapping.

I think this mapping might be supported for the ESP32, but I do not really remember the status.

versamodule, 2018-12-01:
> I think this mapping might be supported for the ESP32, but I do not really remember the status.
It is. I just assumed that since I saw ESP32 in the code it seemed odd that it didn't 

Anyways, Thanks again for all the awesome help.  You can close this ticket.

closed 2018-12-01 by versamodule

olikraus, 2018-12-01:
:)