olikraus/u8g2 #758: Partial refresh after deep sleep

lesept777, 2018-11-30:
Here is what I want to do. A simple watch displaying the time as HH:MM each minute on an oled display, and put the esp32 in deep sleep during the interval.
Additionally, I want to update the display by changing only the digits that changed. For example, at time 05:17, I just want to change the 7, all the other numbers should remain on the display.

But when the esp32 wakes from deep sleep, I have to declare and init the display, which erases it entirely. So this implies that I need to display the time completely.

Is there a solution to prevent the display from erasing?

I tried to bypass the display.init() instruction when it's not the first boot, but the display freezes. I've tried to to replace display.begin(); by display.initDisplay(); but the display erases and nothing appears on it.

The init function is
`bool InitSensorsDisplays () {
  // Sensors
  dht.setup(DHT_PIN, DHTesp::DHT22);
  bool AcceleroSuccess = InitAccelero ();
  // Displays
  display1.setI2CAddress(0x3C * 2);
  display2.setI2CAddress(0x3D * 2);
  display1.begin();
  display2.begin();
  display1.setFont(u8g2_font_helvB18_tr);
  display2.setFont(u8g2_font_helvR08_tr);
  return AcceleroSuccess;
}
`
Thanks for any advice or solution

lesept777, 2018-11-30:
The final apllication will use e(paper displays, so if there is a soution for e-paper and not for oled, it's ok for me...

olikraus, 2018-11-30:
Yes, this is a problem if you have a reset line and the reset line is controlled by u8g2.
I came across this problem also in my own projects.

The init starts here and receives a flag whether to do a power on reset or a soft start of the display:
https://github.com/olikraus/u8g2/blob/master/sys/arm/stm32l031x6/clock/main.c#L437

The actual magic happens here:
https://github.com/olikraus/u8g2/blob/master/sys/sdl/clock/guifn.c#L379

This code mimics the the normal u8g2 startup without execution of the reset (and without clearing the display).  For power on reset, the normal procedure continues:
https://github.com/olikraus/u8g2/blob/master/sys/sdl/clock/guifn.c#L393

Hope this helps....

lesept777, 2018-11-30:
I'll have look, thanks. What do you call 'reset line' ? I wake up the esp32 either by EXT0 interrupt or by timer

lesept777, 2018-11-30:
Question: I have 2 displays (called display1 and display2). Will the line
`gui_LoadData();`
handle both of them ?

lesept777, 2018-11-30:
Ouch !

> Arduino : 1.8.2 (Windows 10), Carte : "LOLIN D32, Default, 80MHz, 921600, None"
> 
> In file included from C:\Users\Chuwi\Documents\Arduino\_METEO\Meteo_ESP32_DeepSleep2\Meteo_ESP32_DeepSleep2.ino:5:0:
> 
> sketch\Accelero.h: In function 'bool InitSensorsDisplays()':
> 
> Accelero.h:36: error: 'gui_LoadData' was not declared in this scope
> 
>      gui_LoadData();
> 
>                   ^
> 
> In file included from C:\Users\Chuwi\Documents\Arduino\libraries\U8g2\src/U8x8lib.h:45:0,
> 
>                  from C:\Users\Chuwi\Documents\Arduino\libraries\U8g2\src/U8g2lib.h:50,
> 
>                  from sketch\Variables.h:107,
> 
>                  from C:\Users\Chuwi\Documents\Arduino\_METEO\Meteo_ESP32_DeepSleep2\Meteo_ESP32_DeepSleep2.ino:1:
> 
> C:\Users\Chuwi\Documents\Arduino\libraries\U8g2\src/clib/u8g2.h:347:44: error: invalid cast from type 'U8G2_SH1106_128X64_NONAME_F_SW_I2C' to type 'u8x8_t* {aka u8x8_struct*}'
> 
>  #define u8g2_GetU8x8(u8g2) ((u8x8_t *)(u8g2))
> 
>                                             ^
> 
> C:\Users\Chuwi\Documents\Arduino\libraries\U8g2\src/clib/u8x8.h:725:32: note: in definition of macro 'u8x8_gpio_Init'
> 
>  #define u8x8_gpio_Init(u8x8) ((u8x8)->gpio_and_delay_cb((u8x8), U8X8_MSG_GPIO_AND_DELAY_INIT, 0, NULL ))
> 
>                                 ^
> 
> sketch\Accelero.h:37:20: note: in expansion of macro 'u8g2_GetU8x8'
> 
>      u8x8_gpio_Init(u8g2_GetU8x8(display1));
> 
>                     ^
> 
> C:\Users\Chuwi\Documents\Arduino\libraries\U8g2\src/clib/u8g2.h:347:44: error: invalid cast from type 'U8G2_SH1106_128X64_NONAME_F_SW_I2C' to type 'u8x8_t* {aka u8x8_struct*}'
> 
>  #define u8g2_GetU8x8(u8g2) ((u8x8_t *)(u8g2))
> 
>                                             ^
> 
> C:\Users\Chuwi\Documents\Arduino\libraries\U8g2\src/clib/u8x8.h:725:58: note: in definition of macro 'u8x8_gpio_Init'
> 
>  #define u8x8_gpio_Init(u8x8) ((u8x8)->gpio_and_delay_cb((u8x8), U8X8_MSG_GPIO_AND_DELAY_INIT, 0, NULL ))
> 
>                                                           ^
> 
> sketch\Accelero.h:37:20: note: in expansion of macro 'u8g2_GetU8x8'
> 
>      u8x8_gpio_Init(u8g2_GetU8x8(display1));
> 
>                     ^
> 
> C:\Users\Chuwi\Documents\Arduino\libraries\U8g2\src/clib/u8g2.h:347:44: error: invalid cast from type 'U8G2_SH1106_128X64_NONAME_F_SW_I2C' to type 'u8x8_t* {aka u8x8_struct*}'
> 
>  #define u8g2_GetU8x8(u8g2) ((u8x8_t *)(u8g2))
> 
>                                             ^
> 
> C:\Users\Chuwi\Documents\Arduino\libraries\U8g2\src/clib/u8x8.h:584:31: note: in definition of macro 'u8x8_cad_Init'
> 
>  #define u8x8_cad_Init(u8x8) ((u8x8)->cad_cb((u8x8), U8X8_MSG_CAD_INIT, 0, NULL ))
> 
>                                ^
> 
[... 198 lines cut ...]
> C:\Users\Chuwi\Documents\Arduino\libraries\U8g2\src/clib/u8g2.h:364:62: note: in expansion of macro 'u8g2_GetU8x8'
> 
>  #define u8g2_SetPowerSave(u8g2, is_enable) u8x8_SetPowerSave(u8g2_GetU8x8(u8g2), (is_enable))
> 
>                                                               ^
> 
> sketch\Accelero.h:47:5: note: in expansion of macro 'u8g2_SetPowerSave'
> 
>      u8g2_SetPowerSave(display2, 0);
> 
>      ^
> 
> Plusieurs bibliothèque trouvées pour "WiFi.h"
> Utilisé : C:\Users\Chuwi\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0\libraries\WiFi
> Non utilisé : C:\Program Files (x86)\Arduino\libraries\WiFi
> Plusieurs bibliothèque trouvées pour "MPU6050.h"
> Utilisé : C:\Users\Chuwi\Documents\Arduino\libraries\MPU6050
> Non utilisé : C:\Users\Chuwi\Documents\Arduino\libraries\Arduino_MPU6050
> exit status 1
> 'gui_LoadData' was not declared in this scope
> 
> Ce rapport pourrait être plus détaillé avec
> l'option "Afficher les résultats détaillés de la compilation"
> activée dans Fichier -> Préférences.
>

olikraus, 2018-11-30:
"gui_LoadData" is part of my own project. You can ignore this call for your problem.

Reset line is the reset input of the display. It is usually only used for SPI based displays.

lesept777, 2018-12-01:
Thanks, I removed that line. However I have another problem, with the display constructor. Here is the code
```
  if (bootCount == 0) {
    // Displays
    display1.setI2CAddress(0x3C * 2);
    display2.setI2CAddress(0x3D * 2);
    //    display1.begin();
    //    display2.begin();
    u8x8_gpio_Init(u8g2_GetU8x8(display1));
    u8x8_cad_Init(u8g2_GetU8x8(display1));
    u8x8_gpio_SetReset(u8g2_GetU8x8(display1), 1);
    u8x8_gpio_Init(u8g2_GetU8x8(display2));
    u8x8_cad_Init(u8g2_GetU8x8(display2));
    u8x8_gpio_SetReset(u8g2_GetU8x8(display2), 1);
  } else {
    u8g2_InitDisplay(display1);
    u8g2_SetPowerSave(display1, 0);
    u8g2_InitDisplay(display2);
    u8g2_SetPowerSave(display2, 0);
  }
  display1.setFont(u8g2_font_helvB18_tr);
  display2.setFont(u8g2_font_helvR08_tr);
```
where display1 and display2 are defined by
```
// Displays
#include <U8g2lib.h>
U8G2_SH1106_128X64_NONAME_F_SW_I2C display1(U8G2_R0, SCL1, SDA1);
U8G2_SH1106_128X64_NONAME_F_SW_I2C display2(U8G2_R0, SCL1, SDA1);
```
It seems incorrect. What are the argument to the functions u8g2_InitDisplay, u8g2_InitDisplay etc ?

olikraus, 2018-12-01:
display1.getU8x8() --> return a pointer to the u8x8 structure
display2.getU8g2() --> return a pointer to the u8g2 structure

lesept777, 2018-12-02:
Thanks, trying to understand your answer. Is this correct?

```
  if (bootCount == 0) {
    // First init of the displays
    display1.setI2CAddress(0x3C * 2);
    display2.setI2CAddress(0x3D * 2);
    //    display1.begin();
    //    display2.begin();
    display1.initDisplay();
    display1.setPowerSave(0);
    display2.initDisplay();
    display2.setPowerSave(0);
  } else {
    // Init displays after wake up
    // https://github.com/olikraus/u8g2/blob/master/sys/sdl/clock/guifn.c#L379
    u8x8_gpio_Init(display1.getU8g2());
    u8x8_cad_Init(display1.getU8g2());
    u8x8_gpio_SetReset(display1.getU8g2(), 1);
    u8x8_gpio_Init(display2.getU8g2());
    u8x8_cad_Init(display2.getU8g2());
    u8x8_gpio_SetReset(display2.getU8g2(), 1);
  }
```
Why se `u8x8_gpio_Init` if I use U8g2lib.h ?

lesept777, 2018-12-02:
I tried this:

```
  if (bootCount == 0) {
    // First init of the displays
    display1.setI2CAddress(0x3C * 2);
    display2.setI2CAddress(0x3D * 2);
    display1.begin();
    display2.begin();
    //    display1.initDisplay();
    //    display1.setPowerSave(0);
    //    display2.initDisplay();
    //    display2.setPowerSave(0);
  } else {
    // Init displays after wake up
    // https://github.com/olikraus/u8g2/blob/master/sys/sdl/clock/guifn.c#L379
    u8x8_gpio_Init(display1.getU8x8());
    u8x8_gpio_Init(display2.getU8x8());
    u8x8_cad_Init(display1.getU8x8());
    u8x8_cad_Init(display2.getU8x8());
    u8x8_gpio_SetReset(display1.getU8x8(), 1);
    u8x8_gpio_SetReset(display2.getU8x8(), 1);
  }
```
and I get a strange result. 
Remember that I use 2 OLED displays, called display1 and display2. On display1, I have the time HH:MM, and on display2 some temperature and humidity data, from a DHT22 sensor.
The ESP32 goes into deep sleep until the start of the next minute (i.e MM becomes MM+1), at that time it wakes up and displays the updated time and new sensor data. 

At first boot, no problem; display1 gives HH:MM and display2 sensor data. But when comes the next wakeup, display2 provides the new data but display1 seems to first display what was on display2 and then changes the MM to MM+1. So I get sensor data mixed with minutes.

There seems to be a buffer transfer from display2 to display1 before calling the time display function, thus this must happen somewhere in the code lines up here.
Have you ever experienced that?

lesept777, 2018-12-02:
More precisely, display2 does not update, it keeps the old sensor data. The new sensor data is written on display1 and then the new minutes number is written above.

lesept777, 2018-12-04:
I tried to move the setI2CAddress instructions out of the if block, thinking that the addresses were lost during deep sleep, but it doesn't change, I still get the mix of buffers.

olikraus, 2018-12-04:
> Why use u8x8_gpio_Init if I use U8g2lib.h ?

U8g2 is based on U8x8. All low level com is handled via U8x8.

> There seems to be a buffer transfer from display2 to display1 

In fact they share the same buffer

```
    display1.setI2CAddress(0x3C * 2);
    display2.setI2CAddress(0x3D * 2);
```

This has to be executed in all cases not only after first boot. I assume RAM is cleared, right?

lesept777, 2018-12-04:
Yes all RAM is cleared in deep sleep of the ESP32, except for the RTC RAM where you can save whatever you want to survive after wakeup.
I tried to do what you said but I still get the same result:
```
  display1.setI2CAddress(0x3C * 2);
  display2.setI2CAddress(0x3D * 2); // those 2 lines are now executed at every wakeup
  if (bootCount == 0) {
    // First init of the displays
    display1.begin();
    display2.begin();
  } else {
    // Init displays after wake up
    u8x8_gpio_Init(display1.getU8x8());
    u8x8_gpio_Init(display2.getU8x8());
    u8x8_cad_Init(display1.getU8x8());
    u8x8_cad_Init(display2.getU8x8());
    u8x8_gpio_SetReset(display1.getU8x8(), 1);
    u8x8_gpio_SetReset(display2.getU8x8(), 1);
  }
```

olikraus, 2018-12-04:
How does your code continue from here? Do you refresh the display content?
If so, how do you do this?

lesept777, 2018-12-04:
Sorry, I don't have my code here, I'll answer this evening (it's noon here)

lesept777, 2018-12-04:
Do you mean that when passing (for example) from 12:34 to 12:35, I have to print the whole 12:35 and not only the 5? Will this make the display flicker?

olikraus, 2018-12-04:
> I have to print the whole 12:35 and not only the 5? 

yes

> Will this make the display flicker?

no (because clearBuffer will clear the uC RAM only, it will not affect the display. The display update happens only with sendBuffer).

lesept777, 2018-12-04:
Thanks, I'll try this evening.

lesept777, 2018-12-04:
I tried and I still have a mix of the buffer(s) : the text on display1 appears also on display2.
[image]
The left display should only display the time, and the one on the right only temp & humi data.

The init:
```
  display1.setI2CAddress(0x3C * 2);
  display2.setI2CAddress(0x3D * 2);
  if (bootCount == 0) {
    // First init of the displays
    display1.begin();
    display2.begin();
    if (DEBUG) Serial << "Begin displays" << endl;
  } else {
    // Init displays after wake up
    // https://github.com/olikraus/u8g2/blob/master/sys/sdl/clock/guifn.c#L379
    u8x8_gpio_Init(display1.getU8x8());
    u8x8_gpio_Init(display2.getU8x8());
    u8x8_cad_Init(display1.getU8x8());
    u8x8_cad_Init(display2.getU8x8());
    u8x8_gpio_SetReset(display1.getU8x8(), 1);
    u8x8_gpio_SetReset(display2.getU8x8(), 1);
  }
```
The sensors data display (called first):
```
void DisplayTH2H (float temperature, float humidite) {
//  display2.clearBuffer();
  display2.setFont(u8g2_font_helvR10_tr);
  CenterText("Interieur :", 10, 2);
  display2.setFont(u8g2_font_fur14_tf);
  char oledTxt[12] = "";
  sprintf(oledTxt, "Temp %.1f\xb0C", temperature);
  display2.setCursor(1, 35);
  display2.print(oledTxt);
  //  CenterText (oledTxt, 25, 2);
  display2.print("\xb0"); // Symbole degrés
  display2.print("C");
  char oledTxt2[11] = "";
  sprintf(oledTxt2, "Humi %.1f", humidite);
  display2.setCursor(4, 60);
  display2.print(oledTxt2);
  //  CenterText (oledTxt2, 60, 2);
  display2.print("%");
  display2.sendBuffer();
}
```
and the time (called after):
```
void DisplayTime () {
  char currentTime[6] = "xx:xx";
  time_t now;
  struct tm * timeinfo;
  time (&now);
  timeinfo = localtime (&now);
  currentTime[0] = (timeinfo->tm_hour + DST) % 24 / 10 + '0';
  currentTime[1] = (timeinfo->tm_hour + DST) % 24 % 10 + '0';
  currentTime[3] = timeinfo->tm_min / 10 + '0';
  currentTime[4] = timeinfo->tm_min % 10 + '0';
  display1.setFont(u8g2_font_inr30_mn);
  byte y = 45;
  CenterText (currentTime, y, 1);
  display1.sendBuffer();
}
```
With the centerText function:
```
void CenterText (char * Text, int y, byte Ndisplay) {
  if (Ndisplay == 2) {
    int x = (display2.getDisplayWidth() - display2.getStrWidth(Text)) / 2;
    if (x < 0) x = 0;
    display2.setCursor(x, y);
    display2.print(Text);
  } else {
    int x = (display1.getDisplayWidth() - display1.getStrWidth(Text)) / 2;
    if (x < 0) x = 0;
    display1.setCursor(x, y);
    display1.print(Text);
  }
}
```

lesept777, 2018-12-06:
I added display1.clearDisplay(); in the time display function, before the call to CenterText. This erases the background but adds some flicker...

olikraus, 2018-12-06:
>  but adds some flicker...

Yes, you should use display1.clearBuffer() instead

olikraus, 2018-12-06:
clearBuffer: Clear the uC RAM only
clearDisplay: Clear the uC RAM and send the empty RAM to the display

lesept777, 2018-12-06:
Perfect now it works like a charm. Thanks !

closed 2018-12-06 by lesept777

olikraus, 2018-12-06:
:)