olikraus/u8g2 #760: screen corruption or flipped characters

gdanov, 2018-12-05:
I'm using Heltec esp32 LoRa board with the sketch bellow. All works well except that after few minutes the characters get flipped in weird way. [this is picture, the orientation of the picture is correct](https://photos.app.goo.gl/Rsdwn2tiysLpi4MfA)

The only hint I see in the console is that after short time (maybe 1 min) I get this:

> [E][esp32-hal-i2c.c:1146] i2cCheckLineState(): Bus Invalid State, TwoWire() Can't init

Here is the sketch
```
//////////////////////////////////////////////////////////////////////////////////////////////////////////////
//                                                                                                          //
//                                     Heltec WiFi Kit 32 NTP Clock                                         //
//                                                                                                          //
//////////////////////////////////////////////////////////////////////////////////////////////////////////////

//
// Notes:
//
//  "Heltec WiFi Kit 32 NTP Clock" is a ntp initialized date and time clock.  The device connects to the
// an ntp time server via wifi and a udp port, obtains the ntp time from the server, adjusts then writes
// the time to the ESP32 rtc (real time clock), and displays the date and time on the built in OLED display.
//
//  Upon startup, the code initializes the serial port, wifi, graphics and udp port.  The serial port is
// used only during initialization to display when the wifi is connected and when the ntp time has been
// received from the ntp server.  Wifi is used to communicate with the ntp server.  The graphics is used
// to display the initialization and operational displays on the built in OLED.  Finally, the udp port
// receives the ntp time from the ntp server.
//
//  The main loop performs two major functions; obtains the time from the ntp server and to update the oled.
// In this example, the time is obtained from the ntp server only once, and upon receipt, is adjusted for
// time zone then written into the ESP32 rtc (real time clock).  The OLED is updated once per pass, and there
// is a 200ms delay in the main loop so the OLED is updated 5 times a second.
//
//  Before compiling and downloading the code, adjust the following settings:
//
//  1) TIME_ZONE  - currently set to -6 for Oklahoma (my home state), adjust to your timezone.
//  2) chPassword - currently set to "YourWifiPassword", adjust to your wifi password.
//  3) chSSID     - currently set to "YourWifiSsid", adjust to your wifi ssid.
//

//////////////////////////////////////////////////////////////////////////////////////////////////////////////
//
// Includes.
//
//////////////////////////////////////////////////////////////////////////////////////////////////////////////

#include                              <time.h>                              // for time calculations
#include                              <WiFi.h>                              // for wifi
#include                              <WiFiUdp.h>                           // for udp via wifi
#include                              <U8g2lib.h>                           // see https://github.com/olikraus/u8g2/wiki/u8g2reference

//////////////////////////////////////////////////////////////////////////////////////////////////////////////
//
// Constants.
//
//////////////////////////////////////////////////////////////////////////////////////////////////////////////

#define   FONT_ONE_HEIGHT               8                                   // font one height in pixels
#define   FONT_TWO_HEIGHT               20                                  // font two height in pixels
#define   NTP_DELAY_COUNT               20                                  // delay count for ntp update
#define   NTP_PACKET_LENGTH             48                                  // ntp packet length
#define   TIME_ZONE                     (+1)                                // offset from utc
#define   UDP_PORT                      4000                                // UDP listen port

//////////////////////////////////////////////////////////////////////////////////////////////////////////////
//
// Variables.
//
//////////////////////////////////////////////////////////////////////////////////////////////////////////////

char      chBuffer[128];                                                    // general purpose character buffer
[... 228 lines cut ...]
      
      struct tm * tmPointer = localtime(& tvTimeValue.tv_sec);
    
      // Display the date.
  
      strftime(chBuffer, sizeof(chBuffer), "%a, %d %b %Y",  tmPointer);
      u8g2.setFont(u8g2_font_6x10_tr);
      u8g2.drawStr(64 - (u8g2.getStrWidth(chBuffer) / 2), 0, chBuffer);
  
      // Display the time.
      
      strftime(chBuffer, sizeof(chBuffer), "%H:%M:%S",  tmPointer);
      u8g2.setFont(u8g2_font_fur20_tn);
      u8g2.drawStr(10, 63 - FONT_TWO_HEIGHT, chBuffer);
  
      // Send the display buffer to the OLED
      
      u8g2.sendBuffer();
  }
    
  // Give up some time.

  delay(300);
}
```

olikraus, 2018-12-05:
hmmm... if you would call u8g2.begin() to refresh the display config ever 10th time when you enter loop(), will it help?
If yes, then probably the wlan signal desturbs the display memory.

gdanov, 2018-12-05:
thanks, set it up to reset the display every 10 secs. clears the error logs, guess it will help with the corruption too. But please note the WiFi is off after the initial set-up and time acquisition.

olikraus, 2018-12-05:
WLAN Impact was just an idea

gdanov, 2018-12-07:
the unit is very stable when driven from batterypack via usb. I don't see this problem. However if it's plugged in the computer the problem appears. Wild guess — the usb-uart adapter is guilty as I also get nasty side effects on the computer side.

closed 2018-12-07 by gdanov

gdanov, 2018-12-07:
> some usb ports from PCs are not capable of handling the current requirements of an ESP32 properly, effectively browning out the hardware causing the issues you're describing, I suspect its more likely the PC's USB rather than the usb->uart adaptor

any hints how to mitigate this? I guess that would cause troubles also while flashing the board?

JamesGKent, 2018-12-07:
well your options are to use another USB port if you pc has a higher power port available (not as common as you'd think) or to use a powered USB hub between your pc and the esp32, I've had issues with esp32 with anything other than 1A or greater rated ports, and a lot of ports on PCs are only rated for 500mA.

and yes it could cause trouble while flashing the board, particularly if you also have power hungry peripherals attached (e.g. screen backlight)

gdanov, 2018-12-07:
thanks a lot, that was really helpful