olikraus/u8g2 #764: Support to export LCD buffer to image

vslinuxdotnet, 2018-12-11:
Hello,

Is possible to export the data in LCD buffer to a bitmap image?

Thank you

olikraus, 2018-12-11:
> Is possible to export the data in LCD buffer to a bitmap image?

If you really mean "LCD buffer" then the answer is "no, not with u8g2".
If your question is:

> Is possible to export the u8g2 display data to a bitmap image?

Then the answer is: It depends...

For example, if you the `_F_` type of constructor, then you can just get the pointer to the data (https://github.com/olikraus/u8g2/wiki/u8g2reference#getbufferptr). Then convert and store the bitmap data somewhere else.

If you want to save a picture which was rendered with u8x8 or with any other type of constructor, then maybe the example code here may work: https://github.com/olikraus/u8g2/tree/master/sys/bitmap

But this is not documented and may not work so easily in an embedded controller.

vslinuxdotnet, 2018-12-11:
OK thank you for the info, what I want is getting a copy of the display data and save it in a picture.
About the controller, Is the ESP8266 and I will write the picture in the SPI flash.

olikraus, 2018-12-11:
>  OK thank you for the info, what I want is getting a copy of the display data and save it in a picture.

Then "getBufferPtr" together with a F-constructor will do the job. The memory structure is described here: https://github.com/olikraus/u8g2/wiki/u8g2reference#memory-structure-for-controller-with-u8x8-support

Some displays (for which u8x8 API is also not available) have a different memory layout.

closed 2018-12-20 by olikraus