olikraus/u8g2 #767: Dear Olikraus,

olikraus, 2018-12-20:
is it possible to use the second SPI interface for the ST7920? Combined with the mySensors library I can't use the first SPI. The u8g2lib with the STM32F103C8T6 and first hardware SPI without mySensor library works very good, the software spi doesn't work for me I don't know why.

Best regards

Elmar Faber

_Originally posted by @elmarfaber in https://github.com/olikraus/u8g2/issues/244#issuecomment-448961044_

olikraus, 2018-12-20:
yes, can be done, but this is probably a bigger effort. I would need some time for this.

Todo:
Add new interface for ST7920 in codebuild.c:1754

elmarfaber, 2018-12-20:
Hello,

it would be nice if you can do this, thank you very much 

Best regards

olikraus, 2018-12-21:
The new constructor includes 2ND: `U8G2_ST7920_128X64_1_2ND_HW_SPI` and is available for 1, 2 and F constructors.

All new constructors are listed here:
https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#st7920-128x64

olikraus, 2018-12-21:
I have created beta release 2.25.2

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

elmarfaber, 2018-12-22:
Hello,
first, I want to thank you for the great help. I try the beta release under  PlatformIO (Visual Studio Code), platform = ststm32, board = genericSTM32F103C8 and the framework = arduino. The board is a bluepill. With SPI1 it works fine but with SPI2 (PB13 - SCK, PB15 - MOSI) it doesn't work.

Best regards

olikraus, 2018-12-22:
which constructor did you use? How does it look like?

elmarfaber, 2018-12-22:
#define displ_cs  PA4
U8G2_ST7920_128X64_F_HW_SPI u8g2(U8G2_R0, displ_cs);

olikraus, 2018-12-23:
From my perspective it is difficult why this doesn't work. There could be several reasons.

One important point is this: Your board files must define SPI_INTERFACES_COUNT (see here for example the mkrzero: https://github.com/arduino/ArduinoCore-samd/blob/master/variants/mkrzero/variant.h#L91).
U8g2 will check for this flag and will create a proper 2ND interface if this flag is 2 or higher.

elmarfaber, 2018-12-23:
The variant.h is a little bit smaller then the sample above:

#ifndef _VARIANT_ARDUINO_STM32_
#define _VARIANT_ARDUINO_STM32_

#define digitalPinToPort(P)        ( PIN_MAP[P].gpio_device )
#define digitalPinToBitMask(P)     ( BIT(PIN_MAP[P].gpio_bit) )
#define portOutputRegister(port)   ( &(port->regs->ODR) )
#define portInputRegister(port)    ( &(port->regs->IDR) )

#define portSetRegister(pin)		( &(PIN_MAP[pin].gpio_device->regs->BSRR) )
#define portClearRegister(pin)		( &(PIN_MAP[pin].gpio_device->regs->BRR) )

#define portConfigRegister(pin)		( &(PIN_MAP[pin].gpio_device->regs->CRL) )

static const uint8_t SS   = BOARD_SPI1_NSS_PIN;
static const uint8_t SS1  = BOARD_SPI2_NSS_PIN;
static const uint8_t MOSI = BOARD_SPI1_MOSI_PIN;
static const uint8_t MISO = BOARD_SPI1_MISO_PIN;
static const uint8_t SCK  = BOARD_SPI1_SCK_PIN;

#endif /* _VARIANT_ARDUINO_STM32_ */

So it wouldn't work for me

olikraus, 2018-12-23:
Hmm... Guess, this needs to be requested from the STM32 Arduino people

elmarfaber, 2018-12-27:
So I found another possibility, in version 4.5 of the platform "ST STM32" there was a function to change the SPI interface. With SPI.setModule(2); I could change the interface to SPI 2 and then then constructor
U8G2_ST7920_128X64_F_HW_SPI u8g2(U8G2_R0, displ_cs); works fine on the second SPI. With the mySensor it doesn't work but with a SDCard on SPI1 it works fine. Thank you very much for this great library and your help. I hope the work for the new function was not in vain and helps others!

deadbird99, 2019-01-24:
I have the very same problem using an ST7920 and MCP41010. ST7920 alone is ok, MCPs alone is ok too, but both results is one of them not working at all.

See #794

olikraus, 2019-01-24:
:)

closed 2019-02-09 by olikraus

Andy2No, 2019-09-08:
> All new constructors are listed here:
> https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#st7920-128x64

Thanks.  Please update the page linked to by the library example "hello world" program, which gives this link in a comment for choosing the constructor:

>> The complete list is available here: https://github.com/olikraus/u8g2/wiki/u8x8setupcpp

The ST7920 does not seem to be on that list so it's taken me a while to find the correct information.

olikraus, 2019-09-08:
Not clear what you mean... 

>  library example "hello world" program

Which example do you mean? There are many...

> The ST7920 does not seem to be on that list 

Yes, u8x8 is not available for ST7920

Andy2No, 2019-09-08:
Okay.  I guess there must be more than one library, and I have installed the wrong one, in that case.

I'm still baffled, but I'm sure I'll work it out eventually.

logiclayer, 2020-01-10:
Hardware and Software communication has a big performance difference on ST7920 chip and LCD ?

olikraus, 2020-01-10:
yes

logiclayer, 2020-01-10:
I've created my own XBM icons for segment digits. It looks like it shows a bit slow via SW mode. But I could not make it work via HW interface. That's my constructor. How can I make it work ?
`u8g2_Setup_st7920_s_128x64_f(&u8g2, U8G2_R0, u8x8_byte_4wire_sw_spi, u8g2_gpio_and_delay_stm32);`

olikraus, 2020-01-10:
Which board do you use? It does not look like an Arduino Environment.
Basically you need to replace u8x8_byte_4wire_sw_spi with a version which writes to your SPI subsystem.
See: https://github.com/olikraus/u8g2/wiki/Porting-to-new-MCU-platform#hardware-spi-communication

logiclayer, 2020-01-10:
I use STM32F103C8T6 and STM32CubeIDE. So please give me a sample about it.

olikraus, 2020-01-10:
U8g2 is a Arduino Lib. I do not know these environments and I do not know your controller. I also do not know u8g2_gpio_and_delay_stm32. Maybe approach the person who wrote this. Or write your own code.

logiclayer, 2020-01-10:
Arduino users are mostly beginner level of hardware and software programmers. STMCube or Keil users are more professional. You have a very good library for LCD. And if you can make it work in these platforms then your library users will grow up much faster. Your library will be like a brand for LCD displays. So my advice is you focus on that. These are my thoughts :)

olikraus, 2020-01-11:
> STMCube or Keil users are more professional. 

Then you should be able to follow the instructions from [here](https://github.com/olikraus/u8g2/wiki/Porting-to-new-MCU-platform#hardware-spi-communication) and implement a callback function for your uC.

> You have a very good library for LCD
 
Thanks

> And if you can make it work in these platforms then your library users will grow up much faster. Your library will be like a brand for LCD displays.

This is a spare time lib with exactly zero commercial interest. Do you really think I am interested in any growth? 

> So my advice is you focus on that. 

:-D  The good thing about a complete private, nc, open source project is: I can ignore advices. You have all the sources and instructions here. Please go ahead and implement the HW SPI callback. Let me know if you have any *specific* questions. I am usually always there to help (although i have to spent my personal spare time for this)

delgadosouza, 2020-05-26:
Dear Olikraus, thank you for this great library!

I have a bluepill STM32F103C8T6 and I would like to use a ST7920 display on the SPI2 (pins PB12 to PB15). On the SPI1 I have another sensor.

I used the SW constructor on those pins and it works fine but slow. Unfortunately, because my project is interactive I can't use it with this speed. 

I am using the official [Arduino_Core_STM32](https://github.com/stm32duino/Arduino_Core_STM32).

The constructor for the second SPI would be

`SPIClass mySPI_2(uint8_t mosi, uint8_t miso, uint8_t sclk, uint8_t ssel)`

by the documentation in their [wiki](https://github.com/stm32duino/wiki/wiki/API#spi).

I tried the HW version:

`U8G2_ST7920_128X64_F_2ND_HW_SPI disp(LCDROTATION, SS_RS_CS, RST_RESET);`

and it compiles but nothing is shown on the screen.

So I tried defining `SPI_INTERFACES_COUNT 2` on the beggining of the `U8x8lib.h` file and got the error:

`\src\U8x8lib.cpp:1000:7: error: request for member 'transfer' in '(SPI_TypeDef*)((1073741824 + 65536) + 12288)', which is of pointer type 'SPI_TypeDef*' (maybe you meant to use '->' ?)`

So I openned the `U8x8lib.cpp` file and tried to hardcode the mentioned constructor for the second SPI and exchanged all occurrences of `SPI1` by `mySPI_2` and it compiled but the screen was blank.

I have sill a lot to learn so I have no ideas how to solve this... Is it really a problem with the core I am using that don't follow the Arduino standard? Do you have any suggestions for a workaround?

delgadosouza, 2020-05-28:
I just received and tried the STM32F401CCU (Black Pill) and have exactly the same behaviour.
I openned a [new issue](https://github.com/olikraus/u8g2/issues/1208#issue-626514693), I hope you can help me...

olikraus, 2020-06-05:
Two devices could share the same SPI port of the CS line is different. Maybe you could use SPI1?