olikraus/u8g2 #769: st7920 monochrome with esp32

paborralho, 2018-12-22:
Hi, first of all thanks for this great library, maybe I am doing something wrong but my skills are very limited.
I don't even know if I should post this as an issue 
 I have a monochrome lcd 128x64 that works fine with this library and the arduino uno, but shows garbage with esp32. 
After some search, found that: st7920 SPI seems to be limited to 600kHz clock rate, so in the file U8x8lib.cpp, in this part:

case U8X8_MSG_BYTE_INIT: //= 20 initialization of SPI
      u8x8->bus_clock = u8x8->display_info->sck_clock_hz;

I forced:
      u8x8->bus_clock = 500000;   //limit clock to 500KHz

and all works fine. Thats my best, it is very brutal I know , I couldn't even pass it to the file as an #define and #ifdef

olikraus, 2018-12-22:
Will the command
```
u8g2.setBusClock(500000);
```
also work (https://github.com/olikraus/u8g2/wiki/u8g2reference#setbusclock). The advantage would be, that you do not need to modify the library sources.

paborralho, 2018-12-23:
I'm sorry, I really am a beginner, I don't fully understand where I can see the value of bus_clock. I tried some Serial.print() within the library U8x8lib.cpp but it seems that Serial.begin() in it causes some interference with SPI, or wire. I will try harder if you explain with me more detail.
Another question, is there a way of using your u8g2.userInterfaceSelectionList(); without begin(six keys); but with keypad keys from other libraries like Keypad.h for instance ? I need to manage many key in this project

olikraus, 2018-12-23:
Please excause. I assume I have confused you. The above mentioned problem description is for me only, because you have revealed a bug in the code of u8g2. 
I have already fixed the code in the above mentioned commit. For the next u8g2 release, also my suggested order of execution should work. 

If the solution from above works for you, then everything is great, don't worry.
Thanks a lot for reporting this problem!

closed 2019-01-14 by olikraus