olikraus/u8g2 #771: U8x8 split fonts (or bigger resolution fonts)

John-Lluch, 2018-12-23:
For the U8x8 library, it would be really nice to have a set of numeric fonts that are bigger resolution and can be displayed on the screen at full resolution. For example 16x16 or 32x32 pixel fonts that could be displayed starting at any row and column. 
Alternatively, a 16x16 font can be provided as a set of 4 related 8x8 fonts that are created by splitting the single characters of the original font into 4 quarters, top-left, top-right, bottom-left and bottom-right in a way that users can combine these 4 fonts on screen to effectively display greater resolution characters, with the already existing functions. 
Something like that would be a fantastic addition to the U8x8 library, and something that I am sure a lot of users would appreciate. 
Due to the particularities of my project I can't use the U8g2 library. However, the the U8x8 library fits perfectly for my needs.
Thanks,

olikraus, 2018-12-24:
Basically the drawTile procedure is there to allow such kind of drawings. In fact I guess, if I would implement this, then I will just call drawTile procedure, so this request is not really a library extension, but more simething like an add on, which could be implemented independently from the library.

But, well, ok, I think about this request.

Adding all the flexibility of a 16x16 and 32x32 fonts with the same API would mean a huge change.
It also will increase the library size. 

Your other idea with the split font, seems to be more easier to implement. No need to change the low level procedures in this case...

olikraus, 2018-12-24:
Ideas for the implementation of the split font feature.
 * Generate multiple fonts
 * Select the "corner" of the glyph during font generation, extend bg_8x8_convert for this: https://github.com/olikraus/u8g2/blob/master/tools/font/bdfconv/bdf_8x8.c#L60
 * Reuse existing drawGlyph procedure to draw the different "corners" of a 16x16 glyph. 
 * Switch between the four fonts

olikraus, 2018-12-24:
I played around with this problem. I have used you split font idea. The result is this:
[image]

It shows a 16x16 font and the same font, but extended to 16x32.

The biggest problem was the font construction, but no lib change was required.
Code is here: https://github.com/olikraus/u8g2/blob/master/sys/sdl/16x16_u8x8/main.c

The four fonts a part of the example.

olikraus, 2018-12-31:
Suitable fonts:
u8g2_font_lucasarts_scumm_subtitle_o

olikraus, 2019-01-02:
I have update the internal font format. Several larger fonts are now part of u8x8: https://github.com/olikraus/u8g2/wiki/fntlist8x8

olikraus, 2019-01-02:
... there is now native support for big 8x8 fonts...

closed 2019-01-02 by olikraus

John-Lluch, 2019-01-03:
Hi, Thank you very much for your later additions including the new fonts!.

In the mean time, I tried the "SSD1306Ascii" library (after I figured out that it also supports the SH1106 128x64 and other displays). 

Unfortunately, there's no documentation for that library, so you need to look at the source code to figure out things, which is not good for everybody. The developer does not seem to be fast at replying to user issues, either. And the name of the library is misleading too because the library supports more displays than its name implies.

However, I think that the SSD1306Ascii library has some good points that you may want to consider for a future version of your U8x8 library. Or maybe you can create a separate library based on the approach of the said one. Please don't take me wrong, your library is excellent and I rather would use yours based on the existing user base and support, but I think that there's something on the other library that deserves a look.

So the thing that I like from the "SSD1306Ascii" is that it will take any font width, including variable width fonts, for direct display. The trick is rows take 8 pixels but columns only use 1 pixel, so for example on a 128x64 display we have 8 rows and 128 columns. Given the appropriate font format, this enables users to display variable width fonts with ease, and big fonts too, at full resolution and without loosing precious screen space. Please see attached picture of what I mean. The big font on the centre is a 24 pixels height, variable width font. The width of most characters is not even multiple of 8, and it doesn't need to be.

[image]

olikraus, 2019-01-03:
Thanks for the feedback. The variable char width is a nice feature, indeed.

olikraus, 2019-01-14:
Until release of the milestone, this feature is available with the latest beta:

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).