olikraus/u8g2 #772: Question:  Would full constructor w/ firstPage/nextPage loop cause memory problems?

keith204, 2018-12-27:
I'm not looking for a solution, just wanting to verify something for peace of mind.  Basically after a few screen updates, my Arduino would freeze up.  I use your library on a lot of microcontrollers around here, so this was baffling me greatly.

At one point I realized that I had the full constructor:
```
U8G2_SSD1306_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0, U8X8_PIN_NONE);
```
and
```
u8g2.firstPage();
do {      
    ...        
} while( u8g2.nextPage() ); 
```
Realizing my folly, I changed to the clear/send method, which fixed the issue (or so I think... it's been running 12 hours now!)  ...including contents of loop in case it matters...
```
u8g2.clearBuffer();  

u8g2.setFont(u8g2_font_6x10_tf);
u8g2.setCursor(0,12);
u8g2.print("Satellites: ");
u8g2.print(num_satellites);

u8g2.setFont(u8g2_font_logisoso24_tn);

u8g2.setCursor(12,50);
if (i_limit < 100)
  u8g2.print(" ");
if (i_limit < 10)
  u8g2.print(" ");
u8g2.print(i_limit);

u8g2.setCursor(75,50);
if (i_mph < 100)
  u8g2.print(" ");
if (i_mph < 10)
  u8g2.print(" ");
u8g2.print(i_mph);

u8g2.setFont(u8g2_font_6x10_tf);
u8g2.setCursor(28,63); u8g2.print("ALERT");
u8g2.setCursor(102,63); u8g2.print("MPH");     
       
u8g2.sendBuffer();
```

I guess what I'm asking is... would it make sense that `_F_` and `do...while` would have caused this?

olikraus, 2018-12-27:
With the `_F_` constructor, both methods are valid. There should be no difference.

keith204, 2018-12-27:
Ok that possibility is what was teasing my mind and prompted me to post the question.  Well, I guess since I don't have a problem now, feel free to close this issue.  Maybe I had the Arduino sitting on a wire clipping or something.

olikraus, 2018-12-27:
If the F-constructor together with firstpage/nextpage loop would cause a crash, then this is a bug... At least i am not aware about something here...

closed 2019-01-02 by olikraus