olikraus/u8g2 #776: Segmentation Fault

WaRZaT, 2019-01-05:
Hi!
I have a segmentation fault with the rev 2.24.3 but not with 2.23.18
The segmentation fault occurs when i use a delay() function.
I tried to use delayMicroseconds()  function => segfault disappears
I tried to move de delay() function at a different place in the code => segfault stays
```

C:\Users\xxx\Documents\Arduino\libraries\U8g2\src\clib\u8g2_setup.c: In function 'u8g2_draw_l90_r0':

C:\Users\xxx\Documents\Arduino\libraries\U8g2\src\clib\u8g2_setup.c:303:1: internal compiler error: Segmentation fault

 }
 ^

```

per1234, 2019-01-05:
It's probably https://github.com/arduino/Arduino/issues/7949
This bug is specific to the 5.4.0-atmel3.6.1-arduino2 version of avr-gcc used by Arduino AVR Boards 1.6.22 and newer. 

#### Here's the traditional workaround:
1. Tools > Board > Boards Manager
1. Wait for downloads to finish.
1. When you move the mouse pointer over "Arduino AVR Boards", you will see a "Select version" dropdown menu appear. Select "1.6.21".
1. Click "Install".
1. Wait for installation to finish.
1. Click "Close".

Due to a [bug](url=https://github.com/arduino/Arduino/issues/7973), this workaround doesn't work with Arduino IDE 1.8.6, but it will work with any other version of the Arduino IDE.

If you have **File > Preferences > Check for updates on startup** checked, the Arduino IDE may occasionally notify you that a new version of Arduino AVR Boards is available, you'll need to refrain from updating back to the new Arduino AVR Boards version, otherwise you'll be back to seeing the segmentation fault error again.

---
#### Alternate workaround:
One of the Arduino developers [recently worked on this and has provided a potential fix by updating avr-gcc to 7.3.0](https://github.com/arduino/Arduino/issues/7949#issuecomment-442969335). There is a special beta testing hardware package you can install via the Arduino IDE's Boards Manager that has the fix:
1. (In the Arduino IDE) File > Preferences
1. In the "Additional Boards Manager URLs" field, enter: http://downloads.arduino.cc/packages/package_avr_7.3.0_index.json
1. If you already have an additional Boards Manager URL in that field, you can separate multiple URLs with commas or click the button on the right side of the field to open a dialog that allows you to add each on its own line.
1. Click "OK"
1. Tools > Board > Boards Manager
1. Wait for the downloads to finish
1. Click on "Arduino AVR Boards by Arduino"
1. Click the "Update" button.
1. Wait for the update to finish.
1. Click "Close".

Previous: https://github.com/olikraus/u8g2/issues/701

olikraus, 2019-01-14:
Is the problem solved? Can we close this?

olikraus, 2019-01-19:
no further feedback... closing...

closed 2019-01-19 by olikraus