olikraus/u8g2 #778: HX1230 set_contrast malfunction

eventus17, 2019-01-08:
Hi,
I'm recently working with cheap chinese LCD display called HX1230. I've notice some issues with the set_contrast function.

https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_pcf8814_hx1230.c

line: 131..137

```
#ifdef U8X8_WITH_SET_CONTRAST
    case U8X8_MSG_DISPLAY_SET_CONTRAST:
      u8x8_cad_StartTransfer(u8x8);
      u8x8_cad_SendCmd(u8x8, (arg_int>>3)|0x80 );	/* 0..31 for contrast */
      u8x8_cad_EndTransfer(u8x8);
      break;
#endif
```
This doesn't work `u8x8_cad_SendCmd(u8x8, (arg_int>>3)|0x80 );`
After changing to this `u8x8_cad_SendCmd(u8x8, 0x80 | arg_int );` contrast function works flawlesly in whole range `0..31`

olikraus, 2019-01-08:
Which values did you use for the u8g2.setContrast command?
The idea is that u8g2.setContrast(255) shell generate the highest contrast and u8g2.setContrast(0) the lowest contrast for all displays and OLEDs.
As a result, the range of the user interface (0..255) must be scaled down to the target range of the hardware (0..31). This is why I used the shift operator here.

In other words: The user does not need to lookup the datasheet to check the possible contrast range, instead the user just could use a proper value between 0 and 255.

eventus17, 2019-01-08:
OK, it's clear now. I've focused on the HX1230 datasheet without digging in the u8g2lib function description. I used values from the comment (0..31)

`u8x8_cad_SendCmd(u8x8, (arg_int>>3)|0x80 );	/* 0..31 for contrast */`
Original code with values 0..255 works as intended.

closed 2019-01-08 by eventus17