olikraus/u8g2 #779: Support for second I2C interface

tobiea27, 2019-01-09:
Hi!
This is my first GitHub post. I hope I'm doing it right ;)

So, I want to use the second I2C HW Interface of the ESP32 with a SH1106 driver. This only works for me with small modifications in this library.

With the U8G2_SH1106_128X64_NONAME_F_2ND_HW_I2C Constructor you can't remap the clock and data pin. This is a problem, because as far as I can see the Arduino Wire library hasn't default pins for WIRE1. [https://github.com/espressif/arduino-esp32/blob/master/libraries/Wire/src/Wire.cpp](https://github.com/espressif/arduino-esp32/blob/master/libraries/Wire/src/Wire.cpp)

With the U8G2_SH1106_128X64_NONAME_F_HW_I2C Constructor you can do that. So I modified it, to work with the second I2C interface, too.

This problem seems to exist in every Constructor with 2ND_HW_I2C. 

I hope the clock and data pin remapping could be implemented for all 2ND_HW_I2C Constructors.

To make it work I have to modify U8g2lib.h and U8x8lib.cpp:

**U8g2lib.h**
original
```C++
class U8G2_SH1106_128X64_NONAME_F_2ND_HW_I2C : public U8G2 {
  public: U8G2_SH1106_128X64_NONAME_F_2ND_HW_I2C(const u8g2_cb_t *rotation, uint8_t reset = U8X8_PIN_NONE) : U8G2() {
    u8g2_Setup_sh1106_i2c_128x64_noname_f(&u8g2, rotation, u8x8_byte_arduino_2nd_hw_i2c, u8x8_gpio_and_delay_arduino);
    u8x8_SetPin_HW_I2C(getU8x8(), reset);
  }
};
```

modified
```C++
class U8G2_SH1106_128X64_NONAME_F_2ND_HW_I2C : public U8G2 {
  public: U8G2_SH1106_128X64_NONAME_F_2ND_HW_I2C(const u8g2_cb_t *rotation, uint8_t reset = U8X8_PIN_NONE, uint8_t clock = U8X8_PIN_NONE, uint8_t data = U8X8_PIN_NONE) : U8G2() {
    u8g2_Setup_sh1106_i2c_128x64_noname_f(&u8g2, rotation, u8x8_byte_arduino_2nd_hw_i2c, u8x8_gpio_and_delay_arduino);
    u8x8_SetPin_HW_I2C(getU8x8(), reset, clock, data);
  }
};
```

**U8x8lib.cpp**
original
```C++
extern "C" uint8_t u8x8_byte_arduino_2nd_hw_i2c(U8X8_UNUSED u8x8_t *u8x8, U8X8_UNUSED uint8_t msg, U8X8_UNUSED uint8_t arg_int, U8X8_UNUSED void *arg_ptr)
{
#ifdef U8X8_HAVE_2ND_HW_I2C
  switch(msg)
  {
    case U8X8_MSG_BYTE_SEND:
      Wire1.write((uint8_t *)arg_ptr, (int)arg_int);
      break;
    case U8X8_MSG_BYTE_INIT:
      if ( u8x8->bus_clock == 0 ) 	/* issue 769 */
	u8x8->bus_clock = u8x8->display_info->i2c_bus_clock_100kHz * 100000UL;
      Wire1.begin();
      break;
    case U8X8_MSG_BYTE_SET_DC:
      break;
    case U8X8_MSG_BYTE_START_TRANSFER:
#if ARDUINO >= 10600
      /* not sure when the setClock function was introduced, but it is there since 1.6.0 */
      /* if there is any error with Wire.setClock() just remove this function call */
      Wire1.setClock(u8x8->bus_clock); 
#endif
      Wire1.beginTransmission(u8x8_GetI2CAddress(u8x8)>>1);
      break;
    case U8X8_MSG_BYTE_END_TRANSFER:
      Wire1.endTransmission();
      break;
    default:
      return 0;
  }
#endif
  return 1;
}
```

modified
```C++
extern "C" uint8_t u8x8_byte_arduino_2nd_hw_i2c(U8X8_UNUSED u8x8_t *u8x8, U8X8_UNUSED uint8_t msg, U8X8_UNUSED uint8_t arg_int, U8X8_UNUSED void *arg_ptr)
{
#ifdef U8X8_HAVE_2ND_HW_I2C
  switch(msg)
  {
    case U8X8_MSG_BYTE_SEND:
      Wire1.write((uint8_t *)arg_ptr, (int)arg_int);
      break;
    case U8X8_MSG_BYTE_INIT:
        u8x8->bus_clock = u8x8->display_info->i2c_bus_clock_100kHz * 100000UL;
      #if defined(ESP8266) || defined(ARDUINO_ARCH_ESP8266) || defined(ESP_PLATFORM) || defined(ARDUINO_ARCH_ESP32)
        /* for ESP8266/ESP32, Wire.begin has two more arguments: clock and data */
        if ( u8x8->pins[U8X8_PIN_I2C_CLOCK] != U8X8_PIN_NONE && u8x8->pins[U8X8_PIN_I2C_DATA] != U8X8_PIN_NONE )
        {
      // second argument for the wire lib is the clock pin. In u8g2, the first argument of the  clock pin in the clock/data pair
      Wire1.begin(u8x8->pins[U8X8_PIN_I2C_DATA] , u8x8->pins[U8X8_PIN_I2C_CLOCK]);
        }
        else
        {
      Wire1.begin();
        }
      #else
        Wire1.begin();
      #endif
    break;
    case U8X8_MSG_BYTE_SET_DC:
      break;
    case U8X8_MSG_BYTE_START_TRANSFER:
#if ARDUINO >= 10600
      /* not sure when the setClock function was introduced, but it is there since 1.6.0 */
      /* if there is any error with Wire.setClock() just remove this function call */
      Wire1.setClock(u8x8->bus_clock);
#endif
      Wire1.beginTransmission(u8x8_GetI2CAddress(u8x8)>>1);
      break;
    case U8X8_MSG_BYTE_END_TRANSFER:
      Wire1.endTransmission();
      break;
    default:
      return 0;
  }
#endif
  return 1;
}
```

Thanks for developing this awsome library!

olikraus, 2019-01-09:
ahm.... ok... got the point...
Request is to allow pin remapping for the 2nd interface, right?

tobiea27, 2019-01-09:
Yes, that's right. Might be a little confusing explanation from my side.

olikraus, 2019-02-09:
this has to be postponed...

v1993, 2019-02-09:
@tobiea27 Hi! I'm working on no-arduino port to ESP32. Development is paused now until I'll get new SPI display, but C side is functional. It does allow you to select pins.

Hope that it will help you later.

olikraus, 2019-05-19:
This is already open for a long time. Unfortunately effort is huge for this topic. Is it still required?

tobiea27, 2019-06-02:
I apologies for my late response. In my project I'm using the workaround mentioned above, which works fine for me.
So I don't need this enhancement right now.

olikraus, 2019-06-02:
Ok, thanks for the feedback.

olikraus, 2020-07-05:
closing this, because "on hold" topics should be closed.

closed 2020-07-05 by olikraus

mnutsch, 2021-04-19:
This post led me down a rabbit hole while I was trying to get 2 screens to work on different GPIO pins at the same time. I am doing this, because I don't want to have to re-solder the displays to have different addresses.

Basically, you should use the standard GPIO pins for the "2ND" constructor (without specifying them). Then you should specify non-standard pins in the regular constructor.

For example:

U8G2_SSD1306_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE, 14, 12); //non-standard pins
U8G2_SSD1306_128X64_NONAME_F_2ND_HW_I2C u8g2b(U8G2_R0, /* reset=*/ U8X8_PIN_NONE); //standard pins: 5, 4