olikraus/u8g2 #780: bdfconv with large font sizes

matphillips, 2019-01-10:
From the comment here https://github.com/olikraus/u8g2/issues/676#issuecomment-436631856

There seems to be a 255 byte limit for the conversion. Is this a restriction of the resulting file format, or of the converter?
I'd like to convert some glyphs to a large size, currently the biggest I can get is 44px at 72dpi.

Is there any way to get it to convert to a much larger size?

olikraus, 2019-01-10:
This is a restriction of the format. In some cases only some extra glyphs like @ or % violate this restriction. In this case you could just remove these chars if you don't require them.

matphillips, 2019-01-10:
ah, ok. Thanks for the answer anyway.

closed 2019-01-10 by matphillips