olikraus/u8g2 #781: Failing to use U8g2 lib with STM32 and 128x64 i2c OLED display (SH1106)

Kart5a, 2019-01-10:
**Board:** STM32F103C series
**Display:** 1.3" i2c OLED display (SH1106) (_Confirmed that works_)
[Pictures here](https://imgur.com/a/pKlmSQz)

**Issue:** Nothing never appears on the screen.

I am trying to run "Helloworld" example code (both full and page buffer).  The constructors that I have tried to uncomment (one at the time of course):
_U8G2_SH1106_128X64_NONAME_1_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);
U8G2_SH1106_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);_

I also tried to use software i2c with my pins but it didn't work either, so I decided to stick with hw I2C. I also tried to use SPI display in the pass but I had no success...

**Pins used:**
SCL - PB10
SDA - PB11
(Using 3.3V)

I also tried to use both: 
_u8g2.SetI2CAddress(0x3D * 2);_
and
_u8g2.SetI2CAddress(0x3C * 2);_
in my setup function. Didn't work.

I also read something about porting to new MCU [here](https://github.com/olikraus/u8g2/wiki/Porting-to-new-MCU-platform#stm32) but I have no clue how to do that or is that even a solution to my problem. U8g2 seems like perfect library so I would like to use it. For some reason nothing never appears on the screen. I am quite a beginner too so don't judge!

olikraus, 2019-01-12:
Which environment do you use? Which STM32 lib? I think the problem is in the improper implementation of the HAL for your uC.

Kart5a, 2019-01-12:
I am using Arduino Software IDE 1.8.7. I installed the "Arduino SAM Boards" from the board manager which allows STM32 programming just like any other arduino board. [Here are the instructions.](https://wiki.stm32duino.com/index.php?title=Installation)
Library should be this U8g2 library and "Helloworld" is an example code from this library.

olikraus, 2019-01-12:
ok, this should work out of the box, provided that the STM32 port is ok, then it should work.
Can you see any activity on the SCL / SDA lines?

Kart5a, 2019-01-13:
I can tell you as soon as I get my hands on my friends multimeter! (I don't own one yet)

I also tested OLED_I2C library which came with Arduino_STM32-master.zip (Downloaded from [here](https://wiki.stm32duino.com/index.php?title=Installation) too). 
I knew this library is for SSD1306. I still tried it and loaded "OLED_I2C_3D_Cube" from the examples and I was able to get something on the screen but it was not moving and there was just random pixels everywhere as expected. I didn't touch the code at all. [Here is a picture of my screen.](https://imgur.com/a/KpyIrbF)

olikraus, 2019-01-14:
Hmm... I do not see pullup resistors here. Just to exclude this as a problem: Can you add pullup resistors (10K) to SDK and SCL lines and test again.

Kart5a, 2019-01-14:
Is there way to use pullup resistors build in stm32? I think it already has pullup resistors... I can go and get a few tomorrow since I don't have any!

olikraus, 2019-01-14:
the internal pullup resistors are usually too weak. I mean they are in the range of 50K Ohm or higher.

Kart5a, 2019-01-14:
Alright! I am gonna get some resistors asap.

Kart5a, 2019-01-15:
I was able to get my hands on some resistors (10K) and measured some voltages. I used both 3.3V and 5V and both measured exactly 3.3V and 5V at any point. Nothing appears on the screen. Here is some pictures:
https://imgur.com/nGXOizW
https://imgur.com/C7uNES8
https://imgur.com/zGY5fJr

Tulsa2008, 2019-01-15:
Olikraus:
Just a general question concerning the use of pull up resistors.  I use SPI hardware with Arduino Megas on a regular basis.  I use the SPI to drive SD chip cards, displays, and other devises.  As a general rule, I get them to work but I have to keep the wires going to and from the devises as short as possible.  I have not been using pullup resistors and sometimes noise is an issue.  

Should I make it a general practice to use pullup resistors??

Thanks for all of your time and your lib.  I use your lib to drive several different displays.
 Ken

olikraus, 2019-01-15:
@Kart5a : I am sorry, that this still doesn't work. Actually, it is difficult to say, what is the problem here because there is some part of the u8g2 software, which was not written by me, but instead by the STM32 Arduino people. Maybe you can test the setup with a software I2C constructor and apply the pin names to the constructor as arguments.

@Tulsa2008 : Pullup resistors are required for I2C bus only. SPI does not require pullup resistors. I am sure there are some tutorials in the www about this topic.

Kart5a, 2019-01-15:
I really appriciate your time and effort you put into this. Thank you very much! I tested software I2C contructor but I had no success... Do you know maybe anyone who I could contact and ask for help? (maybe some STM32 Arduino people!?) I am very furstrated but I am not giving up. I have planned to use STM32 in my project and everything else is pretty much built around it. Do you have any suggestions?

Kart5a, 2019-01-16:
I had one Pro Micro laying around so I decided to test out U8g2 lib with it. I managed to get something on the screen but not without problems. [Here is the video](https://youtu.be/E4cx2TDPpqs). Do you know what may cause these problems? This is just a test since I cannot use Pro Micro in my project (too little storage).

Tulsa2008, 2019-01-16:
Olikraus:
Thanks for your response.  I didn't think pullup resistors were required for SPI but was confused about you commit on the subject.  Thanks for clarifying the issue.
Ken

olikraus, 2019-01-16:
@Kart5a 
Regarding the video: It works nicely, but the power supply is not sufficient. An OLED will draw a lot of current once something gets displayed. Infact the current increases with the number of white pixel on the display. Better use a reliable power supply instead of the USB bus.

Kart5a, 2019-01-16:
Alright! I broke my Pro Micro USB port so I am not able to test it... Is this the case with STM32 too?

olikraus, 2019-01-16:
>  Is this the case with STM32 too?

could be

olikraus, 2019-01-19:
people from the stm32 arduino project maybe...

effeci, 2019-01-28:
Just try to find the screen address by i2c scanner. Then confirm the lib is using the same address. Also f103 has 2 i2c interfaces where the first one is at PB 6 and pb7

effeci, 2019-01-28:
Oh and as well try roger Clark stm32duino framework. If still don't recognize anything try to specify in wire which interface are you using.

crackingthecodethomas, 2019-02-15:
We just had a similar issue using the SSD1327 I2C library. Remove the setI2CAddress member function and allow the library to try and auto detect the display. Whenever we used the set function, the library did not work but would compile.

olikraus, 2019-02-16:
Let me quote from the reference manual at https://github.com/olikraus/u8g2/wiki/u8g2reference#seti2caddress

> I2C address multiplied with 2 (the lowest bit must be zero)

meltdown03, 2019-02-17:
Try printing a String at location (0,20), My ESP8266 has an offset issue with the U8G2 lib. The U8X8 lib works perfectly though. I spent hours trying to get something on the screen until I changed the font and saw a few pixels light up. I also added an `if (!u8g2.begin()) {Serial.print("Display not started");}` conditional for debugging.

olikraus, 2019-02-17:
Unless chang

Serjy1710, 2019-05-07:
> Try printing a String at location (0,20), My ESP8266 has an offset issue with the U8G2 lib. The U8X8 lib works perfectly though. I spent hours trying to get something on the screen until I changed [...]

Hi meltdown03, 
could you please share sample code to proper init this OLED display with ESP8266? I've connected D3 as CLOCK and D5 as DATA, but could'n get any pictures on display - just darkness... I'm trying to use latest U8g2 library v.2.25.10 installed directly form Arduino IDE v.1.8.5.  
thank you for help

olikraus, 2019-06-20:
.... hm closing this issue... please reopen if required...

closed 2019-06-20 by olikraus