olikraus/u8g2 #782: u8x8 drawString taking progmem?

siochs, 2019-01-12:
Thanks for sharing this great library!
This is not an issue, it's a question about the API.
I assumed that `u8x8.drawString` takes `const char *` from RAM. But it seems, that it works also with `PROGMEM` strings? I am on ESP32 and Arduino Framework:
```
static const char a_pgm_string[] PROGMEM = "From Flash";
U8X8_SSD1306_128X64_NONAME_SW_I2C display(/* clock=*/ OLED_SCL, /* data=*/ OLED_SDA, /* reset=*/ OLED_RST);

void testTheDisplay()
{
  display.setCursor(0, 0);
  display.printf_P(a_pgm_string); // works

  display.setCursor(0, 1);
  display.printf( PSTR(a_pgm_string) ); // works

  display.drawString(0, 2, PSTR(a_pgm_string)); // works too!
  display.drawString(0, 3, a_pgm_string ); // even this works!?
}
``` 
Maybe this question sounds noobish, but I am a little confused Which one is the correct use of the API now?

siochs, 2019-01-12:
@olikraus thanks for enlightening me :-) Exactly, on AVR platforms, I used to place "strings" in PROGMEM.
What I found in the web was a little confusing:
[In this thread](https://esp32.com/viewtopic.php?t=1995), the const attribute is enough to keep data in flash (ESP-IDF). But in the [Arduino docs](https://arduino-esp8266.readthedocs.io/en/latest/PROGMEM.html) it seems that this "feature" was ported. It reads as if PROGMEM is required for storing in flash when using Arduino framework and ESP-Platform!?
In my case, I am not aiming to port the code to an other platform.

olikraus, 2019-01-12:
> It reads as if PROGMEM is required for storing in flash when using Arduino framework and ESP-Platform!?

Indeed, I think this is an AVR specific problem. I assume for ESP you can ignore the complete problem. Whether the string is placed in Flash or RAM does not matter.

olikraus, 2019-01-12:
Yes indeed, for I should be more precise: For ESP32 you can ignore the problem.

Maybe to clarify your problem:

First problem:
Some microcontrollers make a difference in accessing RAM or Flash. This means: The compiler must generate different code to access data, depending whether it is in RAM or Flash. This problem applies only to some controllers (ESP8266 and AVR to my knowledge, maybe others also)

Second Problem:
We (as a programmer) have to tell, where the data (e.g. the string) has to be placed. The implementation often goes different ways here, but usually some kind of section command is used for this purpose. For the AVR port of the gcc the PROGMEM keyword was introduced. But actually this just resolves to something like __attribute__((section(".blablabla"))). Of course the blablabla will be different between each system.

Now in principle, gcc can generate different code (problem 1) based on the section where the data is located (problem 2), but this is not so easy, because, as a programmer of a sub procedure, you do not want to care where the argument is located, but in fact, this is not possible: We can not simply write code, which works for strings which are located in Flash and RAM on some systems. As a consequence, for example AVR has introduced two procedures for this task (with and without _P)

On other systems (like ARM), the first problem does not exist, because the code is identical for RAM and Flash access. So we can use all the standard string manipulation procedures. ARM type systems allow a nice simplification: They just use the const keyword to put the code into Flash and reuse all existing string procedures.

siochs, 2019-01-14:
Thank you for clarification. Awesome. I close this issue now.

closed 2019-01-14 by siochs

olikraus, 2019-01-14:
ok :-)