olikraus/u8g2 #783: U8g2lib Invisible Pixels

grundiprogrammierer, 2019-01-13:
Hey, 
iam using the 128x128 1,5" I2C OLED Display with the U8g2lib.h
I use also the U8G2_SSD1327_EA_W128128_1_HW_I2C u8g2(U8G2_R0, U8X8_PIN_NONE); in my Code.
It runs good, but there are only 128x96 Pixels.
If i use 128x128 pixels, the others are invisible. 
It starts form the middle, there are 16 pixels on top and on the bottom who are invisible.
Sorry for my bad englisch :) 

Ill hope somebody can help me;)
Thanks a loooooot

olikraus, 2019-01-13:
Can you provide the name and datasheet of your display?
Can you send a picture of the u8g2 logo example?

grundiprogrammierer, 2019-01-17:
Here is a link for the display description.
I haven't found a datasheet of this display. Ill hope this helps you too.
http://www.raspberrypiwiki.com/index.php/1.5_inch_OLED_Shield

And here is the Picture of the u8g2 logo example with the following line of Contructor List : 

U8G2_SSD1327_EA_W128128_1_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);

[image]

Here is another example : The U8g2 page_buffer GraphicsTest Example with the following line of Contructor List : 

U8G2_SSD1327_EA_W128128_1_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);

[image]

Some lines of the display are invisible or deleted. I'll hope you understand my problem. 
Give me a feedback, if you need something more from me.

Thank you and best regards !!

olikraus, 2019-01-18:
Can you repeat the test with the `U8G2_SSD1327_MIDAS_128X128_1_HW_I2C` constructor?
What will be the results?

grundiprogrammierer, 2019-01-19:
WOW Thank you !!!
It works perfect!!!

I use the U8G2_SSD1327_MIDAS_128X128_1_HW_I2C constructor with the following parameters: 

U8G2_SSD1327_MIDAS_128X128_1_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);

I tried the GraphicsTest and the Logo Example and it works perfect.
Full display is used.

Thank you oli for your help !!!!

olikraus, 2019-01-19:
Great!
Will close this...

closed 2019-01-19 by olikraus