olikraus/u8g2 #784: Support of SSD1318 based display

feistyz, 2019-01-15:
Hi,

Is it possible to add a SSD1318 based 128x96 display? All relevant info attached.

Thanks!
[image]

[SSD1318_1 0.pdf](https://github.com/olikraus/u8g2/files/2761109/SSD1318_1.0.pdf)

olikraus, 2019-01-15:
hmmm the datasheet does not contain the commands for this device.
Can you use the SSD1306 noname constructor together with this display and send me a picture of the HelloWorld example?

feistyz, 2019-01-16:
Also, which commands are missing? I can try to get the info from the manufacturer.

olikraus, 2019-01-16:
> Using U8G2_SSD1306_128X64_NONAME_F_4W_HW_SPI the display stays blank. Are you expecting to see something?

My hope was, that the commands are the same. But this does not seem to be true.
The SSD1318 is an intelligent controller, which accepts commands to init the display and to show something on the display. Without knowning the commands, nothing can be implemented.
This list of all commands is usually part of the datasheet. 

As an example, see chapter 9 of the SSD1306: https://cdn-shop.adafruit.com/datasheets/SSD1306.pdf

feistyz, 2019-01-17:
That makes perfect sense. I will see if I can get the list from the mfg.

feistyz, 2019-01-18:
I got some info back... It's not the datasheet I wanted but I am hoping it is good enough. 

In the attached zip is a bring up sequence (startup.txt) and sample code for SSD1317 which allegedly has very similar command set to the SSD1318.

Let me know if this is enough to make it wall work.

[Display Sample Code.zip](https://github.com/olikraus/u8g2/files/2771685/Display.Sample.Code.zip)

olikraus, 2019-01-19:
hmmm... so I need to do the reverse engineering of the code, extract the necessary commands, put everything into u8g2 and hope that this is good enough to drive your SSD1318.
So the risk what I see are the mistakes during reverse engineering and the unknown differences between SSD1317 and SSD1318. So there is a high probability that we do all the steps, but the display still stays blank due to the above mentioned risks.

Moreover, this is just a spare time project. I currently put 1 to 2 h per week into u8g2 and the above tasks will for sure take much more time. With the mentioned risks, I doubt whether it is worth to start working on this.

feistyz, 2019-01-19:
Is there a way to contact you directly?

olikraus, 2019-01-19:
my e-mail address in there in each c file...

olikraus, 2019-01-20:
device: ssd1318_128x96

feistyz, 2019-01-20:
Thanks! I will test. 
Is there an Arduino build I can grab?

olikraus, 2019-01-20:
I just started with the implementation. That comment was just a note for myself... as I said.... some reverse engineering is required :-(

feistyz, 2019-01-20:
Oh sorry. I misunderstood. I thought it meant you wanted me to test.

olikraus, 2019-01-20:
in the example code from above... the initial brightness is very low: level 15 out of 255 possible levels. During testing, we should remember increasing this value, if nothing is visible...

olikraus, 2019-01-20:
code analysis: from what i see at the moment, code and the additional text file seem to be consistent. It also fits to the rare information from the pdf datasheet for the SSD1318. :-)

olikraus, 2019-01-20:
datasheet analysis: I see one difference. In the datasheet section 6.8.2, it is adviced to turn on the internal charge pump (8Dh 04h/44h/84h/C4h ACh 01h). However this sequence is not used in the setup code. Is this intentionally left out? Or not supported by the SSD1317? The command table would be good here... another topic, which needs to be tested with the real device I guess...

feistyz, 2019-01-20:
My guess is that it is not supported by SSD1317. 
But I agree. Testing will tell us.

olikraus, 2019-01-20:
If will put the following constructors into the examples, but all the other interfaces are also supported.
The I2C might have an issue. Will you test with SPI or I2C? SPI should be prefered.

```
//U8G2_SSD1318_128X96_1_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 13, /* data=*/ 11, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);  
//U8G2_SSD1318_128X96_1_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8); 	
//U8G2_SSD1318_128X96_F_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 13, /* data=*/ 11, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);  
//U8G2_SSD1318_128X96_F_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8); 	
```

feistyz, 2019-01-20:
This display only has SPI exposed.

olikraus, 2019-01-20:
ok then... after 2h of reading and programming, here is an initial version of u8g2 with ssd1318 support. Beta release 2.25.4

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

olikraus, 2019-01-20:
I do not expect, that everything works right now. At least I hope something is visible.
Testing advice:
 * Use SPI if possible
 * If you do not see anything, try to increase brightness: locate u8x8_d_ssd1318.c,  https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_ssd1318.c#L77, mabe replace 0x00f by 0x07f
 * If you still not see anything, try to activate the charge pump, uncomment https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_ssd1318.c#L65 

good luck

feistyz, 2019-01-20:
Trying

feistyz, 2019-01-20:
It is not turning on :-(
I tried all the variations you suggested and still no luck.

I will dig in more tomorrow.

olikraus, 2019-01-20:
One more version: Maybe test what happens, if you uncomment https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_ssd1318.c#L67

I have one doubt here: Even if the software part is fully correct, there are many options for faults on the hardware side:
 * Did you apply the correct caps for the charge pump?
 * Can you measure some higher voltage at the caps? Can you measure this >6V at one of the cap wires (as mentioned in section 6.9 of the datasheet)
 * How does the display look like? You have a flex connector only?
 * Did you setup the correct communication interface? 
 * Are the GPIO levels limited to 3.3v only?
 * Is the overall wiring correct?

olikraus, 2019-01-26:
I have created beta 2.25.5. It includes the new `U8G2_SSD1318_128X96_XCP` constructor.
Can you sent a picture of the HelloWorld example with this constructor?

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

feistyz, 2019-01-26:
Using this constructor:
U8G2_SSD1318_128X96_XCP_F_4W_HW_SPI u8g2(U8G2_R0, CS_PIN, DC_PIN, RESET_PIN);

I get this:
[image]

feistyz, 2019-01-26:
Yeah sorry... it is a bit hard to take a steady picture. 
This is what is on the screen:
[image]

feistyz, 2019-01-26:
It works!
[image]

feistyz, 2019-01-27:
Of course. 

There is no full frame flip mode. right?

olikraus, 2019-01-27:
I don't remember, but you can just use the full buffer constructor also.

feistyz, 2019-01-27:
No. I used this constructor:
U8G2_SSD1318_128X96_F_4W_HW_SPI u8g2(U8G2_R0, CS_PIN, DC_PIN, RESET_PIN);

feistyz, 2019-01-27:
Full Code:
```

#include <Arduino.h>
#include <U8g2lib.h>

#ifdef U8X8_HAVE_HW_SPI
#include <SPI.h>
#endif
#ifdef U8X8_HAVE_HW_I2C
#include <Wire.h>
#endif

#define CS_PIN 10
#define DC_PIN 9
#define RESET_PIN 6

U8G2_SSD1318_128X96_F_4W_HW_SPI u8g2(U8G2_R0, CS_PIN, DC_PIN, RESET_PIN);

// End of constructor list

void setup(void) {
  u8g2.begin();
  u8g2.setFont(u8g2_font_6x10_tr);
}

void draw(const char *s)
{
  u8g2.firstPage();
  do {
    u8g2.drawStr(0,15,"Hello World!");    
    u8g2.drawStr(0,30,s);    
    u8g2.drawFrame(0,0,u8g2.getDisplayWidth(),u8g2.getDisplayHeight() );
  } while ( u8g2.nextPage() );
  delay(2000);
}

void loop(void) {

  u8g2.setDisplayRotation(U8G2_R0);
  u8g2.setFlipMode(0);
  draw("R0,F0");
  u8g2.setFlipMode(1);
  draw("R0,F1");

  u8g2.setDisplayRotation(U8G2_R1);
  u8g2.setFlipMode(0);
  draw("R1,F0");
  u8g2.setFlipMode(1);
  draw("R1,F1");

  u8g2.setDisplayRotation(U8G2_R2);
  u8g2.setFlipMode(0);
  draw("R2,F0");
  u8g2.setFlipMode(1);
  draw("R2,F1");

  u8g2.setDisplayRotation(U8G2_R3);
  u8g2.setFlipMode(0);
  draw("R3,F0");
  u8g2.setFlipMode(1);
  draw("R3,F1");
}

```

olikraus, 2019-01-27:
ah, ok, fixed another bug in the u8g2 ssd1318 code. thanks!

do you need another release or do you pull the update directly from the git repo (fix is already done here)?

feistyz, 2019-01-27:
Happy to retest.
A release would be a bit easier.

olikraus, 2019-01-27:
no problem, here is beta 2.25.7:

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

feistyz, 2019-01-27:
No longer mirrored :)

[flipmode.mov.zip](https://github.com/olikraus/u8g2/files/2799883/flipmode.mov.zip)

olikraus, 2019-01-27:
wow, perfect. then I think all is done from my point of view :-)

feistyz, 2019-01-27:
I agree. Thank you!

closed 2019-02-09 by olikraus

ragofp, 2021-11-06:
@feistyz What's the part number of the display used here? I am shortlisting high PPI displays for wearable applications. Any lead would be much appreciated.

feistyz, 2021-11-06:
For a truly low ppi OLED you will need to pay someone to build a custom display which is not cheap.
Off the shelf you can start with this one: https://www.seeedstudio.com/0-5-Inch-OLED-display-Arduino-shield-p-2194.html

ragofp, 2021-11-07:
Thanks, @feistyz. I tried a few options. 128x64, 128x32, 64x48.. For my application, I'll need high density. The display you've shown looks pretty appropriate. Is it from WiseChip? What would it cost approximately?

feistyz, 2021-11-07:
I believe that display is from Panasonic. It's $15 from that link.

ragofp, 2021-11-07:
@feistyz sorry for the confusion. I was referring to the display you had posted earlier. :) [Link](https://user-images.githubusercontent.com/10438760/51793826-b69dce00-217b-11e9-836a-94a00f334de6.png)

feistyz, 2021-11-07:
@ragofp That is a proprietary custom display.