olikraus/u8g2 #785: Is there a way to detect if the display is on?

luebbe, 2019-01-16:
I have an application where a SSD1306 is connected to an ESP12F via I2C. The display is powered via a GPIO pin of the ESP. Since the device is running on batteries, the ESP is sent into light sleep after a configurable amount of time. When doing so, the GPIO power pin for the display is set low.
Light sleep was chosen in order to have a quicker wake-up time when a button is pressed on the device as opposed to having to go through the entire boot process.
Upon boot or waking up from light sleep, the power pin is set high, display.begin() is called and a 100ms delay is added before sending output to the display.
This works 100% of the time upon cold boot (power on), but it is very unreliable upon waking up from light sleep. Sleeping for 2500ms or calling display.begin() a second time after 500 ms seems to be enough for the display to be really "awake" after the GPIO has been set high. This feels very hacky and is too slow for the user.
Ideally I would like to have non-blocking code after wake up, check if the display is awake in loop() and act accordingly.
So is there a way to detect if the display reacts?

olikraus, 2019-01-16:
Hmmm... it is not really clear to me what you want to do. I mean, I do not really understand your problem.

However, the simple answer is: There is no way to detect whether the display is ready with u8g2. U8g2 is a write only library.

Since I didn't understand your problem, I can just give some general advice:

1.
Turning off the power supply for the display completly, will definitly require a very long startup phase. I have seen displays, which require more than a second to be ready after power supply is applied. So the question is, why don't you use the display sleep mode to power down the display? The power consumption will be very small, but the power up will be very quick. I have a battery based system here with a ssd1306 display, which run for over a year.
I  have described a cold/warm start mechanism here: https://github.com/olikraus/u8g2/issues/758

2. 
You can detect whether the display is available by pinging the display via its I2C address. It it answer with an ACK signal, then at least the display controller is ready.

luebbe, 2019-01-17:
Thanks @olikraus. These are already some very good pointers. I didn't even know that displays have a sleep mode. I'll read through #758 and see how far it takes me. Does the sleep mode only exist for SPI or also for I2C displays? I only have an I2C.
If not I'll try the ping approach.

olikraus, 2019-01-17:
> Does the sleep mode only exist for SPI or also for I2C displays? I only have an I2C.

It is there for both. In fact there is a u8g2 function for it: https://github.com/olikraus/u8g2/wiki/u8g2reference#setpowersave

closed 2019-01-19 by olikraus

luebbe, 2019-01-20:
Thanks a lot! setPowersave did exactly what I wanted to achieve.