olikraus/u8g2 #787: 1.3" OLED working with u8g but not with u8g2

Chade, 2019-01-18:
Hi,

I tested a 1.3" I2C OLED Display (https://www.az-delivery.de/products/1-3zoll-i2c-oled-display) with the u8glib successfully (using U8GLIB_SH1106_128x64 constructor).
Now I wanted to switch to u8g2, but using this constructor U8G2_SH1106_128X64_NONAME_1_HW_I2C, it is not working (Display remaining black).
I tried my own code and also the provided examples from u8glib and u8g2 with the same result.
Arduino (Mini Pro) and wiring remaining the same.

Best regards,

Stefan

Chade, 2019-01-18:
Found out, that the display standalone is working with u8g2.
There are two more I2C devices on the bus, a RTC and a FM Radio module.
The FM Radio module is causing the problems, without it its working fine.

The question remains, why is u8g working in my setup, u8g2 not?
I guess, there has to be some difference during initialization(?), so that with the Radio module on the I2C bus u8g2 can't establish a connection to the display.
That fits another observation I made:  With the Radio module present, u8g2 is "hanging in the ".begin()" function.

I already checked the device addresses, there semmed to be no conflict there.

Any ideas?

Best regards,
Stefan

olikraus, 2019-01-19:
One big difference between u8g and u8g2 is the I2C communication.
u8glib: low level register programming on the target uC
u8g2: Wire.h

So actually, u8g2 should be more portable if all the libs use the official Arduino interfaces (like Wire.h for I2C). 

If you say, that u8g2 works without the radio part, then I assume it is a problem of the radio lib. Does the radio lib use Wire.h too?

One generic question: Do you have pullup resistors ob both lines? Without pullups the ATMega processor might hang with the I2C bus.

And one more remark: This topic is also discussed in the U8g2 FAQ.

Chade, 2019-01-21:
Hi,

the tip regarding the pullups was spot on.
Although the radio module already has 4k7 pullups, I added another pair of pullups for good measure, and now it's working.
There are still some stability issues, but overall it now works as intended.

Thanks very much!

closed 2019-01-21 by Chade