olikraus/u8g2 #788: ESP32 - Screenful of dots

william-ferguson-au, 2019-01-19:
I'm trying to use u8g2 in an ESP32 project built using Make.

If I add component.mk to the components/U8g2_Arduino folder
```
COMPONENT_SRCDIRS:=src src/clib
COMPONENT_ADD_INCLUDEDIRS:=src src/clib
```

then I can successfully compile and link the following against the u8g2 C++ libraries

```
#include <U8g2lib.h>
#include "driver/gpio.h"
#include "esp_log.h"

// OLED
#define OLED_POWER    GPIO_NUM_23
#define OLED_RESET    GPIO_NUM_26
#define OLED_CLOCK    GPIO_NUM_14
#define OLED_DATA     GPIO_NUM_27

U8G2_SSD1306_128X64_NONAME_F_SW_I2C* display;

static const char* TAG = "main";

extern "C" {
	void app_main();
}

void app_main(void)
{
    display = new U8G2_SSD1306_128X64_NONAME_F_SW_I2C(U8G2_R0, /* clock=*/ OLED_CLOCK, /* data=*/ OLED_DATA, /* reset=*/ OLED_RESET);
    display->begin();

    display->setFont(u8g2_font_ncenB14_tr);
    ESP_LOGI(TAG, "FONT_HEADER height=%u", display->getMaxCharHeight());

    display->drawStr(0, 24, "Hello World");

    display->sendBuffer();

    ESP_LOGI(TAG, "Display initialised");

    while (true) {
    	// do nothing
    	sleep(1);
    }
}
```

But when it runs 

1. Height for the font is 1, and ascent and descent return 0.
2. **The display is filled with dots**.

If I compile against the C libraries only, then I can get the display to render properly, but the font height still returns 1, and ascent and descent return 0.

I would prefer to use the C++ libraries, but I definitely need to be able to get font height.

Any ideas on what I am missing?

olikraus, 2019-01-19:
Unfortunately this project only supports the Arduino IDE integration. I do not know anything about your integration for ESP32. 

I think there is a ESP32 board package for the Arduino IDE, which should work.

william-ferguson-au, 2019-01-19:
Thanks @v1993
I'd offer to help with the port, but I'd probably be more of a hindrance.
Is it visible anywhere yet?

v1993, 2019-01-19:
@william-ferguson-au Only in local FS for now :)

It's in early alpha. General stuff is completed (in fact, it fits into just few fuctions), but HW SPI/I2C is in work. Not tested yet too.

After getting C-side stuff workin I'll have to tweak C++ bidings generator too (TBH, its code scaries me now).

v1993, 2019-01-20:
@william-ferguson-au if you're interested, you can check my fork: https://github.com/v1993/u8g2

It's still WIP, but plain C with SW I2C should work.

It's my `main.c`:

```
#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_system.h"
#include "esp_spi_flash.h"
#include <U8x8lib.h>
#include <u8g2.h>

u8g2_t u8g2;

void app_main()
{
	u8g2_Setup_ssd1306_i2c_128x64_noname_f(&u8g2, U8G2_R0, u8x8_byte_sw_i2c, u8x8_gpio_and_delay_espidf);
	//u8x8_Setup(u8g2_GetU8x8(&u8g2), u8x8_d_ssd1306_128x64_noname, u8x8_cad_ssd13xx_i2c, u8x8_byte_sw_i2c, u8x8_gpio_and_delay_espidf);
	u8x8_SetPin_SW_I2C(u8g2_GetU8x8(&u8g2), 26, 27, U8X8_PIN_NONE);
	u8g2_InitDisplay(&u8g2);
	u8g2_ClearDisplay(&u8g2);
	u8g2_SetPowerSave(&u8g2, 0);
	u8g2_SetFont(&u8g2, u8g2_font_7x14_tf);
	u8g2_DrawStr(&u8g2, 50,50, "Hello World!");
	u8g2_SetDrawColor(&u8g2, 2);
	u8g2_DrawBox(&u8g2, 0, 0, 255, 127);
	u8g2_SendBuffer(&u8g2);
	u8g2_SendBuffer(&u8g2);
	while(1) vTaskDelay(10);
}
```

HW I2C/SPI are implemented but not tested (which means that they will 99% likely fail). C++ bindings are different task (I'll do them after getting C part working completely).

NOTE: use traditional `make` build for now. CMake will be done later too.

Patrik-Berglund, 2019-01-22:
Hi,

Here's a port implemented in C from nkolban for the ESP32 (ESP-IDF), have been using it for over a year now and without problems.

https://github.com/nkolban/esp32-snippets/tree/master/hardware/displays/U8G2

william-ferguson-au, 2019-01-23:
Thanks @Patrik-Berglund , but I am already using Neil Kolban's port.

Question for you:
What values do you get back from 

```
u8g2_SetFont(&u8g2, <chooseAnyFont>);

u8g2_GetMaxCharHeight(&u8g2);
u8g2_GetAscent(&u8g2);
u8g2_GetDescent(&u8g2));
```

I get 1, 0, 0 respectively.

olikraus, 2019-01-23:
Strange, this should not depend on the low level code. Which font did you use? Will it happen with all fonts?

william-ferguson-au, 2019-01-23:
No idea if it happens for all fonts. I haven't found a font it doesn't occur for.
It occurs for these:

u8g2_font_helvB14_tr;
u8g2_font_helvR10_tr;
u8g2_font_helvR08_tr;

u8g2_font_8x13_mf

olikraus, 2019-01-24:
ok... need to look at this more closely

william-ferguson-au, 2019-04-27:
@olikraus I can't remember exactly what the cause was, but it was definitely something to do with my setup, so I'm closing this.

closed 2019-04-27 by william-ferguson-au

olikraus, 2019-04-28:
ok...