olikraus/u8g2 #789: Porting to platform with

v1993, 2019-01-19:


olikraus, 2019-01-19:
U8g2 itself is a pure C library. For Arduino however it calls the C++ Arduino HAL and adds a C++ lib API wrapper. Both, Arduino wrapper and HAL are put together in the same c++/h files. As a consequence: If you want the C++ U8g2 API but not the C++ Arduino HAL, then some code needs to be copied out the existing C++ code. 

I am not 100%, but I think this has been done by the nocemcu people already: https://nodemcu.readthedocs.io/en/dev-esp32/modules/u8g2/

In general porting instructions are here: https://github.com/olikraus/u8g2/wiki/Porting-to-new-MCU-platform

v1993, 2019-01-19:
Thank you. Now I think about architecture: is it good idea to place ESP32 specific stuff in `sys/esp-idf`? It will be included from `CMakeLists.txt` or `component.mk`. I also plan adding config (for ESP32 menuconfig menu) for 16bit mode and fonts to make u8g2 feel like nice component.

Also, where can I find pure C exmples? I feel like there is a lack of them.

olikraus, 2019-01-19:
> Now I think about architecture: is it good idea to place ESP32 specific stuff in sys/esp-idf? It will be included from CMakeLists.txt or component.mk

I have no idea about this. CMakeLists.txt or component.mk in the u8g2 are part of a pull request. I can not tell you something about these files.

> Also, where can I find pure C exmples? 

Have a look here: https://github.com/olikraus/u8g2/tree/master/sys
All directories except "arduino" will contain pure c examples.

Here is a "hello world" example for the STM32L031:
https://github.com/olikraus/u8g2/blob/master/sys/arm/stm32l031x6/u8x8_test/main.c

Once I created a blog entry also:
https://drolliblog.wordpress.com/2017/11/02/stm32l031x6-eval-board/

> I feel like there is a lack of them.

I personally feel, that I created more docs and examples than many other Arduino libs...

v1993, 2019-01-19:
Thanks you againg and sorry for blaming you about no docs. Your Arduino/C++ reference is just great! I just had some problems about pure C documentation.

v1993, 2019-01-19:
> Thanks you again and sorry for blaming you about no docs. Your Arduino/C++ reference is just great! I just had some problems about pure C documentation.

olikraus, 2019-01-19:
:-)

Let me know if I can help you with u8g2 related problems.

v1993, 2019-01-19:
> > Now I think about architecture: is it good idea to place ESP32 specific stuff in sys/esp-idf? It will be included from CMakeLists.txt or component.mk
> 
> I have no idea about this. CMakeLists.txt or component.mk in the u8g2 are part of a pull request. I can not tell you something about these files.

There are no problems with changing those files, just wonder is `sys/platformname` a right place to store platform-specific stuff (I/O frunctions for ESP32 in this case).

olikraus, 2019-01-19:
> There are no problems with changing those files, just wonder is sys/platformname a right place to store platform-specific stuff (I/O frunctions for ESP32 in this case).

You mean below u8g2? Well sure, it you like to do so and send a corresponding pull request then I can accept this.

v1993, 2019-01-19:
Thank you! Another question: how presice should be all the `U8X8_MSG_DELAY_*` stuff should be at waiting?

There are three options: use `vTaskDelay` (round time to ticks, pretty rough), `ets_delay_us` for busywait correct time (much better at timing, but not let anything else run while waiting), or `ets_delay_us` with disabeling interrupts with `portDISABLE_INTERRUPTS`/`portENABLE_INTERRUPTS` (100% correct timing, but extremely bad at long and not that presice waitings due to not be able to handle interrupts).

I suggest using `vTaskDelay` on `≥ 10ms` delays and `ets_delay_us` on smaller scale. Are there any parts which should be more accurate at timing (so disable interrupts)? If so, it's still better to disable them in actual calling code and not callback.

Thanks.

olikraus, 2019-01-19:
> Another question: how presice should be all the U8X8_MSG_DELAY_* stuff should be at waiting?

It always can be more than requested, so: It does not need to be precise.

> Are there any parts which should be more accurate at timing (so disable interrupts)?

I think the only sensitive message is the I2C delay. But this is only required for software emulated I2C.

> Where code generator and template files are?

You probably do not need to take care on the code generator. "codebuild" only connects the displays with the communication interfaces: It defines which types of interfaces are available for which display.

The hardware specific files are either in the sys/<platform>/common directory or (an exception) for  Arduino in the u8g2/cppsrc directory.

> My current idea it to leave csrc as is

Yes please. "csrc" should not contain any hardware specific code.

v1993, 2019-01-19:
Are Arduino display C++ binding auto generated? If yes, because I want to support them on ESP-IDF too, it will require tweaking code generator.

Am I right?

My goal is to provide C++ bindings compatible with Arduino interface too, because C++ is commonly used on ESP32 too.

olikraus, 2019-01-19:
> Are Arduino display C++ binding auto generated?

If you mean the constructor classes here, then yes:
https://github.com/olikraus/u8g2/blob/master/cppsrc/U8g2lib.h#L404

olikraus, 2019-01-19:
Or you create a copy of the codebuild Code and add your modifications there.

v1993, 2019-01-19:
To be fair, current code generator looks like hell for me ☺.

I'll look at it when get pure C working.

Do you have anything againist re-writing in in something non-C? I'm good (enough to write such thing) with C++ and lua.

v1993, 2019-01-19:
Can you take a look an my test implementation for `gpio_and_delay`, please?

```
uint8_t u8x8_gpio_and_delay_espidf(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, U8X8_UNUSED void *arg_ptr)
{
    uint8_t i;
    switch(msg)
    {
    case U8X8_MSG_GPIO_AND_DELAY_INIT:
        for( i = 0; i < U8X8_PIN_CNT; i++ )
            if ( u8x8->pins[i] != U8X8_PIN_NONE )
            {
            bool success;
                if ( i < U8X8_PIN_OUTPUT_CNT )
                {
					success = u8x8_setpinoutput_espidf(u8x8->pins[i]);
                }
                else
                {
					success = u8x8_setpininput_espidf(u8x8->pins[i]);
                }
                if (!success) return 0;
            }

        break;

    case U8X8_MSG_DELAY_NANO:
        ets_delay_us(arg_int==0?0:1);
        break;

    case U8X8_MSG_DELAY_10MICRO:
        /* not used at the moment */
        break;

    case U8X8_MSG_DELAY_100NANO:
        /* not used at the moment */
        break;

    case U8X8_MSG_DELAY_MILLI:
        vTaskDelay(arg_int);
        break;
    case U8X8_MSG_DELAY_I2C:
        /* arg_int is 1 or 4: 100KHz (5us) or 400KHz (1.25us) */
        ets_delay_us(arg_int<=2?5:2);
        break;
    case U8X8_MSG_GPIO_I2C_CLOCK:
    case U8X8_MSG_GPIO_I2C_DATA:
		auto pin = u8x8_GetPinValue(u8x8, msg);
        if ( arg_int == 0 )
        {
            if (
            (!u8x8_setpinoutput_espidf(pin)) ||
            (gpio_set_level(pin, 0) != ESP_OK)
            ) return 0;
        }
        else
        {
            if (!u8x8_setpininput_espidf(pin)) return 0;
        }
        break;
    default:
        if ( msg >= U8X8_MSG_GPIO(0) )
        {
            i = u8x8_GetPinValue(u8x8, msg);
            if ( i != U8X8_PIN_NONE )
            {
                if ( u8x8_GetPinIndex(u8x8, msg) < U8X8_PIN_OUTPUT_CNT )
                {
                    gpio_set_level(i, arg_int);
                }
                else
                {
                    u8x8_SetGPIOResult(u8x8, gpio_get_level(i));
                }
            }
            break;
        }

        return 0;
    }
    return 1;
}
```
Are there any visible flaws? Thanks.

olikraus, 2019-01-19:
> To be fair, current code generator looks like hell for me 

then, it is better to work on a copy... ;-)

> Do you have anything againist re-writing in in something non-C?

I do not care if you work on your own repo ;-)
For pull requests: New files in new directories should be ok. For the existing code I prefer C over C++

v1993, 2019-01-19:
Hmm, seems there is a problem…

Arduino introduce global SPI state, but ESP-IDF does not. Doing so would mean big problems with more than displays running in parallel. It would be really nice to have non-interputtable version where SPI state don't have to be stored between calls. Any ideas?

Possible solution is to add SPI and I2C states to `u8x8_t` struct on ESP32. How bad is it? In my opinion, that's ok, but you're main developer after all…

v1993, 2019-01-19:
BTW, do you have an IRC/Jabber? It probably would be faster to communicate there (nothing personal, feel free to reject with or without reason).

v1993, 2019-01-19:
Allright, got it! There IS global state on hardware level (`SPI1` and `SPI2`), so I make one "universal" function which can work with any SPI descriptor and call it from `SPI(1/2)` handlers.

v1993, 2019-01-19:
Well, just got that I was doing HW SPI all wrong… It's better try to get ESP-IDF driver working and NOT try to adapt one from arduino-esp32.

I have some questions now:

1. How big data transfers usually are? There are huge optimisations for < 32 byte ones.
2. Do all SPI stuff have to be syncronized?
3. Do U8G2 read anything? I'm pretty sure it isn't, just want to be 100% clear about it.

olikraus, 2019-01-19:
> Possible solution is to add SPI and I2C states to u8x8_t struct on ESP32. How bad is it?

Yes, it will break the independence from SPI and I2C

> BTW, do you have an IRC/Jabber?

No

> How big data transfers usually are? There are huge optimisations for < 32 byte ones.

I2C transfers are limited to 32 bytes. Any other transfers can have any length. 

> Do all SPI stuff have to be syncronized?

Against what?

> Do U8G2 read anything? I'm pretty sure it isn't, just want to be 100% clear about it.

U8g2 does not read from displays (but from GPIOs)

Just a general notes: I do not have much time to work on U8G2. So I can not do more than give comments on your questions. 
The core part of u8g2 has to be C and it must not depend on any low level, system specific code... at least if you want me to accept pull requests.

v1993, 2019-01-19:
Ok, thank you.

> > Do all SPI stuff have to be syncronized?
> 
> Against what?

In this case syncronized means no caching in driver, i.e. use polling (write means waiting until transfer is complete) and not interrupt (things goes into queue and are done in background) SPI mode (ESP-IDF specific stuff).

P.S.: I was talking about HW SPI transfers, or to be presice `arg_int` for `U8X8_MSG_BYTE_SEND` message. Sorry for not being clear.

v1993, 2019-01-19:
Also, if adding stuff to `u8x8_t` is bad, can I use `U8X8_WITH_USER_PTR` and store my stuff by that pointer?

v1993, 2019-01-19:
Hi. My porting is going ok yet. There is a question: can I be sure that memory pointed by `arg_ptr` is HW SPI callback is on stack? Thanks.

olikraus, 2019-01-19:
> Also, if adding stuff to u8x8_t is bad, can I use U8X8_WITH_USER_PTR and store my stuff by that pointer?

Yes, sure, that is the purpose of the user ptr

> There is a question: can I be sure that memory pointed by arg_ptr is HW SPI callback is on stack?

probably not.

v1993, 2019-01-19:
Interesting thing: it's possible to use two displays with all same pins except CS which is handled by driver (even from two cores)!

However, I want to know one thing: is it possible to trace destruction of u8x8? I know that it's rare thing, but it would be nice to delete device in such a case.

Thanks.

v1993, 2019-01-19:
And another (more actual) question: can I make few SPI transactions instead of one?

olikraus, 2019-01-20:
> Interesting thing: it's possible to use two displays with all same pins except CS (and DC I think) which is handled by driver (even from two cores)!

Yes, sure, just create two different u8x8 / u8g2 structures (classes).

> However, I want to know one thing: is it possible to trace destruction of u8x8?

What do you want to trace? U8g2/U8x8 do not use dynamic memory.

> I know that it's rare thing, but it would be nice to delete device in such a case.

There are many nice features. All of them will cost resources. And people already complain about u8g2 size. Nevertheless, i think this is very much possible.

v1993, 2019-01-20:
At first, what about my last question?

Arduino have SPI API like this:

* Begin transaction
* Write byes
* End transaction

ESP-IDF, on other hand, does it in different way:

1. * Add transaction (complete) into queue
   * Request results later
2. * Do synchronous transaction (complete memory required again)
Under complete memory I mean that only bytes available at moment of beginning transaction will be sent in it.

So, is it ok to perform complete transaction on every "write" and ignore transaction begin/end (CS managed by ESP-IDF SPI driver itself)?

olikraus, 2019-01-20:
> So, is it ok to perform complete transaction on every "write" and ignore transaction begin/end (CS managed by ESP-IDF SPI driver itself)?

Unknown. It probably will depend on the display controller. It is like this: The display controller will receive commands from u8g2. A command sometimes include a command byte and an argument bytes, so it may consist of two (or more) bytes. A controller may allow the reception of these two bytes in two different transactions, but this is not 100% clear to me. You need to study the graphics controller datasheets for this. In u8g2 it is ensured, that command and args are always sent in the same transaction (the CS line does not change between command byte and argument byte of the same full command).

v1993, 2019-01-20:
And about "destructor":

U8g2 itself don't use dynamic memory, but some implementations (like mine) may need to or require some hardware descriptors. Possible implemetation is quite simple:

* Add `U8X8_MSG_GPIO_AND_DELAY_CLEAN` and pass it to main callback.
* Let C users call `u8x8_destroy()` or something like that.
* For C++, call it from actual destructor.

Souldn't be hard or breaking, right?

> A command sometimes include a command byte and an argument bytes, so it may consist of two (or more) bytes.

Hmm, seems pretty bad. Is it accepable to just send them as usually or will I have to use [these fields](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/peripherals/spi_master.html#_CPPv2N17spi_transaction_t3cmdE)? That would be complicated.

Another solution is to allocate buffer for SPI stuff and do actual transaction on "end transaction".

> You need to study the graphics controller datasheets for this.

I'm not ready to do so on all displays supported by u8g2…

Thansk.

v1993, 2019-01-20:
And another question: what is max size of SPI transaction? I have to set in bus configuration.

Can I rely on default 4094 bytes?

olikraus, 2019-01-20:
> Can I rely on default 4094 bytes?

It will probably much shorter. For I2C it is limited to 32 for SPI I assume it will not exceed 512 bytes. The only exception might be e-paper devices, which might require larger blocks.

v1993, 2019-01-20:
Ok, thanks. I'll add into config then (with 512 as default).

v1993, 2019-01-20:
Should CS line be positive or negative during communication?

Default driver's behaviour is to set it ti negative, but there is a flag `SPI_DEVICE_POSITIVE_CS`. Should I use it?

P.S.: what about destructor idea? Thanks.

v1993, 2019-01-20:
> > And yet another question: can SPI mode change for device?
> 
> Yes of course.

I mean, for one device between different transactions? If still yes, that's pretty bad, because it mean that I have to remove and add device again, reducing speed a lot.

> > P.S.: what about destructor idea? Thanks.
> 
> U8g2 is fine with the C++ default constructor, because there is no further dynamic memory allocation.

As I already said, implementation may need to do internal cleanup. I have to `malloc` things to store custom stuff.

> > Should CS line be positive or negative during communication?
> 
> Depends on the display:
> [/csrc/u8x8.h@`master`#L224](https://github.com/olikraus/u8g2/blob/master/csrc/u8x8.h#L224)

If `chip_enable_level == 1`, should it be negative or positive during activity?

olikraus, 2019-01-20:
> > > And yet another question: can SPI mode change for device?
> > 
> > Yes of course.
> I mean, for one device between different transactions? If still yes, that's pretty bad, because it mean > that I have to remove and add device again, reducing speed a lot.

no

> > > P.S.: what about destructor idea? Thanks.
> > 
> > U8g2 is fine with the C++ default constructor, because there is no further dynamic memory allocation.
> As I already said, implementation may need to do internal cleanup. I have to `malloc` things to store > custom stuff.

I assume you will create a new interface, so of course, you should provide a proper constructor.

> > > Should CS line be positive or negative during communication?
> > 
> > Depends on the display:
> > [/csrc/u8x8.h@`master`#L224](https://github.com/olikraus/u8g2/blob/master/csrc/u8x8.h#L224)
> If `chip_enable_level == 1`, should it be negative or positive during activity?

`chip_enable_level` is not a variable, it is a calibration value which returns the level. If your transfer is started, you should set the chip select level like this:
```
  case U8X8_MSG_BYTE_START_TRANSFER:
      u8x8_gpio_SetCS(u8x8, u8x8->display_info->chip_enable_level);  
```

> Can say one thing for sure: SW I2C for SSD1306 from pure C is working just fine!

nice

v1993, 2019-01-20:
I'm having a strange problems when try to get HW I2C working.

Display correctly responds with ACK on data and adress, but nothing works. If I try rebooting from SW to HW implementation (without changing screen), some black lines appears, like this:

[image]

Can you guess any reason? Thanks.

olikraus, 2019-01-20:
Difficult to say. Speed might be to high. Or the picture just reflects the content from the last correct update with some garbage from the restart.

v1993, 2019-01-20:
At first, I started SW I2C version which draw the picture. Then flased HW I2C version, then rebooted few times after finishing (more lines after each restart).

Display changes only when actual sending occours (I use logging to PC terminal).

I tried changing speed, but can't get working display this way.

Tried speeds:

* 400000Hz (from U8G2)
* 100000Hz
* 10000Hz
* 1000Hz

Any other suggestions?

olikraus, 2019-01-20:
Your low level code might be wrong

v1993, 2019-01-20:
Hmm, I have a theory: probably `i2c_master_write(u8x8->user_ptr, arg_ptr, arg_int, true);` don't copy memory but only store pointer? In this case it may point to invalid memory at time of transaction.

Only a theory…

olikraus, 2019-01-20:
user_ptr just is a pointer value, which you provide. It does not copy anything (not sure if this helps)

v1993, 2019-01-20:
Yep, I know. I store transaction pointer there.

Just checked source code of that `i2c_master_write`, looks like it doesn't copy memory. So I really probably am right…

Is there any limit on how many writes may go in one I2C transaction?

v1993, 2019-01-20:
Got HW I2C working :smile:

I copy incoming data into dynamically mallocated memory and then free it after sending. It have pretty big perfomance impact.

Are I2C transactions limited to some size? Not "write" calls to I2C callback, but I2C transactions themselves. This way it would be possible to allocate buffer only once.

Also, how many times per transaction "write" callback can be called?

Thank you.

v1993, 2019-01-21:
Aaand my luck just ran out: SPI capable diplay isn't working :(

I'll have to order another one and wait long time before will be able to test it.

Are there ESP32 and SPI users?

v1993, 2019-01-25:
Hi, @olikraus, it's me again. Can you, please, provide some hints on codebuild's structure and functions?  I want to try to re-write it in a bit more readable manner.

Thank you.

olikraus, 2019-01-27:
what exactly do you want to know?

v1993, 2019-01-27:
What files are generated by it?

What different structures does?

olikraus, 2019-01-27:
> What files are generated by it?

The following files are generated:
u8g2_d_memory.c
u8g2_d_setup.c
The following files are modified:
u8g2.h
U8g2lib.h
U8x8lib.h
u8g2setupc.md
u8g2setupcpp.md
u8x8setupc.md
u8x8setupcpp.md

> What different structures does?

struct interface: Describes the communication interfaces like SPI etc
struct controller: Describes the controller and the supported displays for this controller

olikraus, 2019-04-03:
i think this can be closed

v1993, 2019-04-05:
Hi @olikraus, I'm back! :smiley:
Finally recivied a working SPI display and got some time to expirement with it. I'm getting weird glitched image and have no idea what could be reason behind it (transaction content itself seems fine). Can you take a look at this video, please? https://youtu.be/Vzd91BQ8rZw

I run following things every frame:

    u8g2.setDrawColor(1);
    u8g2.drawBox(0, 0, 255, 127);
    u8g2.sendBuffer();

Similliar results when I draw some text.

Program works 100% percent correctly when I use I²C. Do you have an idea what could be happening? I don't think that it is garbage memory because I've double-checked things there with hex dumping and everything seems fine.

olikraus, 2019-04-05:
Hmmm... Did you enable 16 bit mode? See FAQ.

v1993, 2019-04-05:
I'm pretty sure I did. Also, why would it work in I²C and fail in SPI because of 8bit mode anyways?

v1993, 2019-04-05:
I think that it is most probably some wrong setting for SPI driver.

As a sidenote, is it ok to let driver handle CS line in auto mode? ESP-32 SPI driver is capable of managing it and I use such ability.

Here's my driver code for reference:

```C++
static uint8_t u8x8_byte_espidf_hw_spi_universal(u8x8_t* u8x8, uint8_t msg, uint8_t arg_int, void* arg_ptr, uint8_t host)
// It can work with any of two SPIs
	{
//#ifdef U8X8_HAVE_HW_SPI
	u8x8_device_info* info = u8x8->user_ptr;
	switch (msg) {
			case U8X8_MSG_BYTE_SEND: ;

				// We add stuff into buffer and then send it in one transaction
				//if (info->spi.bufsiz != 0)
				//	hexDump("Old buf", info->spi.buffer, info->spi.bufsiz);
				size_t oldsize = info->spi.bufsiz;
				info->spi.bufsiz += arg_int;
				//hexDump("Appending", arg_ptr, arg_int);
				info->spi.buffer = heap_caps_realloc(info->spi.buffer, info->spi.bufsiz, MALLOC_CAP_DMA); // FIXME: MALLOC_CAP_32BIT may crash stuff
				memcpy(info->spi.buffer + oldsize, arg_ptr, arg_int);
				//hexDump("New buf", info->spi.buffer, info->spi.bufsiz);

				break;
			case U8X8_MSG_BYTE_INIT: ;
				info->type = SPI;
				info->spi.device = NULL;
				info->spi.buffer = NULL;
				info->spi.bufsiz = 0;
				spi_bus_config_t busconf = {};
				busconf.mosi_io_num = u8x8->pins[U8X8_PIN_SPI_DATA];
				busconf.miso_io_num = -1;
				busconf.sclk_io_num = u8x8->pins[U8X8_PIN_SPI_CLOCK];
				busconf.quadwp_io_num = -1;
				busconf.quadhd_io_num = -1;
				busconf.max_transfer_sz = 0; // FIXME: add it into config
				busconf.flags = 0; // FIXME: add stuff here?
				if (u8x8->bus_clock == 0) 	/* issue 769 */
					u8x8->bus_clock = u8x8->display_info->sck_clock_hz;
				/* disable chipselect */
				//u8x8_gpio_SetCS(u8x8, u8x8->display_info->chip_disable_level);

				ESP_ERROR_CHECK(spi_bus_initialize(u8x8_spibuses[host], &busconf, host + 1)); // NOTE: user can set up it manually

				spi_device_interface_config_t conf = {};
				conf.command_bits = 0;
				conf.address_bits = 0;
				conf.dummy_bits = 0; // FIXME: Hmmm, may cause problems?
				conf.mode = u8x8->display_info->spi_mode;
				conf.duty_cycle_pos = 0; // I'm not sure
				conf.clock_speed_hz = u8x8->bus_clock; // TODO: add setting to overclock displays :)
				conf.input_delay_ns = 0;
				conf.spics_io_num = u8x8_GetPinIndex(u8x8, U8X8_MSG_GPIO_CS);
				conf.flags = SPI_DEVICE_POSITIVE_CS | SPI_DEVICE_NO_DUMMY;
				if (u8x8->display_info->chip_enable_level) conf.flags |= SPI_DEVICE_POSITIVE_CS; // FIXME: may be right in complete reverse
				conf.queue_size = 1; // I don't quite get it, but ok…
				spi_bus_add_device(u8x8_spibuses[host], &conf, &(info->spi.device)); // FIXME: remove in destructor

				break;

			case U8X8_MSG_BYTE_SET_DC:
				u8x8_gpio_SetDC(u8x8, arg_int);
				break;

			case U8X8_MSG_BYTE_START_TRANSFER:

				// Nothing to do here
				break;

			case U8X8_MSG_BYTE_END_TRANSFER: ;
				// Docs says that it's better to have 32-bit aligned memory
				// Doing it here should make stuff faster
				//info->spi.buffer = heap_caps_realloc(info->spi.buffer, info->spi.bufsiz, MALLOC_CAP_DMA);
				spi_transaction_t transaction = {};
				transaction.flags = 0;
				transaction.length = (info->spi.bufsiz) * 8; // Bits
				transaction.tx_buffer = info->spi.buffer;
				transaction.rx_buffer = NULL; // U8g2 don't care about feedback, bad guys!
				spi_device_polling_transmit(info->spi.device, &transaction);
				free(info->spi.buffer);
				info->spi.buffer = NULL;
				info->spi.bufsiz = 0;

				break;
			default:
				return 0;
			}

	return 1;
	}
```
P.S.: can you add destructor event, please? Something like `U8X8_MSG_DESTROY` passed to GPIO callback, being ignored by anyone who don't have to clear stuff and handled by ones who do. I really need it!

olikraus, 2019-04-05:
> P.S.: can you add destructor event, please?

There is already a default destructor: https://github.com/olikraus/u8g2/issues/815

> Here's my driver code for reference:

Maybe there is something wrong in your SPI code. At least it looks complicated to me.

v1993, 2019-04-05:
>There is already a default destructor: #815

That destructor story is TOTALLY different. That person just had no understanding of basic C++ concepts, including fact that destructors are called automatically by language. I, on other hand, what to have C-side destructor (or cleanup function which is same thing called the other way) which could be called eiter manually (if using C interface, like `u8x8_destroy(u8x8);`) or automatically by C++ (declare C++ destructor which calls C one).

> At least it looks complicated to me.

Well, most of code is just creating various structures if you look closely. The rest is maintaining simple buffer which gets extended/allocated by `heap_caps_realloc`, which is pretty much same as `realloc` except for last argument.

v1993, 2019-04-05:
Just figured out faulty string:

    conf.spics_io_num = u8x8_GetPinIndex(u8x8, U8X8_MSG_GPIO_CS);
Replaced it with

    conf.spics_io_num = u8x8_GetPinValue(u8x8, U8X8_MSG_GPIO_CS);
Still face the same issue tho :\

olikraus, 2019-04-05:
> That destructor story is TOTALLY different. 

Well, I can only say this for u8g2: If you want me to create a destructor, then the question is what shell I do inside the u8g2 destructor? Or in other words: How should the destructor differ from the default destructor which is already there.

> Well, most of code is just creating various structures if you look closely.

My motivation is not that high, to look into this closely. I mean, my own code works correctly. Your code makes use of many  other functions, which I do not know. And: I already looked into it closely: For example you call `heap_caps_realloc` but then it is released with `free`. Is this correct? I do not know this, at least it sounds strange.

v1993, 2019-04-06:
As for destructor thing, it's probably easier to implement and make pull request rather than describe (yeah, I sometimes suck at describing :disappointed:).

> And: I already looked into it closely: For example you call `heap_caps_realloc` but then it is released with `free`. Is this correct? I do not know this, at least it looks strange.

It is correct. According to [ESP-IDF docs](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/system/mem_alloc.html#_CPPv414heap_caps_freePv):

> `void heap_caps_free(void *ptr)`
> Free memory previously allocated via `heap_caps_malloc()` or `heap_caps_realloc()`.
> …
> In IDF, `free(p)` is equivalent to `heap_caps_free(p)`.

But it's probably better ro replace it with caps version for better readability and potentially better compatiblity with futher IDF versions.

ESP-IDS SPI driver use 32bit blocks and say to handle non-32bit ones in some way unclear for me. When I do so, however, display doesn't show anything at all. Aligning memory to 32bits do nothing too.

P.S.: can you reopen this, please? I don't feel that comfortable discussing problems in closed issue :).

[86 more replies]