olikraus/u8g2 #790: ERC12864-1 (buydisplay.com) contrast issue - fixed

i5mile, 2019-01-19:
Hi,

I used the u8g2 lib for ERC12864-1  from buydisplay.com.  I found that the contrast wasn't right. The pixels which is supposed to be OFF had a faint dark color. I tweaked the init sequence and made a couple of changes as shown below. (Pls look for *** in the comments where the change was made)

static const uint8_t u8x8_d_st7565_erc12864_init_seq[] = {
    
  U8X8_START_TRANSFER(),             	/* enable chip, delay is part of the transfer start */
  
  U8X8_C(0x0e2),            			/* soft reset */
  U8X8_C(0x0ae),		                /* display off */
  U8X8_C(0x040),		                /* set display start line to ... */
  
  U8X8_C(0x0a0),		                /* ADC set to reverse */
  U8X8_C(0x0c8),		                /* common output mode */
  //U8X8_C(0x0a1),		                /* ADC set to reverse */
  //U8X8_C(0x0c0),		                /* common output mode */
  // Flipmode
  // U8X8_C(0x0a0),		                /* ADC set to reverse */
  // U8X8_C(0x0c8),		                /* common output mode */
  
  U8X8_C(0x0a6),		                /* display normal, bit val 0: LCD pixel off. */
  U8X8_C(0x0a2),		                /* LCD bias 1/9 - *** Changed by Ismail - was 0xa3 - 1/7 bias we were getting dark pixel off */
  U8X8_C(0x02f),		                /* all power  control circuits on (regulator, booster and follower) */
  U8X8_CA(0x0f8, 0x000),		/* set booster ratio to 4x (ST7567 feature)*/
  U8X8_C(0x027),		                /* set V0 voltage resistor ratio to max  */
  U8X8_CA(0x081, 0x05),		       /* set contrast, contrast value, EA default: 0x016 - *** Changed by Ismail to 0x05 */ 
  
  U8X8_C(0x0ae),		                /* display off */
  U8X8_C(0x0a5),		                /* enter powersafe: all pixel on, issue 142 */
  
  U8X8_END_TRANSFER(),             	/* disable chip */
  U8X8_END()             			/* end of sequence */
};

Basically, the LCD bias was set to 1/7 (0x0a3) which was not right for the display. It should be 1/9 (0xa2). Further, I have to play around with the contrast setting. 0x05 works fine on my display.
With these changes I can get good crisp pixels.

I though this might help someone out there...

Thanks,
-Ismail

olikraus, 2019-01-19:
thanks for the feedback. Maybe I should create an alternative version for this constructor...

olikraus, 2019-01-20:
device: erc12864_alt

olikraus, 2019-01-20:
I have created beta release 2.25.4

new constructor starts with U8G2_ST7565_ERC12864_ALT

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

olikraus, 2019-01-20:
Can you test this? Thanks.

i5mile, 2019-01-20:
Hi Olikraus,

Sorry, I don't use C++/Arduino environment. I used your lib in plain old C.
I will test the changes but it would take a while for me to setup.

olikraus, 2019-02-09:
no further feedback, closing...

closed 2019-02-09 by olikraus