olikraus/u8g2 #791: Chopping second line of text on 1327 with esp32 dev board greekeit.

DivinerGregg, 2019-01-20:
ssd 1327 is cutting the second line of simple text. I have tried several constructors. 

Link to what I bought. 
https://www.amazon.com/gp/product/B07FHLFQWB/ref=ppx_yo_dt_b_asin_title_o08__o00_s00?ie=UTF8&psc=1

They recommended 
U8G2_SSD1327_EA_W128128_1_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE); 

[image]

``

//U8G2_SSD1327_SEEED_96X96_F_SW_I2C u8g2(U8G2_R0, /* clock=*/ SCL, /* data=*/ SDA, /* reset=*/ U8X8_PIN_NONE);  // Seeedstudio Grove OLED 96x96
//U8G2_SSD1327_SEEED_96X96_F_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE); // Seeedstudio Grove OLED 96x96
//U8G2_SSD1327_EA_W128128_F_SW_I2C u8g2(U8G2_R0, /* clock=*/ 5, /* data=*/ 4, /* reset=*/ U8X8_PIN_NONE);
//U8G2_SSD1327_EA_W128128_F_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);

//-------------tested------------------
//U8G2_SSD1327_EA_W128128_1_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE); //from amazon didn't work
//   "1" is amazon recommended                                       clock, data
// U8G2_SSD1327_EA_W128128_F_HW_I2C u8g2_P(U8G2_R0,U8X8_PIN_NONE,   22,   21);
//U8G2_SSD1327_EA_W128128_F_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE, 22, 21); //clock and data  worked RESULT: buffer sucks  "F" is not from amazon
//U8G2_SSD1327_MIDAS_128X128_2_HW_I2C u8g2(U8G2_R0, (/* reset=/ U8X8_PIN_NONE, (/ clock=/ SCL, / data=*/ SDA));
//U8G2_SSD1327_EA_W128128_1_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE, 22,21);   //vendor recommendation *added clock&data RESULT:worked but some garbage, line 2 not working  

#include <Arduino.h>
#include <U8g2lib.h>
#include <Wire.h>     // i2c

U8G2_SSD1327_MIDAS_128X128_2_HW_I2C u8g2(U8G2_R0, (/* reset=/ U8X8_PIN_NONE, (/ clock=/ SCL, / data=*/ SDA));

void setup(void) {
  u8g2.begin();
}

void loop(void) {
  u8g2.clearBuffer();					// clear the internal memory
//  u8g2.setFont(u8g2_font_ncenB08_tr);	// choose a suitable font
  u8g2.setFont(u8g2_font_6x10_tf);
  
  u8g2.drawStr(0,10,"Hello World!");	// write something to the internal memory
  u8g2.drawStr(30,20,"Line 2");  // write something to the internal memory
  u8g2.sendBuffer();					// transfer internal memory to the display
  
  delay(1000);

olikraus, 2019-01-20:
use  `U8G2_SSD1327_EA_W128128_F_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE)`
with **`_F_`**.

DivinerGregg, 2019-01-20:
Thought I tried that one. Anyways, it works. Thanks.