olikraus/u8g2 #792: U8G2 and SD

froussel, 2019-01-21:
Hi and happy new year,

With Arduino IDE and ESP8266
[code]

#include <Arduino.h> 
#include <U8g2lib.h>
#include <SPI.h> 
#include <Wire.h> 
#include <TinyGPS++.h> 
#include <SoftwareSerial.h> 
#include <SD.h>

/* ESP8266         SD(adaptateur sd --> microsd)
 *
 * D5         ---     CLK
 * D6         ---     MISO
 * D7         ---     MOSI
 * 3.3v                3.3v
 * GNd                GND
 * D1         ---     CS
 * 
 * 
 * testscreen LCD UC1701x 
 * 
 *      Screen      ESP8266
 *
 * 5    LedA         NC
 * 6    Vss           GND
 * 7    Vdd          3.3v
 * 8    SCLK        D5 SCK
 * 9    SDA          D7 MOSI
 * 10   RS           D2 dc
 * 11   Reset       D3 reset
 * 12   CS           D8  
 */

U8G2_UC1701_MINI12864_F_4W_SW_SPI u8g2(U8G2_R2, /* clock=*/ D5, /* data=*/ D7, /* cs=*/ D8, /* dc=*/ D2, /* reset=*/ D3);  

const int CS_SD = D1;               // for SD

void setup(void)     
{ u8g2.begin();  
  u8g2.firstPage();
  do {  u8g2.setFont(u8g2_font_courB14_tr); 
        u8g2.drawStr(1,15,"Ecran OK");
     } while ( u8g2.nextPage() );
  Serial.begin (115200); 
  Serial.println("   ");
  Serial.println(" Initialisation écran passée  ");
  delay(1000);
  Serial.println("Initialisation SD card...");
  if (!SD.begin(CS_SD))                         // SD detectee ? 
    {   Serial.println("Pas de SD détectée");
         u8g2.firstPage();
         do {  u8g2.setFont(u8g2_font_courB14_tr); 
               u8g2.drawStr(1,1,": Pas de SD");
            } while ( u8g2.nextPage() );
         Serial.println("Ici : rien n'a changé sur l'écran ");
         return;            }
}

void loop(void) 
{ 
}

[/code]

The second screen on the UC1701x is not displayed
The first screen remains displayed

I suspect a pb between the u8g2 library and SD library

A solution ? Please...

v1993, 2019-01-21:
Please, format your code as github needs it: use ```.

That will make understanding a lot easier.

froussel, 2019-01-21:
`#include <Arduino.h>
#include <U8g2lib.h>
#include <SPI.h>
#include <Wire.h>
#include <TinyGPS++.h>
#include <SoftwareSerial.h>
#include <SD.h>

/* ESP8266 SD(adaptateur sd --> microsd)
*

D5 --- CLK
D6 --- MISO
D7 --- MOSI
3.3v 3.3v
GNd GND
D1 --- CS
testscreen LCD UC1701x
 Screen      ESP8266
5 LedA NC
6 Vss GND
7 Vdd 3.3v
8 SCLK D5 SCK
9 SDA D7 MOSI
10 RS D2 dc
11 Reset D3 reset
12 CS D8
*/
U8G2_UC1701_MINI12864_F_4W_SW_SPI u8g2(U8G2_R2, /* clock=/ D5, / data=/ D7, / cs=/ D8, / dc=/ D2, / reset=*/ D3);

const int CS_SD = D1; // for SD

void setup(void)
{ u8g2.begin();
u8g2.firstPage();
do { u8g2.setFont(u8g2_font_courB14_tr);
u8g2.drawStr(1,15,"Ecran OK");
} while ( u8g2.nextPage() );
Serial.begin (115200);
Serial.println(" ");
Serial.println(" Initialisation écran passée ");
delay(1000);
Serial.println("Initialisation SD card...");
if (!SD.begin(CS_SD)) // SD detectee ?
{ Serial.println("Pas de SD détectée");
u8g2.firstPage();
do { u8g2.setFont(u8g2_font_courB14_tr);
u8g2.drawStr(1,1,": Pas de SD");
} while ( u8g2.nextPage() );
Serial.println("Ici : rien n'a changé sur l'écran ");
return; }
}

void loop(void)
{
}`

froussel, 2019-01-21:
`
#include <Arduino.h>
#include <U8g2lib.h>
#include <SPI.h>
#include <Wire.h>
#include <TinyGPS++.h>
#include <SoftwareSerial.h>
#include <SD.h>

/* ESP8266 SD(adaptateur sd --> microsd)
*

D5 --- CLK
D6 --- MISO
D7 --- MOSI
3.3v 3.3v
GNd GND
D1 --- CS
testscreen LCD UC1701x
 Screen      ESP8266
5 LedA NC
6 Vss GND
7 Vdd 3.3v
8 SCLK D5 SCK
9 SDA D7 MOSI
10 RS D2 dc
11 Reset D3 reset
12 CS D8
*/
U8G2_UC1701_MINI12864_F_4W_SW_SPI u8g2(U8G2_R2, /* clock=/ D5, / data=/ D7, / cs=/ D8, / dc=/ D2, / reset=*/ D3);

const int CS_SD = D1; // for SD

void setup(void)
{ u8g2.begin();
u8g2.firstPage();
do { u8g2.setFont(u8g2_font_courB14_tr);
u8g2.drawStr(1,15,"Ecran OK");
} while ( u8g2.nextPage() );
Serial.begin (115200);
Serial.println(" ");
Serial.println(" Initialisation écran passée ");
delay(1000);
Serial.println("Initialisation SD card...");
if (!SD.begin(CS_SD)) // SD detectee ?
{ Serial.println("Pas de SD détectée");
u8g2.firstPage();
do { u8g2.setFont(u8g2_font_courB14_tr);
u8g2.drawStr(1,1,": Pas de SD");
} while ( u8g2.nextPage() );
Serial.println("Ici : rien n'a changé sur l'écran ");
return; }
}

void loop(void)
{
}
`

olikraus, 2019-01-21:
which second display?

froussel, 2019-01-21:
Excuse my english
I have only one screen, the display of u8g2.drawStr(1,15,"Ecran OK");
is correct it is the second display that should clear then display
u8g2.drawStr(1,1,": Pas de SD");
which is not done

olikraus, 2019-01-23:
```
u8g2.drawStr(1,1,": Pas de SD");
```
by default, the reference point for the chars is the lower left edge. So the text will be above the visible part of the display. Use
```
u8g2.drawStr(1,20,": Pas de SD");
```
instead.

v1993, 2019-01-24:
Again, use ``` on both sides of code to format it better way. Just an advice.

olikraus, 2019-01-24:
> It seems that every time we ask an SPI device an action we have to redo an order before

ok, understand. But at this point, there is nothing much what i can do, I just use the SPI/I2C subsystem in the way as it has been adviced by the Arduino people. At least I hope, that SD still will work with u8g2. 

Maybe I should test the cooperation of SD card and u8g2 once.... but i will test this with my own Arduino Uno not for ESP8266. Infact the ESP8266 HAL might have a problem here also...

froussel, 2019-01-24:
Thanks
please see : [http://forum.arduino.cc/index.php?topic=587880.0](url)
May be see SPI/I2C subsystem's men
My english and my C++ are too poor :-))

olikraus, 2019-04-05:
looks like I missed this issue. Nevertheless, I can not say much here. U8g2 sticks to the Arduino SPI API suggestions. If it doesn't work with SD then it may not be the fault of u8g2.

closed 2019-05-19 by olikraus

olikraus, 2019-11-23:
Just for reference: I have created a demo for reading and writing XBM data from and to SD Card: https://github.com/olikraus/u8g2/blob/master/sys/arduino/u8g2_page_buffer/LoadFromSD/LoadFromSD.ino

fikretakarsu, 2020-06-01:
selam
u8g2 ile  sd kart daki dosya isimlerini listelemek ve seçilen dosyadan veri yüklemesi yapmaya çalıştım başaramadım yardım rica ederim