olikraus/u8g2 #793: SSD1327 128x128 form waveshare no lines

Andrzej1000, 2019-01-21:
Hello.
SSD1327 128x128 form waveshare.
Arduino DUE does not display the top and bottom some lines.

"
#include <U8g2lib.h>
#include <Wire.h>
U8G2_SSD1327_EA_W128128_1_SW_I2C u8g2(U8G2_R0, /* clock=*/ A1, /* data=*/ A0, /* reset=*/ U8X8_PIN_NONE);
"

It looks like it did not light up the first  about 20 and  the last about 20 lines (I did not count exactly).
Please, help me display all 128 lines?
What to check ...

olikraus, 2019-01-21:
what about the other constructor?
`U8G2_SSD1327_MIDAS_128X128_1_SW_I2C`

Andrzej1000, 2019-01-21:
Thank you very much!
It works like that:
U8G2_SSD1327_MIDAS_128X128_1_SW_I2C u8g2(U8G2_R0, /* clock=*/ A1, /* data=*/ A0, /* reset=*/ U8X8_PIN_NONE);
and
U8G2_SSD1327_MIDAS_128X128_1_2ND_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);

Thank you - you can close the topic
;)

closed 2019-01-21 by Andrzej1000

olikraus, 2019-01-22:
:)