olikraus/u8g2 #794: Using u8g2 and SPI

deadbird99, 2019-01-23:
Hi. 

I'm using a ST7920 SPI display, coupled with 8 MCP41010 SPI potentiometers.

When I use u8g2 do draw on my display only, everything's fine.
When I set my potentiometers using SPI, it's ok too.

But when I try to do both, u8g2 can't draw anymore.

Is there a way to place an ST7920 and the MCP41010 on the same SPI line and make them work altogether using u8g2 (for the display) and regular SPI functions for the MCPs?

Thanks a lot for your help.

AnHardt, 2019-01-24:
The latest [ST7920 datasheet](http://www.hpinfotech.ro/ST7920.pdf) page 26 second paragraph, tells us CE is broken, don't use it with together with other devices on the bus.

Mostly only a garbed display is observed, but depending on what is send to the other devices worse thing can happen.

Safest is to ether use a separate spi-bus for the ST7920 or use some AND-gates to keep SCLK and SI quite when CE is not selected.

AnHardt, 2019-01-24:
The datasheet describes a, compared to other SPI devices, very different (broken) behavior of the controllers CE pin. In former versions of the datasheet a normal  behavior was described.

deadbird99, 2019-01-24:
And there is no way to patch this behavior in the code?

AnHardt, 2019-01-24:
Maybe - the day we can fix a flat tire with a firmware update.

deadbird99, 2019-01-24:
See #767, exactly the same issue.

AnHardt, 2019-01-24:
Hardware reset, reinitialize, reprint could work. Never tried. Up to now reprint was enough.

deadbird99, 2019-01-24:
Found the solution in #767. Using HW SPI in beta 2.25.2 solves the issue.

The constructor:
```C++
U8G2_ST7920_128X64_F_HW_SPI u8g2(U8G2_R0, 30, U8X8_PIN_NONE);
```

closed 2019-01-24 by deadbird99

olikraus, 2019-01-24:
I like "self-solving" issues. :-)