olikraus/u8g2 #796: Allow each display to have its own buffer

futureshape, 2019-01-27:
I am driving multiple displays from the same Arduino-compatible board and would like to have a way to have a separate buffer for each display. 

The reason is that each display shows a dynamically generated animation with only part of the picture changing in each re-draw. In the current situation, making updates to one display also overwrites what's displayed on the other one.

Current alternatives are:

1. Clear the display buffer every time - this means I'll need to re-create the previous frame of the animation, which is computationally expensive
2. Keep a copy of the buffer for each display in my code - will result in always having 1 more buffer than needed, and unnecessary back-and-forth copying of memory
3. Solution previously discussed in #540, but is a temporary work-around that requires manual modification of the library code

Possible fixes: 

1. In display setup constructor, set a flag that you want the display to allocate its own memory instead of sharing a buffer. This would get passed all the way to the functions in `u8g2_d_memory.c`, who could return a dynamically allocated buffer instead of a static one. Drawback: a lot of code updates to add this to every constructor
2. Implement a `U8G2::setBufferPtr(uint8_t *buf)` that allows the user to allocate their own buffer and pass it to the library, essentially changing the pointer stored internally. 

I can try to implement the second solution and open a pull request unless you see any potential drawbacks/side effects?

olikraus, 2019-01-27:
Users already complain about RAM and FLASH size with u8g2, so I do not want to grow the lib without important reason. I guess 99.9% of all users will drive only one display, so I assume the current concept is good enough.

I see two options:
1. Store the content of the buffers externally. You can get the address of the internal buffer with the direct access API. You could backup and restore the content.
2. We add dynamic memory allocation, but encapsulate everything in #ifdefs

futureshape, 2019-01-27:
Agree, it's very much an edge case, and it only makes sense in MCUs with a lot of memory (I'm using it with ESP8266). 

I've managed to implement a `U8G2::setBufferPtr(uint8_t *buf)` method which seems to work OK, so I'll raise a pull request.

v1993, 2019-01-27:
@futureshape Please, implement it on C side and let C++ only call it on internal object.

futureshape, 2019-01-28:
@olikraus quick question - where is the best place to raise a pull request so it eventually propagates to the Arduino library?

Should I do it in https://github.com/olikraus/u8g2 or in https://github.com/olikraus/U8g2_Arduino or both?

olikraus, 2019-01-28:
This project would be correct (u8g2)

olikraus, 2020-01-18:
this is probably solved by #1052 (and others), so closing this.

closed 2020-01-18 by olikraus