olikraus/u8g2 #797: Collision detection?

darkoromanov, 2019-01-28:
Hi,

does the library provide a way to check collisions between drawn pixels? In other words, is there a simple way to check if 2 or more bitmaps are rendered on the same pixels?

I'm developing a simple retro game with an Arduino Nano and I have to efficiently check when 2 shapes (bitmaps) collide. Thanks.

darkoromanov, 2019-01-28:
Hi Oliver,

thanks for the prompt response.  I'll keep the bounding boxes methods as second option. 

There is also a third way which is bounding boxes plus pixel-by-pixel check to make the give exact precision to the detection. In this case the pixel values would be taken from the bitmaps.

The full buffer way sounds interesting, I'm checking your examples but I'm not sure what to look for exactly. I guess I should have a buffer in memory representing the state of the screen that is going to be shown, is it correct? Then I could use it as a mask with the current buffer and check for 1s, correct? So, is there a function to get the current buffer?

olikraus, 2019-01-28:
> So, is there a function to get the current buffer?

Yes, the buffer is already allocated by u8g2 and you can also get the pointer to the same: https://github.com/olikraus/u8g2/wiki/u8g2reference#getbufferptr

I have once added an example, where I play around with the internal buffer:

https://github.com/olikraus/u8g2/blob/master/sys/arduino/u8g2_page_buffer/DirectAccess/DirectAccess.ino

darkoromanov, 2019-01-29:
Thanks Oliver!

I hope I'll be able to do that, 20 years of coding with Java, PHP, c# and all sort of high level languages and yet I feel lost with pointers and low level stuff, what a n00b :)

olikraus, 2019-01-29:
hehe opposite with me. Started with C more than 20 years back. Did lot of C++, Java, JavaScript, but still miss pointers and do not trust the automatic memory deallocation...

closed 2019-04-05 by olikraus