olikraus/u8g2 #799: [Font request] new font q_pixia for persian+arabic unicode

IMAN4K, 2019-02-01:
Hi @olikraus 

I'm just trying to add `q_pixia`:
```
./otf2bdf -r 72 -p 14 q_pixia.ttf -o q_pixia_14.bdf
./bdfconv -f 1 -m '32-128,$600-$6ff,$FB50-$FBB1,$FE70-$FEFF' -n q_pixia_14 -o out/q_pixia_14.c fonts/q_pixia_14.bdf
```
But there's a problem, the text is not shown on LCD while the same text with [`samim_14`](https://github.com/olikraus/u8g2/wiki/fntgrppersian#samim_14) is shown properly!
I don't know where is the problem ?!
Wha do you think?

[q_pixia_v1_01.zip](https://github.com/olikraus/u8g2/files/2820024/q_pixia_v1_01.zip)
[q_pixia_14_c.zip](https://github.com/olikraus/u8g2/files/2820042/q_pixia_14_c.zip)

olikraus, 2019-02-01:
I have used the following commands for the conversion:
```
	otf2bdf -a -r 72 -p 13 -o tmp.bdf q_pixia.ttf
	bdfconv -v -f 1 -m '32-65534' tmp.bdf -o u8g2_font_q_pixia.c  -n u8g2_font_q_pixia
```
this is the generated c-code:
[u8g2_font_q_pixia.txt](https://github.com/olikraus/u8g2/files/2823191/u8g2_font_q_pixia.txt)

here is a picture of the converted font:

[image]

IMAN4K, 2019-02-02:
@olikraus 
Thanks and Regards

closed 2019-02-02 by IMAN4K