olikraus/u8g2 #803: Pimoroni 128x128 OLED SH1107 offset

jjgarciajz, 2019-02-03:
Hello,
First of all thanks for the amazing work with this library :)

I am working with a [Pimoroni 128x128 OLED breakout board](https://shop.pimoroni.com/products/1-12-oled-breakout) based on the SH1107 chip connected to a SparkFun Pro Micro 5v. After my first test I noticed the display show quite big X offset.

After reading the issue #501 I've discovered this board doesn't require the 96 pixels X offset that exists in the u8x8_d_sh1107.c file. 

So, after changing u8x8_d_sh1107.c lines 374 and 375 to:
  /* default_x_offset = */ 0,
  /* flipmode_x_offset = */ 0,
The display shows contents properly. 

I guess I should notify this :) 
Kind regards.

olikraus, 2019-02-09:
Thanks for the information.

New constructor: 
U8G2_SH1107_PIMORONI_128X128_1_HW_I2C u8g2(U8G2_R0, /* reset=*/ 8);

I have created a beta release for this. Can you test the new constructor? Will it work?

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

olikraus, 2019-02-09:
this is part of 2.25 now, but will be moved to 2.26 for testing

olikraus, 2019-02-13:
Any feedback?

jjgarciajz, 2019-02-13:
Hi, sorry for the delay. I have been busy with work this week. I will test
the new constructor this weekend and give you proper feedback.
Thanks and kind regards.

olikraus, 2019-02-14:
No problem... take you time...

jjgarciajz, 2019-02-16:
Hello, sorry again for the delay.
I've tested the new constructor with sample code provided (graphicstest and spacetrash) and it works perfectly. 
If you need me to test anything else more specific just ask.
Thanks!

olikraus, 2019-02-16:
Thanks for testing.
Will it work with the u8g2 page buffer FlipMode.ino?
The frame should not change and should be visible all the time.
Thanks a lot!

jjgarciajz, 2019-02-16:
The FlipMode.ino code seems to work. Frame is continuously shown and text turns.

Probably is not related (I'm quite new to this) but I've found out any attempt to show Unicode characters it shows the ? character instead. 

Please check this short video:
https://youtu.be/Qx4R-vCw4i4

I'm using a SparkFun Pro Micro 5v/16Mhz board.

Regards!

olikraus, 2019-02-17:
Thanks a lot for the video. I the flipmode example works perfect. Regarding the "?" problem: Which IDE do you use? Which Version of the IDE?

jjgarciajz, 2019-02-17:
I use Arduino IDE v1.8.8 (the latest I think). The problem with unicode characters happened to me with the stable library version too, even before I posted this issue.

olikraus, 2019-02-17:
hmm strange. I have IDE v1.8.4 here and unicode works without problems with the internal editor. Maybe I should check a later version.

jjgarciajz, 2019-02-17:
Quite strange indeed. I've checked the code of graphicstest and the other unicode examples and the Arduino IDE shows correctly the unicode characters.
I've also tried other uC boards (Arduino Nano, Arduino Uno) with same results.

Do I need to install anything else apart of the library itself? (like fonts, etc...) Maybe I'm skipping something.

Just in case, I will try 1.8.4 and write you back as soon as possible.

olikraus, 2019-02-17:
No extra software is required. It should work out of the box.

AnHardt, 2019-02-17:
Try saving and reloading the sketch.

jjgarciajz, 2019-02-17:
@AnHardt that did the trick!

Before I saw AnHardt's response, I was trying to upload the sketch with my iMac + Arduino IDE 1.8.8 (normally I use a Surface Pro 4 to "play" with UCs). The unicode characters were shown properly this time. I thought it was some issue with the Surface.

Then I tried the Surface Pro again, saving the modified sketch as a new project before uploading, and worked perfectly.

Thanks for the help :)

jjgarciajz, 2019-02-17:
Confirmed the unicode "issue" and the solution provided testing on a third Windows PC. 
Apparently on Windows, example sketches that make use of unicode characters need to be saved as a new sketch to upload correctly.
This doesn't seem to happen on Mac or Linux, though.

olikraus, 2019-02-18:
Hmmm... Never had such a problem on Linux...

jjgarciajz, 2019-02-18:
Yes, as I mentioned, the issue only affects to Windows.

closed 2019-04-05 by olikraus