olikraus/u8g2 #805: No BitmapP() Function

Catch2223, 2019-02-05:
In u8glib there use to be a bitmapP() function that would allow for bitmaps to saved in the program memory and it really helped in saving some dynamic memory for larger images. It would be great if this function could be brought to u8g2lib because when dealing with bitmaps its pretty easy to start to run out of memory.

I'm actually requesting this because I am using an Arduino Pro Micro and for some reason with my sketch at 88% dynamic memory used it bricks the bootloader and I have to reflash it if I want to program over USB. If I could save bitmaps to program memory I would be much better off. Any input would be great!

Joennuh, 2019-02-05:
I think you are searching for drawXBM (dynamic memory) and drawXBMP (PROGMEM). Documentation can be found on  https://github.com/olikraus/u8g2/wiki/u8g2reference#drawxbm 
Am I correct?

olikraus, 2019-02-05:
I know the format is different between drawBitmapP and drawXBMP, but during transition from U8glib to U8g2 i removed drawBitmapP. Will be drawXBMP ok?

closed 2019-04-05 by olikraus