olikraus/u8g2 #807: Need to add a TTF font to U8G2

dominicklee, 2019-02-06:
Hi Oliver,

I have a weather icons font that I want to integrate in the U8G2. Could you please convert the attached "Climacons.ttf" file so that it can be used in U8G2 and show me how to install it?

[Climacons.zip](https://github.com/olikraus/u8g2/files/2838655/Climacons.zip)

Can you also show me how to convert such TTF fonts for U8G2 in Windows? Thanks!

olikraus, 2019-02-07:
it needs to be converted to bdf first. For this I need to know the desired pt size. 64pt and 128pt look good, but the icons will be large.

dominicklee, 2019-02-07:
Can you make a 28pt font (for 5 rows of icons on a 128x128 screen) and a 36pt font (for 2 rows of icons)?

dominicklee, 2019-02-07:
For example, like this:

[image]

olikraus, 2019-02-09:
I need to use the pt argument for the conversion tools. I do not know what pt=64 means and what the resulting height will be, but maybe the font author should know the minimal pt size for scan line conversion.

olikraus, 2019-02-09:
with pt=64 the result looks like this:
[image]

So the hight will be around 30 pixel. The quality will be very poor with any value blow 64.

olikraus, 2019-02-09:
same picture with pt=32. You can see that the thermometer is broken:
[image]

olikraus, 2019-02-09:
hmm... I think the icons are from here: http://adamwhitcroft.com

olikraus, 2019-02-09:
@AdamWhitcroft what would be the prefered pt size at 72 dpi to convert the icons to bitmaps?

closed 2019-04-05 by olikraus