olikraus/u8g2 #808: different thickness of chevron in open_iconic_*_1x

tobiea27, 2019-02-08:
Hi,
I'm currently using the open_iconic_*_1x font and I noticed a asymmetric thickness/size of the chevrons. The up and left facing ones are smaller/slimmer. This doesn't look very great on the screen.

Image:
[image]

In the bigger fonts they seems fine.

AnHardt, 2019-02-08:
That's completely normal for automatically converted fonts (vector to pixel). The smaller the worse.
Manually reworking the .bdf-fonts is the only way to come out of this.

olikraus, 2019-02-09:
Yes, indeed. I also decided not to modify any fonts. You could try to contact the original font author for this problem.

tobiea27, 2019-02-09:
Okay, thanks for the help.

I only use this 4 symbols as navigation indicators and not the whole font. So, I decided to us a different method. 
Drawing with drawXBMP() is a little bit more straight forward then modifying the font.

Maybe this will help somebody in the future:
```cpp
const unsigned char chevron_up[] = {0x00, 0x42, 0xE7, 0x7E, 0x3C, 0x18, 0x00, 0x00};
const unsigned char chevron_down[] = {0x00, 0x00, 0x18, 0x3C, 0x7E, 0xE7, 0x42, 0x00};
const unsigned char chevron_left[] = {0x10, 0x38, 0x1C, 0x0E, 0x0E, 0x1C, 0x38, 0x10};
const unsigned char chevron_right[] = {0x08, 0x1C, 0x38, 0x70, 0x70, 0x38, 0x1C, 0x08};

enum ICON
{
  ICON_NONE = 0,
  ICON_CHHEVRON_LEFT,
  ICON_CHHEVRON_RIGHT,
  ICON_CHHEVRON_UP,
  ICON_CHHEVRON_DOWN
};

void draw_icon(int x, int y, ICON icon)
{
  switch(icon)
  {
    case ICON_CHHEVRON_UP:
      display.drawXBMP(x, y, 8, 8, chevron_up);
    break;

    case ICON_CHHEVRON_DOWN:
      display.drawXBMP(x, y, 8, 8, chevron_down);
    break;

    case ICON_CHHEVRON_LEFT:
      display.drawXBMP(x, y, 8, 8, chevron_left);
    break;

    case ICON_CHHEVRON_RIGHT:
      display.drawXBMP(x, y, 8, 8, chevron_right);
    break;
  }
}

closed 2019-02-09 by tobiea27

AnHardt, 2019-02-09:
If only using the chevrons, you could extract a tiny, 4 symbols, 'chevron-font' from a nicer looking one.