olikraus/u8g2 #811: need help for interfacing ssd1306 Display

alexiskyriakou, 2019-02-09:
Hello,

I am using Atmel Studio and i am trying to drive an SSD1306 display over SPI.

I have try the example for ATmega328 without success. i am getting the following error
u8x8_avr.h: No such file or directory	

Can anyone provide me any guide on how to use this or any example code which is working?

Thank you,
Alexis

olikraus, 2019-02-09:
will this help? https://github.com/olikraus/u8g2/wiki/u8g2as7

alexiskyriakou, 2019-02-09:
I already try that guide but i am getting the following error

u8x8_avr.h: No such file or directory

olikraus, 2019-02-09:
"u8x8_avr.h" is not required in the example mentioned above. If you follow exactly the guide from above, you will not face the problem.

alexiskyriakou, 2019-03-03:
I have manage to make it work but the framerate (fps) is to slow. An i am using the following

u8g2_Setup_ssd1306_128x64_noname_f( &u8g2, U8G2_R0, u8x8_byte_4wire_sw_spi, u8x8_avr_gpio_and_delay );

Any ideas

alexiskyriakou, 2019-03-03:
Here is my main.c file

https://github.com/alexiskyriakou/AtmelProject/blob/master/TPA3118_AudioAmplifier/TPA3118_AudioAmplifier/main.c

olikraus, 2019-03-03:
I get a "page not found" error by your link

alexiskyriakou, 2019-03-04:
Can you try now???

https://github.com/alexiskyriakou/AtmelProject

olikraus, 2019-03-05:
U8g2 Low Level Code Looks ok

alexiskyriakou, 2019-03-05:
Is there any other way to try to make it faster??

olikraus, 2019-03-05:
Sure, you can use hardware driven SPI instead. 
https://github.com/olikraus/u8g2/wiki/Porting-to-new-MCU-platform#atmel-avr

alexiskyriakou, 2019-03-16:
i am trying to use the Hardware SPI but i can not display any massage on the screen.

I have change my display to a nokia one, with pcd8544 controller but again nothing.

I am using the following code

#include <u8g2.h>
#include <util/delay.h>
#include <u8x8_avr.h>
#include <avr/power.h>

#define CS_DDR DDRC
#define CS_PORT PORTC
#define CS_BIT 1

#define DC_DDR DDRC
#define DC_PORT PORTC
#define DC_BIT 2

#define RESET_DDR DDRD
#define RESET_PORT PORTD
#define RESET_BIT 2

u8g2_t u8g2;

uint8_t
u8x8_gpio_and_delay (u8x8_t * u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr) {
	// Re-use library for delays
	if (u8x8_avr_delay(u8x8, msg, arg_int, arg_ptr))
	return 1;

	switch (msg) {
		// called once during init phase of u8g2/u8x8
		// can be used to setup pins
		case U8X8_MSG_GPIO_AND_DELAY_INIT:
		CS_DDR |= (CS_BIT);
		DC_DDR |= (DC_BIT);
		RESET_DDR |= (RESET_BIT);
		break;
		// CS (chip select) pin: Output level in arg_int
		case U8X8_MSG_GPIO_CS:
		if (arg_int)
		CS_PORT |= (CS_BIT);
		else
		CS_PORT &= (CS_BIT);
		break;
		// DC (data/cmd, A0, register select) pin: Output level in arg_int
		case U8X8_MSG_GPIO_DC:
		if (arg_int)
		DC_PORT |= (DC_BIT);
		else
		DC_PORT &= (DC_BIT);
		break;
		// Reset pin: Output level in arg_int
		case U8X8_MSG_GPIO_RESET:
		if (arg_int)
		RESET_PORT |= (RESET_BIT);
		else
		RESET_PORT &= ~(RESET_BIT);
		break;
		default:
		u8x8_SetGPIOResult(u8x8, 1);
		break;
	}
	return 1;
}

int main (void) {
	u8g2_Setup_pcd8544_84x48_f(&u8g2, U8G2_R0, u8x8_byte_avr_hw_spi, u8x8_gpio_and_delay);
	u8g2_InitDisplay(&u8g2);		//Send initialization code to the display
	u8g2_SetPowerSave(&u8g2, 0);
	u8g2_SetFont(&u8g2, u8g2_font_miranda_nbp_tr);

	while (1) {
u8g2_ClearBuffer(&u8g2);
//u8g2_SetFontMode(&u8g2,1);
//u8g2_SetDrawColor(&u8g2,1);
u8g2_DrawRBox(&u8g2,0,12,82,15,3);
u8g2_SetDrawColor(&u8g2,2);
u8g2_DrawStr(&u8g2, 2,22, "u8g3 is awesome");
u8g2_SendBuffer(&u8g2);
	}
}

With the following defines in u8x8_avr.c
#define SCK_DDR DDRB
#define SCK_PORT PORTB
#define SCK_BIT 5

#define MOSI_DDR DDRD
#define MOSI_PORT PORTD
#define MOSI_BIT 3

any suggestions?

olikraus, 2019-04-07:
What is the status here? Can I close this issue?

alexiskyriakou, 2019-04-30:
Hello again,

Do you have any example with PCD8544 display with hardware SPI interface?

Thank you

olikraus, 2019-05-01:
I have only examples for Arduino. Will this help?

alexiskyriakou, 2019-05-06:
No because i am working with Atmel studio 7 and a custom board

Thank you

olikraus, 2019-05-06:
hmmm...

closed 2019-05-19 by olikraus