olikraus/u8g2 #813: maximum i2c transfer size

pabigot, 2019-02-11:
The stub I2C callback documented in the [porting guide](https://github.com/olikraus/u8g2/wiki/Porting-to-new-MCU-platform#hardware-i2c-communication) suggests that allocating 32 bytes of local buffer storage to aggregate data is sufficient.

`u8g2_Setup_ssd1306_128x64_noname_f` ends up attempting to transmit 132 bytes in one go.

Is there a generic known maximum length for the I2C transfer, or do I need to determine it per-device by experimentation?

pabigot, 2019-02-11:
Or is that sample just misleading, and it's sufficient to do something like this:
```
    case U8X8_MSG_BYTE_SEND:
      i2c_transfer(u8x8_GetI2CAddress(u8x8) >> 1, arg_int, (uint8_t *)arg_ptr);
      break;
    case U8X8_MSG_BYTE_START_TRANSFER:
    case U8X8_MSG_BYTE_END_TRANSFER:
      break;
```
(i.e. each `BYTE_SEND` is self-contained/complete)?

olikraus, 2019-02-11:
> u8g2_Setup_ssd1306_128x64_noname_f ends up attempting to transmit 132 bytes in one go.

This should not happen. From where do you know this? How did you measure this?

pabigot, 2019-02-11:
The (C++) implementation I'm working on sets up with:
```
  u8g2_Setup_ssd1306_128x64_noname_f(&u8g2,
                                     U8G2_R0,
                                     msg_twi0,
                                     gpio_delay_cb);
  u8x8_SetI2CAddress(&u8g2.u8x8, 0x3C);
```
with
```
uint8_t
msg_twi0 (u8x8_t* u8x8,
          uint8_t msg,
          uint8_t arg_int,
          void* arg_ptr)
{
  static uint8_t buffer[256];    // u8g2 documented to never exceed 32
  static uint8_t count{};
// stuff
    case U8X8_MSG_BYTE_START_TRANSFER:
      count = 0;
      break;
    case U8X8_MSG_BYTE_END_TRANSFER:
      // transmit code here
      break;
    case U8X8_MSG_BYTE_SEND: {
       if (1 < arg_int) {
        printf("send %u from %p, %u already\n", arg_int, arg_ptr, count);
      }
      auto incount = count;
      memcpy(buffer + count, static_cast<const uint8_t *>(arg_ptr), arg_int);
      count += arg_int;
      if (sizeof(buffer) < count) {
        printf("count from %u to %u\n", incount, count);
        uptime::sleep(100);
        while (true);
      }
    }
      break;
```
From this I'm getting:
```
send 128 from 0x2000028f, 4 already
send 128 from 0x2000030f, 4 already
send 128 from 0x2000038f, 4 already
```
The call sequence is four SEND operations of one byte each, followed by a send of 128 bytes.

olikraus, 2019-02-11:
Actually it should be brocken down to smaller pieces: 
https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_cad.c#L509

Do you use the latest version?

pabigot, 2019-02-11:
Ah; no.  Seems I was back a couple revs past 2.23.18 (4dd7a0127151a155ab3223d5c724c88b65624de7 to be exact).

However, I just updated to current master fcee05824436a01d48366265b453836f2a446ee5 and see the same behavior.

Which makes sense, because I used `u8g2_Setup_ssd1306_128x64_noname_f` which sets up to use `u8x8_cad_001` rather than `u8g2_Setup_ssd1306_i2c_128x64_noname_1` which uses `u8x8_cad_ssd13xx_fast_i2c`, which is the one reworked to limit the amount of data fed into the Arduino driver buffers.

Switching probably wouldn't work in my case, because I do need all the data in a single I2C bus transfer: I am in fact using the Adafruit OLED featherwing, but not with Arduino.

I think I know enough of what's going on to move forward, but it also looks like the 32-byte maximum per send is not true for all displays and platforms.

olikraus, 2019-02-11:
Yes, true, the 32-byte limit is for I2C only, but should be there for all plattforms...

closed 2019-04-03 by olikraus