olikraus/u8g2 #815: no destructor?

DaleSchultz, 2019-02-11:
Hi

I am working on a library that sometimes uses g8u2. In order to get things to compile I therefore always have to declare a g8u2 object, even if a particular project does not use a display.

In order to save resources in such projects I though I would call the destructor on the dummy u8g2 object.

I tried:
` ~u8g2;`
and
` ~u8g2();`

but the compiler complains with:

> error: no match for 'operator~' (operand type is 'U8G2_SSD1306_128X64_NONAME_F_SW_I2C')

Does this mean there is no destructor defined? or am I calling it incorrectly?

Is there a way to free up the resources?  `delete` needs a pointer to u8g2.

Alternatively is the en 'empty' constructor we can use that will allow compilation of all the methods but not actually work, but minimizes resources? Or which constructor uses the least resources?

olikraus, 2019-02-12:
Indeed there is no destructor, because there is nothing what needs to be deallocated (actually I am not 100%, because U8g2 is derived from print class and I do not know whether print has a destructor).

U8g2 is developed for embedded systems with very small amount of RAM. Often new/delete/malloc/free are not required and should be avoided in embedded systems. As a consequence, U8g2 is a complete static library. Everything is allocated at compile/link time.

> if a particular project does not use a display.

maybe use an #ifdef conditional compile here

> In order to save resources in such projects I though I would call the destructor on the dummy u8g2 object.

Not sure whether this will really save something. As soon as the compiler/linker sees the u8g2 constructor or setup function, the required resources are reserved/foreseen by the linker.

> Is there a way to free up the resources? `delete` needs a pointer to u8g2.

I assume the u8g2 object itself could be allocted via new. So delete should work in such a case, However the frame buffer will stay in such a case.

> Alternatively is the en 'empty' constructor we can use that will allow compilation of all the methods but not actually work, but minimizes resources? Or which constructor uses the least resources?

There is  a U8G2_NULL class if i remember correctly, but I also remember, that there was a bug with it. It needs to be tested better.

DaleSchultz, 2019-02-12:
ok thanks very much. I will try the U8G2_NULL constructor and also the 'new' approach if I run out or resources.

Pharap, 2019-03-05:
@DaleSchultz 

Your problem is that you're trying to call the destructor incorrectly.

You'd need to call it as `u8g2.~U8G2_SSD1306_128X64_NONAME_F_SW_I2C();`
(The pattern is `object.~Type();`.)

Though as @olikraus says, you probably won't gain anything by calling it because it won't deallocate the memory assigned for the object.

---

@olikraus 

If a class didn't have a destructor then you wouldn't be allowed to instantiate it and you'd get a compiler error if you tried.
```cpp
class Object
{
public:
	~Object() = delete;
};

Object object;
```

An instantiable object _always_ has to have a destructor, even if that destructor does nothing.

DaleSchultz, 2019-03-05:
many thanks!

olikraus, 2019-03-05:
Actually i am a little bit confused by this issue. From my understanding, a default destructor is created if there is not explicit destructor.

Pharap, 2019-03-06:
@olikraus 

Yes, that's right. That's what I'm saying - there is actually a constructor, @DaleSchultz was just trying to call it incorrectly.

So basically the issue's premise of "there's no destructor" is incorrect,
the real issue was @DaleSchultz not calling the destructor correctly.

However, as both you and I have said, calling the destructor won't actually 'clean up resources' as such.

Assuming `u8g2` is statically allocated (i.e. a global variable) then the memory can never be reclaimed.
If it were dynamically allocated (through `new`) then it could be deallocated (through `delete`),
but as you've said that wouldn't reclaim the statically allocated frame buffer.

It would be possible to modify the code to accept an external frame buffer (that could be either statically or dynamically allocated), but I'm not sure what the implications of that would be.
Obviously it would increase the object size by the size of an extra pointer, but it might also prevent certain compiler optimisations and/or cause a performance impact.
It might be worth investigating if you have time though, especially if anyone else has a similar issue where they want to be able to reclaim some memory.

Either way, I think that's a separate issue.

---

@DaleSchultz, if you are interested in saving memory then it's probably better to open a new issue asking about how to save memory, assuming you haven't already found a way to do so.

olikraus, 2019-03-06:
Usually all Arduino Libs and the Core Arduino Firmware come without dynamic memory allocation. Adding dynamic allocation to U8g2 would also require the inclusion of code for new/malloc and delete/free.

Pharap, 2019-03-06:
Arduino has `new` and `delete` available:

https://github.com/arduino/ArduinoCore-avr/blob/master/cores/arduino/new.h

(They just call `malloc` and `free` of course - the compiler is the thing that inserts the calling of the constructors and destructors.)

You wouldn't have to add the dynamic allocation to the library, just the option to use an external RAM buffer.
It would then be up to the user to allocate the buffer (statically or dynamically) and dispose of the buffer when it's no longer needed.

Obviously that has the downside that they have to manually delete the buffer rather than being able to rely on a destructor to do it for them (in the C++ API at least),
but it avoids tying the allocation method to the library, which in turn would mean the user is free to use `malloc` in C and `new` in C++.

It would however be possible to write a wrapper class that handles buffer allocation. E.g.
```cpp
class StaticU8G2 : public U8G2
{
private:
	static buffer[bufferSize];
	
public:
	StaticU8G2() : U8G2(buffer) {}
};

class DynamicU8G2 : public U8G2
{
public:
	DynamicU8G2() : U8G2(new char[bufferSize]) {}
	
	// It might be better to not bother with a copy constructor,
	// making the object non-copyable might make more sense
	DynamicU8G2(const DynamicU8G2 & other)
	{
		// copy other's buffer
	}
	
	DynamicU8G2(DynamicU8G2 && other) : buffer(other.buffer)
	{
		other.buffer = nullptr;
	}
	
	~DynamicU8G2()
	{
		delete this->buffer;
	}
	
	// It might be better to not bother with a copy operator,
	// making the object non-copyable might make more sense
	DynamicU8G2 & operator =(const DynamicU8G2 & other)
	{
		// copy other's buffer
		return *this;
	}
	
	DynamicU8G2 & operator =(DynamicU8G2 && other)
	{
		delete this->buffer;
		this->buffer = other.buffer;
		other.buffer = nullptr;
		return *this;
	}
};
```
(There might be more to handle than just the buffer, but I'll assume the buffer is the only resource you need to care about here.)

If the library were pure C++ and C didn't have to be accounted for then there would be a better solution using templates and allocators.

olikraus, 2019-03-06:
All true.
I just think, that dynamic memory allocation is usually not used in embedded systems.

Pharap, 2019-03-06:
It's just an example/suggestion.
It depends whether your userbase would find it useful or not.

Knowing more about why @DaleSchultz would like to free up some resources would be useful,
but as I say, that should probably be considered a separate issue.

DaleSchultz, 2019-03-06:
I created my own Class library that accepts commands over TCP and carries out these commands. Sometime these commands drive an OLED screen and sometimes not. I thought that when the application was not using an OLED screen I could perhaps reduce the footprint by calling the destructor. 

Since it would not free up any memory, it would not help anyway.
I am using ESP8266 have not run out of memory (yet!)

Pharap, 2019-03-07:
It sort of makes sense, I can see how getting rid of the framebuffer when it wasn't needed could be useful.
But it's a bit drastic because you'd have to reinitialise the screen every time,
and you'd have to be able to provide that memory or your program could crash.

If you're not struggling for space then you probably don't need to worry about memory use yet.
(I.e. you might be looking for memory prematurely.)

But even then, it would make sense to look to other parts of the code first.
Usually there's some simple savings that can be made before the more drastic ones.

If the screen driving code is only used occaisionally and there are no easier savings to make,
then you might be better off seeing if you can avoid using a framebuffer altogether.
Without intending to drive you away from this library, another library might be more suitable for that.

ESP chips have quite a lot of memory anyway, so you should be fine for a while.

closed 2019-04-03 by olikraus