olikraus/u8g2 #816: updateDisplayArea places some junk on screen

DaleSchultz, 2019-02-12:
I find that updating certain tiles using updateDisplayArea  places garbage near the top of the screen, outside away from the tiles specified. If I then call updateDisplay() the screen display is corrected showing that the buffer is good.

`U8G2_SSD1306_128X64_NONAME_F_SW_I2C   u8g2  (U8G2_R0, /* clock=*/ D5, /* data=*/ D3);`

```
u8g2.begin();
u8g2.clearBuffer();
u8g2.updateDisplayArea(14,8,1,1);  // places garbage near top of screen 
delay(2000);     // provide time to see the garbage
u8g2.updateDisplay(); // repaints screen as it should be (all blank)
```

DaleSchultz, 2019-02-12:
OK, I forgot the tiles were zero based and I was requesting to update row 8, beyond the buffer, so it was writing data from beyond the buffer wrapping around onto row zero.

Perhaps the code should not attempt to write data for tiles beyond the screen limit?

closed 2019-02-12 by DaleSchultz

olikraus, 2019-02-12:
Hmmm... Ok thanks. Will think about an extra check.

olikraus, 2019-02-12:
I have updated the documentation.