olikraus/u8g2 #817: UC1608 240x128 unpredictable display

meanderingwalrus, 2019-02-12:
Hello,

I'm trying to use the UC1608 chip with a 240x128 display from BuyDisplays.com and a Teensy3.5 with the 8080 interface.

I've been having unpredictable results. For a time, the display was only displaying data every two times the board is programmed. After a short while the display would work after every program. Now the display will not display anything. The display has been unpredictable like this for the last 4 days of debgging.

I'm using `U8G2_UC1608_240X128_F_8080 u8g2(U8G2_R0, D0, D1, D2, D3, D4, D5, D6, D7, ENABLE, CS, DC, RESET);` as a constructor.

My thinking is that because the screen *sometimes* displays perfectly the BUS, CS,CD and ENABLE must be correctly defined and connected. I'm unsure what else could be wrong with the software/setup.

Any help or guidance would be greatly appreciated. thank you

```#include <Arduino.h>
#include <Encoder.h>

#include <U8g2lib.h>

// Display Definitions
#define D0      32
#define D1      31
#define D2      30
#define D3      29
#define D4      28
#define D5      27
#define D6      26
#define D7      25
#define ENABLE  3 // AKA  Write
#define CS      6 // AKA /CE
#define DC      4 // AKA C/D
#define RESET   5
#define DISP_RD 2

//misc
#define BK_LT_PWR   22
#define _SHTDWN     23

// Input
#define ENC1        14
#define ENC2        15

U8G2_UC1608_240X128_F_8080 u8g2(U8G2_R0, D0, D1, D2, D3, D4, D5, D6, D7, ENABLE, CS, DC, RESET);

Encoder myEncoder(ENC1, ENC2);
int prev_pos = 0;

void setup() {
  
  pinMode(_SHTDWN,OUTPUT);
  digitalWrite(_SHTDWN,HIGH);
  
//  pinMode(ENABLE,OUTPUT);
//  pinMode(CS,OUTPUT);
//  pinMode(DC,OUTPUT);
//  pinMode(RESET,OUTPUT);
  
  u8g2.begin();
  delay(250);
  u8g2.setFontMode(1);
  
  pinMode(BK_LT_PWR,OUTPUT);
  analogWrite(BK_LT_PWR,240); 
}

void loop() {
  u8g2.setDrawColor(0); // clears written pixels to transparent
  u8g2.drawBox((prev_pos+2)%150, 100-17, 100, 17); // clears old text
  
  prev_pos = myEncoder.read();

  u8g2.setDrawColor(1);
  u8g2.setFont(u8g2_font_ncenR08_tr);
  u8g2.setCursor((prev_pos+2)%150, 100); //draws text at a position defined by encoder
  u8g2.print("This text does not print");
  u8g2.sendBuffer();

  delay(10);
}```

olikraus, 2019-02-13:
To me it sounds like a hardware problem. Many resons could be there: Power supply not good enough, wrong/broken caps, broken wires.
Did you solder all connections?

meanderingwalrus, 2019-02-13:
It definitely does sound like a hardware problem, doesn't it? 
The display has only a FPC cable as connection and the SMD ZIF socket looks well soldered in, have tried two of the PCBs I designed but no luck with the 2nd.

olikraus, 2019-02-13:
> Powering via laptop USB port, should be good enough, right?

Depends. If you use the same port for the backlight, then current might be too high

> Unsure what caps you are referring to, but the data sheet suggested two to put on the board and I followed it's recommended but will double check

Use ceramic caps as close as to the FPC connector. Also put one or more ceramic cap into the power supply lines also close to the FPC connector.

meanderingwalrus, 2019-02-17:
Thanks for your help! Issue resolved, stupidly connected the ~RD pin of display to Teensy but didn't do anything with it. It defaulted to LOW and was obviously confusing the display a little. All works fine now

closed 2019-02-17 by meanderingwalrus

olikraus, 2019-02-18:
:)