olikraus/u8g2 #820: SSD1322 : newhaven display, only 248 pixels width even with 16 bit mode enabled

Sanzache, 2019-02-14:
Hello Olikraus & the others, 
I'm using this display : [(http://www.newhavendisplay.com/specs/NHD-3.12-25664UCW2.pdf)] 
with Arduino mkr1000 
and this constructor : U8G2_SSD1322_NHD_256X64_F_4W_HW_SPI

Before 16bit mode, the returned width was 240 and I had misalignment problems as seen in other issues. I looked in the issues and found the fix but uncommenting the #define.
With 16 bit mode enabled, the returned widtht is 248, the misalignments problems are solved but I still cannot acces the last 8 pixels.

What am I doing wrong?

Thanks for the help ! :)

Sanzache, 2019-02-14:
Okay my bad, deleting the lib, then reinstalling the last update and uncommenting the  #define U8G2_16BIT **before** recompiling the project solved the issue !

olikraus, 2019-02-14:
:-)

Sanzache, 2019-02-14:
Thanks a lot for that wonderful lib Oli, it helped me a lot!!!
You can close this issue if you want.
Maybe it will help someone someday :)

closed 2019-04-06 by olikraus