olikraus/u8g2 #821: setI2CAddress Member Function not using correct I2C Address

crackingthecodethomas, 2019-02-15:
We have discovered a bug with the setI2CAddress function. It appears to require a modified I2C Address to work properly. Our display address when scanned is 0x3C. When used in this function nothing would appear on the display. However if we bit shifted it 1 left (0x3C << 1) and used 0x78, the display again would show the proper text. I believe the library is just directly using the address as the first byte which doesn't work for the 7 bit addresses. The R/W bit is left out. There is another post referring to STM32 i2c (SH1106) which i believe has the same issue and have posted a comment there as well. Thanks so much for this useful library.

olikraus, 2019-02-16:
I replied to the other post also.

I took me 30 sec to find this SSD1306 OLED from my lab: 
[image]

Whatever I would do or implement, if people do not know anything about I2C bus and also do not read the manual, they will be confused.

No doubt, I could also highlight a lot of other important things in the manual...

olikraus, 2019-02-16:
Anyhow. Thanks a lot for jumping in at #781 
I am happy about anyone supporting me in providing answers to the community.
Thanks!

LoRaTracker, 2019-03-05:
Just found this problem myself, I could not understand why the setI2CAddress() function actually stopped the display working. 

Mostly when troubleshooting an I2C device, such as a display, I run an I2C scanner to check the address is correct................. 

Very confusing that the I2C scanner, and just about every other I2C device I have used, uses an address format the u8x8 does not accept.

olikraus, 2019-03-05:
The value you have optained from the scanner has to be multiplied by two. This is mentioned in the u8g2 user manual. So I would say that the address is not accepted. It has to be applied correctly according to the user manual.

closed 2019-04-07 by olikraus