olikraus/u8g2 #822: Support for reverse display

fornellas, 2019-02-16:
I just found out that the [SH1106](https://www.displayfuture.com/Display/datasheet/controller/SH1106.pdf) has support for setting normal / reverse display (page 30). The current setup actually set it to normal (0x0A6), but there seems to be no interface to set it to reverse. I fiddled with it, and sending 0x0A1 works to set it to reverse.

How about, just like u8g2_SetContrast() (U8X8_WITH_SET_CONTRAST), we add u8g2_SetReverse() (U8X8_WITH_SET_REVERSE) to allow just that?

I can cut a pull request with this implementation for SH1106 / SSD1306, for which I can test. Other displays can come later.

olikraus, 2019-02-16:
As mentioned here 
https://github.com/olikraus/u8g2/blob/master/doc/faq.txt#L252, 
the reverse command could be sent like this:

```
  u8x8_cad_StartTransfer(u8g2.GetU8x8());
  u8x8_cad_SendCmd( u8g2.GetU8x8(), 0x0a1);
  u8x8_cad_EndTransfer(u8g2.GetU8x8());
```

I am not happy with adding such a reverse command to the lib itself. It would mean to update all the displays and check all datasheet whether the reverse command is possible or not. And finally I would need to test all the displays, which would be a big effort.

fornellas, 2019-02-17:
Fair point. I would love to have this command sequence as something of the shelf, even if it is specific for some displays only, so people don't have to go through the datasheet as I have.

How would you feel about having display specific functions then, for example `u8g2_sh1106_SetReverse()`? For example, I found out that SH1106 supports scrolling, which will not be available to all others, and still useful.

olikraus, 2019-02-17:
Maybe I should add an extra command to the display. 
I think something like 
`u8g2.SendCommand(0x0a1)`
could be implemented.
We could then have an extra wiki page which describes nice commands for some specific controller.

fornellas, 2019-02-18:
TBH, I fell in love with u8g2, precisely because it has loads of things of the shelf, including the awesome HAL, for which I've sent patches for AVR previously. Having display specific functionality as of the shelf functions, IMHO, seems more user friendly that having a wiki page with random hex codes for each display... we already have the datasheets for that.

I'm happy to send a pull request with [SH1106](https://www.displayfuture.com/Display/datasheet/controller/SH1106.pdf) scroll and reverse functionality, with functions, using any of the 2 suggestions I gave above, or something else you come up with. If you're interested, of course.

olikraus, 2019-02-20:
Meanwhile I think more about some expert examples, which send commands directly to the OLED.
I do hesitate to expand the HAL, but would prefer a simplified send command instead.

fornellas, 2019-02-20:
So how about at least having the defines for the hex values at least?

fornellas, 2019-02-20:
Perhaps at least have the defines for the hex values for each display command?

olikraus, 2019-02-20:
I have added the sendF command.

It looks like this: https://github.com/olikraus/u8g2/blob/master/sys/arduino/u8g2_page_buffer/HardwareInverse/HardwareInverse.ino#L296

The "c" stands for command. You could also write "ca", then the function takes two more args, one command and one arg.

There is also a pure C variant of the same function.

olikraus, 2019-02-21:
Added another example for hardware scroll:
https://github.com/olikraus/u8g2/blob/master/sys/arduino/u8g2_page_buffer/HardwareScroll/HardwareScroll.ino#L293

It also gives a better idea of the SendF command.

fornellas, 2019-02-21:
It is what it is... Thanks for the example.

witc, 2019-08-12:
> I have added the sendF command.
>
> It looks like this: https://github.com/olikraus/u8g2/blob/master/sys/arduino/u8g2_page_buffer/HardwareInverse/HardwareInverse.ino#L296
>
> The "c" stands for command. [...]

Dear Olikraus,
I can not find the pure C variant for SendF() anywhere.  Yould you please navigate me?

Thank you very much. Jan.

olikraus, 2019-08-12:
It is not there, but you could copy the code from here: https://github.com/olikraus/u8g2/blob/ff3098326fa90ceebdd48ca7a09f2cfa0d333ba7/cppsrc/U8g2lib.h#L66

I can also add this function once...

olikraus, 2020-01-26:
`void u8x8_SendF(u8x8_t * u8x8, const char *fmt, ...)` is implemented for u8x8

olikraus, 2020-01-26:
added `void u8g2_SendF(u8g2_t * u8g2, const char *fmt, ...)`

olikraus, 2020-01-26:
done

closed 2020-01-26 by olikraus