olikraus/u8g2 #824: Help with RLE Decoding

roberto314, 2019-02-21:
Hello everybody, i am writing a python screen simulator. Purpose is to simulate the position of different fonts on a virtual display to reduce the "code - flash - look - repeat" routine with microcontrollers.
First draft uses the bdf fonts in "u8g2/tools/font/bdf/*.bdf" and works well enough.
Code:
`
	from graphics import *
	from bdflib import reader
	
	FONTP='../ChibiStudio/u8g2/tools/font/bdf/'
	LCDx=240
	LCDy=128
	ZOOMx=2
	ZOOMy=2
	#---------------- Block 1 -----------------
	FONT1='6x9.bdf'
	LINE1_T='Hallo Welt!'
	LINE1_X=0
	LINE1_Y=18
	#---------------- Block 2 -----------------
	FONT2='10x20.bdf'
	LINE2_T=' *+,-./0123'
	LINE2_X=60
	LINE2_Y=20
	
	def makePoint(x, y, win):
		# If Zoom = 1 make point, else make black filled rectagle.
	    rect = Rectangle(Point(x*ZOOMx,y*ZOOMy), Point(x*ZOOMx+(ZOOMx-1), y*ZOOMy+(ZOOMy-1)))
	    rect.setFill("black")
	    rect.draw(win)
	
	def makeGLine(x, y, data, width, win):
		for i in range(width):
			mask = 2**(width-i-1)
			#print('Making Point: ', i, ' Mask: ', mask)
			if ((data & mask) > 0):
				makePoint(x, y, win)
			x = x+1
	
	def makeGlyph(x, y, data, width, win):
		length = (len(data))
		for i in range(length):
			#print('Making Line: ', i)
			makeGLine(x, y, data[i], width, win)
			y = y-1
	
	def main():
	    
		win = GraphWin('LCD Display - Click anywhere to quit', LCDx*ZOOMx, LCDy*ZOOMy)
		win.setBackground('white')
	
		font = reader.read_bdf(open(FONTP + FONT1, "rb"))
		width = int(font[b"AVERAGE_WIDTH"] / 10)
		x = LINE1_X
		for t in range(len(LINE1_T)):
			l = font[ord(LINE1_T[t])]
			makeGlyph(x, LINE1_Y, l.data, width, win)
			x = x + width
	    
		font = reader.read_bdf(open(FONTP + FONT2, "rb"))
		width = int(font[b"AVERAGE_WIDTH"] / 10)
		x = LINE2_X
		for t in range(len(LINE2_T)):
			l = font[ord(LINE2_T[t])]
			makeGlyph(x, LINE2_Y, l.data, width, win)
			x = x + width
	
	
		win.getMouse()
		win.close()
	
	main()
`
The second version uses the actual RLE encoded fonts from "u8g2/csrc/u8g2_fonts.c" and decodes the actual font. I am lost however the decoding works. I read up on the documentation here (the wiki on github), the thread on microcontroller.net (https://www.mikrocontroller.net/topic/443768?goto=5362290) and i found a pdf (2018-04-05-2.U8G2_Doku.pdf), also wikipedia and various other sources but i just don't get the decode process.
For Example i have the font 8x13.bdf and the character space which is:
STARTCHAR space
ENCODING 32
SWIDTH 568 0
DWIDTH 8 0
BBX 8 13 0 -2
BITMAP
00
00
00
00
00
00
00
00
00
00
00
00
00
ENDCHAR
and encoded: 
Header:
Number of Glyphs:  18
Bounding Box Mode:  0
m0 Bit width of Zero-Bit run-length encoding in bitmap:  3
m1 Bit width of One-Bit run-length encoding in bitmap:  2
Bit width of glyph bitmap width (bitcntW):  3
Bit width of glyph bitmap height (bitcntH):  4
Bit width of glyph bitmap x offset (bitcntX):  3
Bit width of glyph bitmap y offset (bitcntY):  5
Bit width of glyph character pitch, delta to next glyph (bitcntD):  5
Font Bounding Box width:  7
Font Bounding Box heigth:  12
Font Bounding Box x Offset:  0
Font Bounding Box y Offset:  255
Ascent (size above baseline) of letter A:  11
Descent (size below baseline) of letter g:  254
Ascent of (:  11
Descent of (:  254
Array offset of glyph A:  0
Array offset of glyph a:  0
Array offset of glyph 0x0100:  206

Fontdata: [32, 0, 110, 12]
Where 32 is the ASCII Code and 0,110,12 the actual RLE Coded Fontdata.

Please help.
Thank you.

This is the second Version of the Python Program without Graphic - just a Parser for the fontdata.
I applogize for the code formatting, it won't work with python (not in the preview anyway).

`#!/usr/bin/env python3
#import mmap
import codecs

FONTS='../ChibiStudio/u8g2/csrc/u8g2_fonts.c'
#FONT="u8g2_font_10x20_tn"
#FONT=b"u8g2_font_profont10_tf"

def decode_font(lines):
	#decode first line
	#print(fd[0])
	fl  = lines[0]	#first line is name and length information
	fd = lines[1:]	#Rest of Font Data
	#print(fl)
	#print(fd)
	#val = fd.split('[', 1)[1].split(']')[0]
	size = int(fl[fl.find(b"[")+1 : fl.find(b"]")])	#Split length info
	print("size: ", size)

	fontdata = []		#make list of data
	for i in fd:
		fontdata.extend(list(i))
	#hdrcomp = list(fd[0]) 	# Header Data is in first line
	hdr = fontdata[:23]		# Header Size is 23 Byte
	print("Number of Glyphs: ", hdr[0])
	print("Bounding Box Mode: ", hdr[1])
	print("m0 Bit width of Zero-Bit run-length encoding in bitmap: ", hdr[2])
	print("m1 Bit width of One-Bit run-length encoding in bitmap: ", hdr[3])
	print("Bit width of glyph bitmap width (bitcntW): ", hdr[4])
	print("Bit width of glyph bitmap height (bitcntH): ", hdr[5])
	print("Bit width of glyph bitmap x offset (bitcntX): ", hdr[6])
	print("Bit width of glyph bitmap y offset (bitcntY): ", hdr[7])
	print("Bit width of glyph character pitch, delta to next glyph (bitcntD): ", hdr[8])
	print("Font Bounding Box width: ", hdr[9])
	print("Font Bounding Box heigth: ", hdr[10])
	print("Font Bounding Box x Offset: ", hdr[11])
	print("Font Bounding Box y Offset: ", hdr[12])
	print("Ascent (size above baseline) of letter A: ", hdr[13])
	print("Descent (size below baseline) of letter g: ", hdr[14])
	print("Ascent of (: ", hdr[15])
	print("Descent of (: ", hdr[16])
	print("Array offset of glyph A: ", int.from_bytes(hdr[17:19], byteorder='big'))
	print("Array offset of glyph a: ", int.from_bytes(hdr[19:21], byteorder='big'))
	print("Array offset of glyph 0x0100: ", int.from_bytes(hdr[21:23], byteorder='big'))
	#print("Fist Line Data: ", hdrcomp)
	#print("Header 21-22: ", hdr[21:23])

	rest = fontdata[23:]
	#print("Rest: ", rest)
	#print("Glyph Data: ")
	glyph = []
	offset = 0
	pos = 0
	nrglyph = hdr[0]
	print("\n")	
	print("Nr. of Glyphs: ", nrglyph)
	offset=rest[1]	# offset to next glyph is at byte 1
	low = 0
	up = offset

	for gn in range(0, nrglyph):
		#ngl = rest[low:up].pop(1)
		ngl = rest[low:up]
		del ngl[1]				# remove the second itel, it is the number of bytes
		#ngl = ngl.pop(1)
		glyph.append(ngl)
		print("Glyph", gn, "ASCII:", chr(rest[low]), "low: ", low, "Up: ", up, "Off: ", offset)
		low = up
		offset = rest[low+1]
		up = up + offset
	print("\n")

	#for i in range(0, nrglyph):	#Print Glyphdata one Glyph at a time
		#print(glyph[i])
	
	glyph.insert(0, hdr) # we put the header data in the first list

	#print("\n")	
	#print("Glyphdata: ", glyph)
	
	return glyph

def u8g2_SetFont(fname):
	lines = []
	ln = 0
	start = 0
	SSTR = ("const uint8_t " + fname).encode('utf-8') # to limt results to one!
	with open(FONTS, 'rb') as s:
		for line in s:
			ln = ln + 1
			if line.find(SSTR) != -1:
				#print(ln)
				start = 1
			if start == 1:
				#lines.append(line)
				#stl = line[2:].rstrip(b'"')
				stl = line.strip()
				stl = stl.rstrip(b'\n')
				stl = stl.rstrip(b';')
				stl = stl.strip(b'"')
				lines.append(codecs.escape_decode(stl)[0])
				if line.find(b';\n') != -1:
					glyph = decode_font(lines)
					#return lines
					return glyph

def decode_glyph(gl, g):
	#print(gl[g][:])
	print("Char: ", chr(gl[g][0]), "Index: ", g, "Data: ", gl[g][:])
	print("Header: ", gl[0][:])
	print("Char: ", (gl[g][0]))
	#print("glyph bitmap width: (RLE) ", gl[g][1])
	#print("glyph bitmap height: (RLE) ", gl[g][2])
	#print("glyph bitmap x offset: (RLE) ", gl[g][3])
	#print("glyph bitmap y offset: (RLE) ", gl[g][4])
	#print("character pitch: (RLE) ", gl[g][5])
	#print("Bitmap: (RLE) ", gl[g][6:])
	return

def find_glyph(gd, ch):
	#print("Length: ", len(gd))
	for i in range(1,len(gd)):		# we have to start at the second list because the first list is the header!
		#print("Loop", i)
		if gd[i][0] == ord(ch):
			#print("Found it..")
			#return gd[i]
			return i
	print("Not found")
	return

def u8g2_DrawStr(gl, x, y, str):
	for ch in str:
		#print(ch)
		g = find_glyph(gl, ch)
		#print("Index: ", g)
		gd = decode_glyph(gl, g)

def main():
	#gl = u8g2_SetFont("u8g2_font_10x20_tn")
	gl = u8g2_SetFont("u8g2_font_8x13_tn")
	#fl = u8g2_SetFont("u8g2_font_logisoso30_tn")
	#print(gl[0][:])
	u8g2_DrawStr(gl, 0, 20, " ")

main()`

olikraus, 2019-02-21:
Just to confirm this, you found this page: https://github.com/olikraus/u8g2/wiki/u8g2fontformat, right?

The actual RLE decoding happens here for the glyph data:
https://github.com/olikraus/u8g2/blob/master/csrc/u8g2_font.c#L564

It basically is a bit sequence of trhee values:
<number of clear pixel> <number of set pixel> <repeat flag>

<number of clear pixel> is stored in variable a, <number of set pixel> is stored in variable b. The flag itself is used to terminate the while loop.

roberto314, 2019-02-21:
I already found the first Link (in fact it was the first page i found).
I try to make sense of the c code.
Thanks.

roberto314, 2019-02-25:
Good morning everyone,
i am sorry to say i am lost.....
I looked at the C-Code and it looks easy enough at first glance but it is very difficult to figure out all the variables and all the functions:
- u8g2_font_decode_get_unsigned_bits
- u8g2_font_decode_len
- u8g2_add_vector_x
- u8g2_add_vector_x
and so on.
If somebody could be so kind and give me an example to get from this: (Font 8x13.bdf character Space, folder: u8g2/tools/font/bdf/8x13.bdf)
STARTCHAR space
ENCODING 32
SWIDTH 568 0
DWIDTH 8 0
BBX 8 13 0 -2
BITMAP
00
00
00
00
00
00
00
00
00
00
00
00
00
ENDCHAR
to this:
Char:    Index:  1 Data:  [32, 0, 110, 12]
Header:  [18, 0, 3, 2, 3, 4, 3, 5, 5, 7, 12, 0, 255, 11, 254, 11, 254, 0, 0, 0, 0, 0, 206]
Char:  32
which is from u8g2_fonts.c Line 7209, Font: u8g2_font_8x13_tn converted to decimal.
Also a second character would be nice for me to check my thoughts.
Maybe:
STARTCHAR slash
ENCODING 47
SWIDTH 568 0
DWIDTH 8 0
BBX 8 13 0 -2
BITMAP
00
00
02
02
04
08
10
20
40
80
80
00
00
ENDCHAR
to:
Char:  / Index:  7 Data:  [47, 79, 66, 236, 106, 159, 99, 0]
Header:  [18, 0, 3, 2, 3, 4, 3, 5, 5, 7, 12, 0, 255, 11, 254, 11, 254, 0, 0, 0, 0, 0, 206]
Char:  47
Thank you.

olikraus, 2019-02-25:
just to save some effort, can you convert  [47, 79, 66, 236, 106, 159, 99, 0] into binary (01 seqeuence) numbers and place it here?

roberto314, 2019-02-25:
Is this ok?
Character SPACE:
Char:    Index:  1 Data:  [32, 0, 110, 12]
Hex Data:  0x20
Binary Data:  0b00100000
Hex Data:  0x00
Binary Data:  0b00000000
Hex Data:  0x6e
Binary Data:  0b01101110
Hex Data:  0x0c
Binary Data:  0b00001100
Binary Data: 
Header:  [18, 0, 3, 2, 3, 4, 3, 5, 5, 7, 12, 0, 255, 11, 254, 11, 254, 0, 0, 0, 0, 0, 206]
Char:  32
Character SLASH:
Char:  / Index:  7 Data:  [47, 79, 66, 236, 106, 159, 99, 0]
Hex Data:  0x2f
Binary Data:  0b00101111
Hex Data:  0x4f
Binary Data:  0b01001111
Hex Data:  0x42
Binary Data:  0b01000010
Hex Data:  0xec
Binary Data:  0b11101100
Hex Data:  0x6a
Binary Data:  0b01101010
Hex Data:  0x9f
Binary Data:  0b10011111
Hex Data:  0x63
Binary Data:  0b01100011
Hex Data:  0x00
Binary Data:  0b00000000
Binary Data: 
Header:  [18, 0, 3, 2, 3, 4, 3, 5, 5, 7, 12, 0, 255, 11, 254, 11, 254, 0, 0, 0, 0, 0, 206]
Char:  47

roberto314, 2019-02-25:
Different formatting (although i don't think that will be an additional difficulty):
Char:  / Index:  7 Data:  [47, 79, 66, 236, 106, 159, 99, 0]
Hex Data:  ['0x2f', '0x4f', '0x42', '0xec', '0x6a', '0x9f', '0x63', '0x00']
Binary Data:  ['00101111', '01001111', '01000010', '11101100', '01101010', '10011111', '01100011', '00000000']
Header:  [18, 0, 3, 2, 3, 4, 3, 5, 5, 7, 12, 0, 255, 11, 254, 11, 254, 0, 0, 0, 0, 0, 206]
Char:  47
Binary2:  00101111 01001111 01000010 11101100 01101010 10011111 01100011 00000000

olikraus, 2019-02-25:
Lets start with the header:
```
  1		bbx_mode	0: proportional, 1: common height, 2: monospace, 3: multiple of 8  --> 0
  2		bits_per_0	glyph rle parameter --> 3
  3		bits_per_1	glyph rle parameter --> 2
  4		bits_per_char_width		glyph rle parameter --> 3
  5		bits_per_char_height	glyph rle parameter --> 4
  6		bits_per_char_x		glyph rle parameter --> 3 
  7		bits_per_char_y		glyph rle parameter--> 5
  8		bits_per_delta_x		glyph rle parameter --> 5
```

ok... will interrupt here and come back to this topic later...

olikraus, 2019-02-25:
ok, I continue. Cosider the data:
[47, 79, 66, 236, 106, 159, 99, 0]
 ['0x2f', '0x4f', '0x42', '0xec', '0x6a', '0x9f', '0x63', '0x00']
 00101111 01001111 01000010 11101100 01101010 10011111 01100011 00000000

The first byte is the ascii code: 47, so we can just consider the rest.
 ['0x4f', '0x42', '0xec', '0x6a', '0x9f', '0x63', '0x00']
01001111 01000010 11101100 01101010 10011111 01100011 00000000

According to this:
https://github.com/olikraus/u8g2/blob/master/csrc/u8g2_font.c#L465
the first two values are width (3 bits) and height (4bits).
So we start with the lowest three bits of the first byte (01001111), which are 111, so the width is 111 (seven).
The remaining bits are 01001. From these bits we take the next 4 bits, which is 1001 (nine), so the height is 9.

to be continued...

roberto314, 2019-02-25:
OK, i can follow you. I already decoded the header:
Bounding Box Mode:  0
m0 Bit width of Zero-Bit run-length encoding in bitmap:  3
m1 Bit width of One-Bit run-length encoding in bitmap:  2
Bit width of glyph bitmap width (bitcntW):  3
Bit width of glyph bitmap height (bitcntH):  4
Bit width of glyph bitmap x offset (bitcntX):  3
Bit width of glyph bitmap y offset (bitcntY):  5
Bit width of glyph character pitch, delta to next glyph (bitcntD):  5

roberto314, 2019-02-25:
Maybe somebody can check if my python script is accessible at: https://bitbucket.org/100prozent/u8g2_display_simulator/src/default/read_fonts_test.py

Output so far:
Char:  / Index:  7 Data:  [47, 79, 66, 236, 106, 159, 99, 0]
Glyph data cut:  [79, 66, 236, 106, 159, 99, 0]
Header:  [18, 0, 3, 2, 3, 4, 3, 5, 5, 7, 12, 0, 255, 11, 254, 11, 254, 0, 0, 0, 0, 0, 206]
Binary Glyph Data wo ASCII:  01001111010000101110110001101010100111110110001100000000
wu:  8 wl:  5 hu:  5 hl:  1
glyph bitmap width: (RLE)  bitwidth:  3 bitstr:  111 dec:  7
glyph bitmap height: (RLE)  bitwidth:  4 bitstr:  1001 dec:  9

roberto314, 2019-02-25:
Ok, i continued with the rest of the header variables, i am not sure about the signed though.
Char:  / Index:  7 Data:  [47, 79, 66, 236, 106, 159, 99, 0]
Glyph data cut:  [79, 66, 236, 106, 159, 99, 0]
Header:  [18, 0, 3, 2, 3, 4, 3, 5, 5, 7, 12, 0, 255, 11, 254, 11, 254, 0, 0, 0, 0, 0, 206]
Binary Glyph Data wo ASCII:  01001111010000101110110001101010100111110110001100000000
Binary Glyph Data wo ASCII inverted:  00000000011000111001111101101010111011000100001001001111
length of bitstring:  56
glyph bitmap width: (RLE)  bitwidth:  3 bitstr:  111 dec:  7 I am at:  3
glyph bitmap height: (RLE)  bitwidth:  4 bitstr:  1001 dec:  9 I am at:  7
glyph bitmap x offset: (RLE)  bitwidth:  3 bitstr:  100 dec:  4 I am at:  10
glyph bitmap y offset: (RLE)  bitwidth:  5 bitstr:  10000 dec:  16 I am at:  15
glyph character pitch: (RLE)  bitwidth:  5 bitstr:  11000 dec:  24 I am at:  20
Bitmap:  000000000110001110011111011010101110

olikraus, 2019-02-25:
let me recall the BDF file, char '/':
BITMAP
00
00
02
02
04
08
the first two lines are zero and are ignored by u8g2:
BITMAP
02
02
04
08
So the first bitmap lines of the char "/" are
00000010
00000010
00000100
00001000

in fact this looks like the upper part of the slash, so let us check this in the encoded version...

['0x4f', '0x42', '0xec', '0x6a', '0x9f', '0x63', '0x00']
```
xhhhhwww pyyyyyxx 1000pppp ------f1
01001111 01000010 11101100 01101010 10011111 01100011 00000000
```
w: Width, length=3, value = 7
h: Height, length = 4, value = 9
starting from here, we have signed values, which have to be adjusted by 1<<(length-1)
https://github.com/olikraus/u8g2/blob/master/csrc/u8g2_font.c#L282
x: x offset, length = 3, value = 4 - (1<<(3-1)) = 0
y: y offset, length = 5, value = 16 - (1<<(5-1)) = 0
p: char pitch: length = 5, value = 24 - (1<<(5-1)) = 8
Now the bitmap ifself starts, with the number of "not set bits"/number of zeros.
0: no of zeros: length = 3, value = 6
1: no of one: length = 2, value = 1
f: the repeat flag, length = 1, value = 1 (repeat!)
So until now, the bitmap contains 6x 0 and 1x1, which is then repeated:
00000010000001
For the final bitmap, there is an automatic wrapping at the width of the bitmap (which is 7 here):
0000001
0000001
The original BDF has a width of 8 bits for the chars. This is not true anymore for u8g2, so the number of columns is 7 (the width of the bitmap). But the bitmap itself (until now) is exactly the same as the bitmap from the BDF file (see above)

Let's continue a little bit:
```
xhhhhwww pyyyyyxx 1000pppp 11000ff1 -------f
01001111 01000010 11101100 01101010 10011111 01100011 00000000
```
f: the repeat flag, length = 1, value = 0 --> no further repeat, so we get the next zero line:
0: no of zeros: length = 3, value = 5
1: no of one: length = 2, value = 1
f: the repeat flag, length = 1, value = 1 (repeat!)
So we have
0000001  6 x zero, 1 x one
0000001  repeat 6 x zero, 1 x one
0000010  5 x zero, 1 x one, then repeat
00001     

Can you see how the bitmap gets constructed?

roberto314, 2019-02-26:
I played a little with my encoder script and i am still unsure about some details.
First let me show you what i've got so far (i inverted the bytes so the bitstream is easier to read):

Header:  [18, 0, 3, 2, 3, 4, 3, 5, 5, 7, 12, 0, 255, 11, 254, 11, 254, 0, 0, 0, 0, 0, 206]
Char:  / Index:  7 Data:  [47, 79, 66, 236, 106, 159, 99, 0]
Glyph data cut and inverted:   [0, 99, 159, 106, 236, 66, 79]
Binary Glyph Data wo ASCII (bytewise inverted):          00000000011000111001111101101010111011000100001001001111
glyph bitmap width:  (RLE)    bitwidth:  3 dec:  7 out:  00000000011000111001111101101010111011000100001001001
glyph bitmap height: (RLE)    bitwidth:  4 dec:  9 out:  0000000001100011100111110110101011101100010000100
Bitval:  4 Power:  4 Result:  0
glyph bitmap x offset: (RLE)  bitwidth:  3 dec:  0 out:  0000000001100011100111110110101011101100010000
Bitval:  16 Power:  16 Result:  0
glyph bitmap y offset: (RLE)  bitwidth:  5 dec:  0 out:  00000000011000111001111101101010111011000
Bitval:  24 Power:  16 Result:  8
glyph character Pitch: (RLE)  bitwidth:  5 dec:  8 out:  000000000110001110011111011010101110
Decoding Bitmap...
No. of zeros:  3 No. of ones:  2
--------------`xxxxxxxOOZZZFOOZZZFFFFFFOOZZZFFOOZZZ`
`Bitmap:         000000000110001110011111011010101110`
zeros:  6 out:  000000000110001110011111011010101
ones:   1 out:  0000000001100011100111110110101 Value:  1
Flag:  1 out:   000000000110001110011111011010 Value:  1
Flag:  0 out:   00000000011000111001111101101
zeros:  5 out:  00000000011000111001111101
ones:   1 out:  000000000110001110011111 Value:  2
Flag:  1 out:   00000000011000111001111 Value:  4
Cnt:  1
Flag:  1 out:   0000000001100011100111 Value:  8
Flag:  1 out:   000000000110001110011 Value:  16
Flag:  1 out:   00000000011000111001 Value:  32
Flag:  1 out:   0000000001100011100 Value:  64
Flag:  0 out:   000000000110001110
zeros:  6 out:  000000000110001
ones:   1 out:  0000000001100 Value:  1
Flag:  0 out:   000000000110 Value:  1
zeros:  6 out:  000000000
ones:   0 out:  0000000 Value:  0
Flag:  0 out:   000000 Value:  0
Cnt:  1
zeros:  0 out:  000
ones:   0 out:  0 Value:  0
Flag:  0 out:    Value:  0
Cnt:  7
Finished. [1, 1, 2, 4, 8, 16, 32, 64, 1, 0, 0]

I am not sure about the repetition.
If a have ones + zeros < width do i fill the rest with zeroes?
I tried that at the second repetition flag because i have 5 zeros + 1 ones = 6, which is one less than 7 but than the first not repeated line is wrong because i have 6 zeros and 1 ones which is 1 but should be 128.
I am sorry but i am afraid i need a ltitle bit more of your excellent explaination :)
Did you check my bitbucket account? Can you get the file?

olikraus, 2019-02-26:
> I am not sure about the repetition.

Repetition means, that a previous zeros and ones pair is repeated. The number if repetitions depends on the flags (or better the flag sequence).
Here are some examples of the flag sequence:
0: no repetition
10: Repeat the previous zeros and once once more (ah sounds odd)
110: repeat it twice
1110: repeat three times
11110: guess what... repeat it four times
111110: so the general rule is: repeat as often as there is a one flag,

>zeros: 6 out: 000000000110001110011111011010101
ones: 1 out: 0000000001100011100111110110101 Value: 1
Flag: 1 out: 000000000110001110011111011010 Value: 1

Not sure what you mean by value, but if the flag is one, you have to output the 6 zeroes and the 1 ones again with the flag. And in fact you have to do the same once again with any other flag which is one until you reach a zero-value flag.

I quote myself:

> Now the bitmap ifself starts, with the number of "not set bits"/number of zeros.
0: no of zeros: length = 3, value = 6
1: no of one: length = 2, value = 1
f: the repeat flag, length = 1, value = 1 (repeat!)
So until now, the bitmap contains 6x 0 and 1x1, which is then repeated:
00000010000001

The bitmap starts with the value 6 and 1, so we have the following bitmap:
0000001

The the repeat flag is set, and the 6 and 1 pair is again connected:
00000010000001

If the would be another one-flag, then it would look like this:
000000100000010000001

Not sure whether this clarifies this...

> If a have ones + zeros < width do i fill the rest with zeroes?

This will not happen, again, I quote myself:

> For the final bitmap, there is an automatic wrapping at the width of the bitmap

Maybe the "/" example is a little bit bad with respect to ths, because accidently zeros + ones is equal to width for the first row, but if you follow my example from the previous post, you will see, that the zeroes and once are just concatenated, while the line break happens independently (just go through my previous example completely)

> I tried that at the second repetition flag because i have 5 zeros + 1 ones = 6, which is one less than 7 but than the first not repeated line is wrong because i have 6 zeros and 1 ones which is 1 but should be 128.

Not sure whether you still talk about the line breaks here. My suggestion is: Just output the zeros and ones as a data stream, then apply the line breaking later.

> Can you get the file?

Probably yes, but I usually do not review any others code...

roberto314, 2019-02-26:
Ok, i think that clarifies things a bit.
The tip:
>Not sure whether you still talk about the line breaks here. My suggestion is: Just output the zeros and ones >as a data stream, then apply the line breaking later.
was very helpful.
The reason i ask if you can get the code from bitbucket was not to review my code, i understand that you won't do that, but to publish the finished script. It could be helpful to others.

olikraus, 2019-02-26:
Sure, it can be placed here, but probably user will require some hints about the purpose and what problems could be solved with your program.

roberto314, 2019-02-26:
Sure, i'll get to that. :)

roberto314, 2019-02-28:
Ok, my script is nearly functional but i am having troubles with the space character.
Output is:
Header:  [18, 0, 3, 2, 3, 4, 3, 5, 5, 7, 12, 0, 255, 11, 254, 11, 254, 0, 0, 0, 0, 0, 206]
Char:    Index:  1 Data:  [32, 0, 110, 12]
Glyph data cut and inverted:   [12, 110, 0]
Binary Glyph Data wo ASCII (bytewise inverted):          000011000110111000000000
glyph bitmap width:  (RLE)    bitwidth:  3 dec:  0 out:  000011000110111000000
glyph bitmap height: (RLE)    bitwidth:  4 dec:  0 out:  00001100011011100
Bitval:  4 Power:  4 Result:  0
glyph bitmap x offset: (RLE)  bitwidth:  3 dec:  0 out:  00001100011011
Bitval:  27 Power:  16 Result:  11
glyph bitmap y offset: (RLE)  bitwidth:  5 dec:  11 out:  000011000
Bitval:  24 Power:  16 Result:  8
glyph character Pitch: (RLE)  bitwidth:  5 dec:  8 out:  0000
Decoding Bitmap...
No. of zeros:  3 No. of ones:  2
Bitmap in: 0000
As you can see, after stripping the RLE Header i am left with four zeros as Bitmap. Shouldn't there be at least six (3-zeros, 2-ones, one flag)?

olikraus, 2019-02-28:
Any output is limited to the number of pixel in the bitmap.
The number of pixel is width x height. Any output is limited by that number. For the char the number of pixel is 0 x 0 = 0. That means there will be no output.
The space is nothing else, than a shift by the glyph character pitch.

roberto314, 2019-03-01:
I am happy to report that there is a working python script at: https://bitbucket.org/100prozent/u8g2_display_simulator.
In fact there are two scripts, one is just a parser and outputs the bitmap like the _bdflib_ python module, the other outputs the actual font in a window which is user-configurable (size, height, zoom).
The Reason for this is simply to position the text on the lcd display without having to reflash the poor microcontroller dozens of times.

olikraus, 2019-03-01:
Nice. Great Job!
Probably I need to learn how to execute Python :-/

roberto314, 2019-03-01:
Well on linux it is quite easy, just make the files executable (chmod ugo+x _file_) and then simply run it from the commandline. You need python3 and codecs which you can find in the package manager.
With Windows i am not sure, i gave that up a while ago.

closed 2019-03-04 by roberto314

olikraus, 2019-03-04:
Thanks for the linux hints :-)