olikraus/u8g2 #825: SSD1309 i2c requires reset pin

JamesGKent, 2019-02-21:
Was testing a SSD1309 based display today and found that the display would not initialise and display anything, I had a look through the datasheet for the controller and it appears that is expected behaviour.

If the reset is required should it be either a compile error or documented in the setup section of the wiki?

Many thanks for the library

olikraus, 2019-02-21:
I personally would claim, that all displays which have a reset pin also require some logic at this pin. So from that perspective I would say, that your comment applies to most of the displays.

But: U8g2 can not replace a datasheet. Still many users are able to use their displays without reading the datasheet in some simple cases (like a I2C SSD1306 OLED), but usually it is required to read the datasheet to understand all the setup requirements of your display. As an open source lib with the limited resources of a spare time programmer, it is simply impossible to document all the extra requirements for each displays (nevertheless, I always try to answer setup related questions here...)

And finally: The fact the the reset is optional in the constructor, does not mean you should not implement the reset. Instead it means, that the reset line can be skipped in such cases where you have your own reset logic (like a dedicated reset controller or a simple RC network)

Well, not sure whether this helps. 
Thanks for using U8g2!

JamesGKent, 2019-02-21:
You make a very good point, I guess I've been lucky that none of the displays I've used until now had this issue.

Your library is much appreciated, I'll close the issue, thanks.

closed 2019-02-21 by JamesGKent

olikraus, 2019-02-21:
:)