olikraus/u8g2 #826: OLED with RF24

TheMisieekPL, 2019-02-21:
Hello!
I want to use my 128x128 OLED from waveshare in SPI mode with NRF24L01 module and I've got some issues. Each one of the modules work without a problem when the other one isn't yet initiated. Oled doesn't show any info when I initiate both libraries. I don't really know what's the problem. Maybe I have to change something in libraries to make it work? Using arduino nano, RF24 library by TMRh20, and U8g2lib by Olikraus. 
Connections:
NRF24L01:
CE - 4
CSN - 5
SCK - 13
MOSI - 11
MISO - 12
OLED: 
DIN - 11
CLK - 13
CS - 10
DC - 9

I tried to use I2C instead of SPI, but it's much slower, and "freezes" the microcontroller for some time so I'm not able to look almost constantly if some data from RF24 has appeared... The communication is initiated by pressing the button connected do pin 3.

Here's the code of receiver with OLED. 
`#include <Arduino.h>
#include <U8g2lib.h>
//#include <Wire.h>
#include <SPI.h>
#include <nRF24L01.h>
#include <RF24.h>

//U8G2_SSD1327_MIDAS_128X128_1_SW_I2C u8g2(U8G2_R0, 5, 4 , 8 );
U8G2_SSD1327_MIDAS_128X128_1_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 13, /* data=*/ 11, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);

RF24 radio(4, 5); // CE, CSN

const byte addresses[][6] = {"00001", "00002"};

boolean buttonstate = 0;
int powtorzenia = 0;

void setup() {
  Serial.begin(9600);
  radio.begin(); 
  radio.openWritingPipe(addresses[1]); // 00001
  radio.openReadingPipe(1, addresses[0]); // 00002
  radio.setPALevel(RF24_PA_MIN);
  pinMode(3, INPUT); 
  u8g2.begin();  

}
void loop() {
    u8g2.firstPage();
  do {
    u8g2.setFont(u8g2_font_ncenB10_tr);
    u8g2.drawStr(0,24,"Hello World!");
  } while ( u8g2.nextPage() );
}
  if(digitalRead(3) == 1){
    Serial.println("button==1");
    buttonstate = 1;
    radio.stopListening();
    radio.write(&buttonstate, sizeof(buttonstate));
  }
  if (buttonstate == 1)
 { 
    radio.startListening();
    if ( radio.available()) {
    while (radio.available()){ 
    radio.read(&powtorzenia, sizeof(powtorzenia));
    Serial.println(powtorzenia);
    }
    }
    
    if (powtorzenia > 100) buttonstate =0;
    
  }
}`

I've found out, that I can call SPI.end(); after using RF24 and then u8g2.begin(); to use OLED again, but screen blinks, and initiation of both modules in turns takes too long, so some data may be gone.

olikraus, 2019-02-21:
Do not mix software emulated SPI with hardware SPI.
This is somehow similar to the first question of the u8g2 FAQ: https://github.com/olikraus/u8g2/blob/master/doc/faq.txt#L9 which applies also to SPI.
The solution is to use the "HW" constructor for your display.

TheMisieekPL, 2019-02-21:
Ohhhh! Thank you sooooo much! I couldn't solve this problem for three days! 

I know it's a bit off topic, but... Do you have any ideas how can I use oled and check simultaneously if any data has appeared on RF? My idea is to update OLED only when RF got something, and keep OLED unrefreshed with u8g2.setAutoPageClear(0) command just not to make it dark. 

Thanks for help!

olikraus, 2019-02-21:
> Do you have any ideas how can I use oled and check simultaneously if any data has appeared on RF?

Probably your could will look like this:

```
if ( data_received ){
  u8g2.firstPage();
  do {
    // draw / print data
  } while ( u8g2.nextPage() );
}
```

"data_received" might be a function call or a variable, which is set to 1 if any data is there. However, how to set the variable depends on the RF lib (which I do not know).

TheMisieekPL, 2019-02-21:
Thank you! I will ask if I have another problem that I won't be able to solve.

Greetings from Poland.

closed 2019-02-21 by TheMisieekPL

BlackGrom, 2019-10-31:
TheMisieekPL
How did you solve your problem i don't understand