olikraus/u8g2 #828: U8x8 future support for 1x Open Iconic font?

DirtyEngineer, 2019-02-24:
Wondering if support for the smallest size Open Iconic fonts for U8x8 is on the road map? The 2x times version seems to work well.  I'm specifically interested in the "open_iconic_arrow_1x" variant.  Great library! Thank you for all your hard work.

olikraus, 2019-02-24:
guess, I just missed this...

olikraus, 2019-03-02:
Ceated new beta 2.26.1

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

closed 2019-03-02 by DirtyEngineer

olikraus, 2019-03-02:
:)